wcbonner/GoveeBTTempLogger #11: Question: Celsius or Fahrenheight?

hancockks, 2020-12-15:
Just pulled the latest code since I saw some errors in the log after some security updates.  I noticed my output has switched to Celsius. Is there an option to determine output?

2020-12-03 02:10:13     67.1729 40.5    89
2020-12-03 02:11:02     67.1729 40.5    89
2020-12-15 13:56:55     20.0346 34.6    86
2020-12-15 13:56:57     20.0345 34.5    86

wcbonner, 2020-12-15:
There's not currently an option to have the log written in Fahrenheit. 

I switched the log to Celsius some time ago because it made debugging new devices much simpler. The devices report values in °C.

hancockks, 2020-12-15:
Thanks! Not really a problem, just wondering if there was a configuration I missed.

closed 2020-12-15 by hancockks