wcbonner/GoveeBTTempLogger #12: temp values are wrong for negative temperatures

kw123, 2021-01-16:
temperature and humidity data is wrong for negative temperature data ( tested using cold spray to generate - C)
Any temperature < 0 gives VERY large positive temp numbers  

spend some time to figure out the 2 complement but it is not that simple. 
The data for the 3bytes + 1 (3 bytes for temp and humidity split at 1000) does not lead to a simple  if data > xx: data -= yy

the one with 2 bytes for temp and 2 bytes for hum is a simple 2 complement for temp (hum is never negative)

wcbonner, 2021-01-16:
Which thermometer are you using?

I thought I'd fixed the problem with the h5074, which is the only one I've left outside, but it doesn't surprise me that I've got problems with the h5075 or h5175.

kw123, 2021-01-16:
int iTemp = int(data[5]) << 16 | int(data[6]) << 8 | int(data[7]);
			Temperature = float(iTemp) / 10000.0;
			Humidity = float(iTemp % 1000) / 10.0;
 
that can not be a negative number for Temperature
I tried: 
if iTemp > 2**23 then  iTemp -= 2**24 
but that gives wrong numbers (~ -870 or so)

kw123, 2021-01-16:
This seems to work:
if iTemp >=8388608: iTemp = 8388608 - iTemp
temp = float(iTemp)/1000.
and for hum then use
hum = abs(iTemp)%1000 / 10.

kw123, 2021-01-16:
8388608 = 2exp(23)

wcbonner, 2021-01-16:
Thanks. I'll see if I can get it fixed tonight when I'm back near my development machine.

I didn't see an answer as to which device you are working with.

kw123, 2021-01-16:
I actually only use your formula. I have several rpi in the house and they collect various info.  It’s all written in python. 
And I have all the Govee temp sensors that use Bluetooth. 
I tried the -C with a small square one - forgot the name 
Karl

kw123, 2021-01-16:
the device name: HS5101

But the problem in the code is for both types: one with 3 bytes (that is the complicated one) but also the one with 2 bytes. There it is a simple 2 complement with 16 bits. 
The 3 byte is not a 2 complement. I have never seen it done that way before.
Looks like they where trying to save a byte in the message and made it really complicated.
 Karl

wcbonner, 2021-01-17:
I updated the two supported models (h5075 and h5177) I have that use the three byte encoding to properly handle negative temperatures. (Tested with the h5075 by putting it in my freezer) 

The two byte data model code of the h5074 had already been tested to work properly with negative temperatures. I believe it's working because of the handling of data types. I was specifically using a short integer to handle the data, so the sign data was in a normal bit.

closed 2021-01-17 by wcbonner