wcbonner/GoveeBTTempLogger #14: Daylight savings time - not changing timestamp

ant-thomas, 2021-03-29:
I'm in the UK and the clocks went forward at the weekend.
The output from the GoveeBTTempLogger doesn't reflect the system time and is now an hour behind.
Is there a way to get the program to change the logged time according to the actual system time?

Only thing I've tried so far is to restart the service, and the output remains 1 hour behind.

Or is the time set on the device?

wcbonner, 2021-03-29:
Which time are you referring to? 

The logs should always be stored as UTC, so it should be unaffected by Daylight Savings Time / Summertime changes.

The SVG graphs attempt to use local time calls for display.

ant-thomas, 2021-03-29:
I'm running the program via the systemd service with this command...
```/usr/local/bin/goveebttemplogger -v 0 -t 1 -l /dev/shm/```

Output is to the log file is....
```2021-03-29 18:43:01     21.0599 59.9    81
2021-03-29 18:43:03     21.0599 59.9    81
2021-03-29 18:43:05     21.0599 59.9    81
2021-03-29 18:43:07     21.0599 59.9    81
2021-03-29 18:43:09     21.0599 59.9    81
2021-03-29 18:43:11     21.0599 59.9    81
```

Current time is 19:43

wcbonner, 2021-03-29:
It sounds to me like you are saying that things are operating as designed. The log file is written as UTC, with no time zone or summertime adjustment. 

Since you happen to match up with UTC during the winter, logs look like they are your local time during the winter, when they are actually representing UTC.

ant-thomas, 2021-03-29:
I see, I wrongly assumed it would use system time.
No problem and no issue then. I will manipulate the output based on system time now I know this. Thanks for the quick response.

closed 2021-03-29 by ant-thomas