wcbonner/GoveeBTTempLogger #15: Excessive CPU use on Raspberry Pi 3

alinsavix, 2021-04-07:
Howdy!

I was trying to set up GoveeBTTempLogger on the Pi3 I am currently using as an octoprint server (to monitor my filament storage humidity), and I got everything working well, except for one problem -- it uses up a steady 20% of the pi's CPU!  This seems a bit excessive, just to be able to write out a few stats to a file once a minute! And on a device that needs to be fairly responsive (due to driving real hardware), this is not ideal.

A quick trace of the process suggests that it is in a fairly tight loop of trying to read from the HCI socket, getting EAGAIN (presumably a nonblocking socket), sleeping for a very short period of time, and repeating. Quick back-of-the envelope math says it's doing this ~10000 times/second.

Is there any way GoveeBTTempLogger could be updated to either use blocking reads (with timeouts for doing things like updating graphs and such, if those features are enabled), or to poll at a much slower rate than it is currently doing? I would really like to be able to use this software!

This is using the provided .deb package, build 2.20210405-2

Thanks!

wcbonner, 2021-04-07:
I'd love to implement your suggestion. I used to know how to configure timeouts on network sockets but haven't figured it out for Bluetooth sockets. If you have pointers to examples or documentation I'd appreciate it.

Wim

alinsavix, 2021-04-07:
Bluetooth sockets _should_ behave basically identically -- linux puts a bunch of effort into making sockets and streams and such behave consistently, regardless of underlying implementation. 

That being said, if I manage to remember (and have time) tomorrow, I'll do a bit of nosing around and try to provide a more concrete answer there.

wcbonner, 2021-04-07:
I've added timeout code to the Bluetooth socket setup. Initially it's not throwing any errors and still seems to be working. I'll run it on my production system and see if I can get a new release package out today.

wcbonner, 2021-04-07:
Quick note about the PiZeroW I've got running this remotely, which is both the lowest power processor, and in a relatively quiet Bluetooth environment.

ps aux results before upgrade:
root     16205 25.7  0.7   4640  3320 ?        Ss   Apr05 668:40 /usr/local/bin/goveebttemplogger -v 0 -l /var/log/goveebttemplogger/ --svg /var/www/html/goveebttemplogger/ --battery 8 --minmax 8

Install new package.. 
Unpacking goveebttemplogger (2.20210407-1) over (2.20210405-2) ...

ps aux results after upgrade:
root      5223 30.6  0.7   4684  3284 ?        Ss   11:52   4:59 /usr/local/bin/goveebttemplogger -v 0 -l /var/log/goveebttemplogger/ --svg /var/www/html/goveebttemplogger/ --battery 8 --minmax 8

I'll let this go for several hours before making any judgments, because during the startup it's running all the work of reading the log files.

wcbonner, 2021-04-08:
The PiZeroW after running the new code for nearly a day is looking very similar to the older, nonblocking, code.

root      5223 26.2  0.7   4640  3288 ?        Rs   Apr07 358:59 /usr/local/bin/goveebttemplogger -v 0 -l /var/log/goveebttemplogger/ --svg /var/www/html/goveebttemplogger/ --battery 8 --minmax 8

I'm going to call this a release, and hopefully you can give me feedback.

alinsavix, 2021-04-13:
Apologies for the delay in getting you feedback on this.

I gave things a quick run, and it looks like your changes for this don't actually address the problem -- the CPU use is pretty much identical, and traces look pretty much the same.

I did a little looking around, and it seems like the most likely culprit is https://github.com/wcbonner/GoveeBTTempLogger/blob/master/goveebttemplogger.cpp#L1918 , which looks like it probably originated with some libbluetooth example code. It seems like the connection is set to be nonblocking, so it just sits there and spins when there's nothing to read.

The weird thing is that looking earlier in the code, it looks like that code actually tries to set the socket to be blocking with a timeout (https://github.com/wcbonner/GoveeBTTempLogger/blob/master/goveebttemplogger.cpp#L1666 and https://github.com/wcbonner/GoveeBTTempLogger/blob/master/goveebttemplogger.cpp#L1671), but there's still this tight loop.

Maybe those calls fail, but I doubt it -- it should print an error message if there's a failure there, and I'm definitely not seeing that (and a quick glance at a trace doesn't show any failures there), so not sure what's up there. Haven't been able to dig much further yet, but that's what I've found so far.

wcbonner, 2021-04-13:
What you are seeing is exactly what I was seeing. After your initial message, I added the socket calls that configure a socket timeout and put the socket in blocking mode. I left the sleep call in place, because I figured if it didn't sleep, it might really be a runaway polling loop.  I suppose I could try using nanosleep instead of usleep, but I have a feeling that the underlying code would be the same. 

One thing I do know is that there were a lot more bluetooth announcements going on in my local environment than I expected. run the program with a -v 9 option to see a dump of all the information I'm discarding..  I don't really think that's why the read calls return so quickly. I believe it's more likely an issue with the way the bluez bluetooth stack operates.

alinsavix, 2021-04-13:
After spending a while typing up possibilities, I think I realize what the issue is.

A simplified version of the setup code:

```
    if (set_receive_timeout succeeds) {
        nonblocking = 1
    } else {
        nonblocking = 0
    }

    ioctl(device_handle, FIONBIO, (char *) &nonblocking)
```

I verified that for the ioctl argument, a 0 means "nonblocking is disabled" (i.e. the socket will be blocking), and anything else means "nonblocking is enabled", so I'm using the variable "nonblocking" to make it easier to think about what value means which.

I think if you look at it that way, you'll see the problem -- if you can set a read timeout (which only applies to blocking reads -- nonblocking reads always return immediately one way or the other), currently the code uses that as a trigger to enable nonblocking I/O -- and if the read timeout _can't_ be set, then it switches to blocking I/0, which will potentially block forever because a timeout hasn't been set.

I'm going to bet if you reverse the logic there, things will look a lot healthier. My pi is offline right now, so I can't immediately test, but I'm pretty sure I got the logic documented the right way around (and getting them reversed would exactly explain the behavior we're seeing)

alinsavix, 2021-04-15:
Hmmm. I flipped the logic here, as I suggested, but the behavior didn't change.

Color me surprised.

I have the feeling I may be spending some quality time in front of a debugger this weekend.

wcbonner, 2021-04-15:
Thanks for looking at it. I have been busy with other things this week. I'd love to have it properly waiting on the socket instead of using a sleep call.

wcbonner, 2021-04-21:
On my Raspberry Pi4 running several things, mpstat (I think from the sysstat package) reports 84% idle, while the PiZeroW reports only 8% idle. The PiZeroW is only running this program and PiAware. The Pi4 is running more stuff.

wim@WimPi4:~ $ mpstat
Linux 5.10.17-v7l+ (WimPi4)     04/21/21        _armv7l_        (4 CPU)

10:49:48     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest  %gnice   %idle
10:49:48     all   11.69    0.00    3.86    0.21    0.00    0.10    0.00    0.00    0.00   84.13

pi@WimPiZeroW:~ $ mpstat
Linux 5.10.17+ (WimPiZeroW)        21/04/21        _armv6l_        (1 CPU)

11:52:16     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest  %gnice   %idle
11:52:16     all   60.33    0.00   28.25    0.01    0.00    2.61    0.00    0.00    0.00    8.79

alinsavix, 2021-04-22:
Hokay, I broke down and fixed this up with a proper nonblocking workflow using select(). The attached patch is more-or-less the normal way to handle nonblocking I/O, using select() to basically go "block here until that fd is actually servicable". I added a bunch of comments about what the various bits are doing.

Submitting a patch this way rather than a proper PR because there's a decent bit of cleanup you'll probably want to do (and my indents are different than yours, courtesy of clang-format ... oops), so making a full-blown PR out of it is probably not really more useful than just attaching the patch.

I admit that I was surprised at just how much data comes in, and how fast it comes in, even with this patch, but it *does* succeed in lowering the CPU use on my Pi3, from ~18% before, to less than 1% now, and process traces show roughly three orders of magnitude fewer syscalls happening. So, this should put things in a good place, I suspect.

Let me know if you have any questions!

[goveebttemplogger.patch.txt](https://github.com/wcbonner/GoveeBTTempLogger/files/6361613/goveebttemplogger.patch.txt)

wcbonner, 2021-04-23:
I've looked over your patch, and I believe I've implemented the select() statement in my program flow so the select is the blocking call instead of sitting in a sleep call.   The code is in github now. I'm going to run it on my production machine overnight and build a release package in the morning. Initially the results looks great.

wcbonner, 2021-04-24:
Running the new release, I just realized I'm getting a bunch of these messages in the syslog:

kernel: [99201.229573] Bluetooth: hci0: advertising data len corrected 31 -> 29

The good thing is that it's only happening on my local machine, not on either of the secondary machines I've got running the exact same release. It's possible that something strange in my local environment is broadcasting malformed BT-LE packets.  but I'm hoping I've not introduced something strange to the BT Stack. 

The following command shows that it didn't start till significantly after my program was running, so I don't think I would have done significantly bad things. Though after 16:54:31, I seem to be getting a couple of these messages a second.

wim@WimPi4:~ $ sudo grep Bluetooth /var/log/syslog | more
Apr 23 10:58:04 WimPi4 goveebttemplogger[7523]: GoveeBTTempLogger Version 2.20210423-1 Built on: Apr 23 2021 at 10:49:27 (listening for Bluetooth Low Energy Advertisements)
Apr 23 14:30:01 WimPi4 kernel: [86463.904693] Bluetooth: hci0: advertising data len corrected 1 -> 0
Apr 23 16:54:29 WimPi4 kernel: [95193.140742] Bluetooth: hci0: advertising data len corrected 31 -> 29
Apr 23 16:54:31 WimPi4 kernel: [95193.770744] Bluetooth: hci0: advertising data len corrected 31 -> 29
Apr 23 16:54:31 WimPi4 kernel: [95194.931398] Bluetooth: hci0: advertising data len corrected 31 -> 29

wcbonner, 2021-05-20:
Running the new code for the past month. The CPU is now low, and the strange syslog messages seem to have been fixed with a system update.

closed 2021-05-20 by wcbonner