wcbonner/GoveeBTTempLogger #16: PI3 Ubuntu Install Failed

jbronikowski, 2021-04-19:
Trying to install but running into dependancy and cannot locate this specifc armhf for libluetooth3
```
ubuntu@ubuntu:~/GoveeBTTempLogger$ sudo apt-get install bluetooth bluez libbluetooth-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
bluez is already the newest version (5.53-0ubuntu3).
libbluetooth-dev is already the newest version (5.53-0ubuntu3).
bluetooth is already the newest version (5.53-0ubuntu3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
ubuntu@ubuntu:~/GoveeBTTempLogger$ make deb
chmod a+x GoveeBTTempLogger/DEBIAN/postinst GoveeBTTempLogger/DEBIAN/postrm GoveeBTTempLogger/DEBIAN/prerm
dpkg-deb --build GoveeBTTempLogger
dpkg-deb: building package 'goveebttemplogger' in 'GoveeBTTempLogger.deb'.
ubuntu@ubuntu:~/GoveeBTTempLogger$ sudo apt install ./GoveeBTTempLogger.deb
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'goveebttemplogger:armhf' instead of './GoveeBTTempLogger.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 goveebttemplogger:armhf : Depends: libbluetooth3:armhf but it is not installable
E: Unable to correct problems, you have held broken packages.
ubuntu@ubuntu:~/GoveeBTTempLogger$

wcbonner, 2021-04-19:
I've not used ubuntu recently. I just did a quick search for the package and it looks like the library should be available. https://packages.ubuntu.com/xenial/libbluetooth3 after I'd initially looked at https://packages.debian.org/sid/libbluetooth3

Have you tried adding a -y option to your apt install command? In my experience -y causes it to automatically try to install dependencies.

jbronikowski, 2021-04-19:
Yep, I did try that. I will continue to see if I can find a workaround and report back.

tylerobara, 2021-04-25:
@jbronikowski I am going to guess that you are running Ubuntu 64bit on your Pi3 like I am. I ran into this issue as well. I changed [this line](https://github.com/wcbonner/GoveeBTTempLogger/blob/59f8ed12d02b9e9da65d2a6014cf152a4d31badd/GoveeBTTempLogger/DEBIAN/control#L5) from `armhf` to `arm64` and ran `make deb` again. This time everything compiled and installed w/o issue and I am pulling data. Although admittedly I am still figuring out how to pull it consistently, the service seems to pull it in at random times.

wcbonner, 2021-04-25:
@tylerobara Thanks for the suggestion. 

I was just looking at https://www.debian.org/doc/debian-policy/ch-controlfields.html#s-f-architecture and trying to figure out if I should be specifying either a list of architectures or any or all on that line to make the package build and install on multiple architectures. Do you have any insight?

tylerobara, 2021-04-27:
> @tylerobara Thanks for the suggestion.
>
> I was just looking at https://www.debian.org/doc/debian-policy/ch-controlfields.html#s-f-architecture and trying to figure out if I should be specifying [...]

No, sorry. TLDR: I guessed. 
This is a step out of my knowledge level. IIRC I was able to run the `make deb` command without issue but `sudo apt-get install ./GoveeBTTempLogger.deb` failed and gave me the error about `armhf`. I knew I had installed Ubuntu 64bit specifically for something I was testing a long time ago. I dug around and saw the line I mentioned above and took a shot. After running through the make and install again, it worked fine. I would not know how to provide a way for this to be compiled programmatically based on the OS of the user. 
No up for me is pulling this data into Home Assistant and determining if I can scale to multiple of these sensors.

closed 2021-06-14 by wcbonner