wcbonner/GoveeBTTempLogger #19: SVG output in Celsius

GertJanH, 2021-06-14:
is there a option to change de SVG output from Fahrenheight to Celsius?

wcbonner, 2021-06-14:
Not currently. The code was written with the parameter in it, but it looks like I've not exposed it on the command line. It should be an easy fix for me to add this week when I'm looking at the code.

GertJanH, 2021-06-14:
that's a quick answer, and that would be very nice if it could be fixed so quickly.

wcbonner, 2021-06-14:
I just built the current code with the new option added..  either -c or --celsius. 

```
Usage: GoveeBTTempLogger/usr/local/bin/goveebttemplogger [options]
  GoveeBTTempLogger Version 2.20210614-1 Built on: Jun 14 2021 at 14:19:43
  Options:
    -h | --help          Print this message
    -l | --log name      Logging Directory [./]
    -t | --time seconds  Time between log file writes [60]
    -v | --verbose level stdout verbosity level [1]
    -m | --mrtg XX:XX:XX:XX:XX:XX Get last value for this address
    -a | --average minutes [5]
    -s | --svg name      SVG output directory
    -c | --celsius       SVG output using degrees C
    -b | --battery graph Draw the battery status on SVG graphs. 1:daily, 2:weekly, 4:monthly, 8:yearly
    -x | --minmax graph  Draw the minimum and maximum temperature and humidity status on SVG graphs. 1:daily, 2:weekly, 4:monthly, 8:yearly
    -d | --download      Periodically attempt to connect and download stored data
```

GertJanH, 2021-06-15:
That was a fast implementation, and its working good

closed 2021-06-15 by wcbonner