wcbonner/GoveeBTTempLogger #26: TIME setting not being honored

timmerk, 2022-03-16:
I'm using a Govee H5074. I just installed [v2.20220111-1](https://github.com/wcbonner/GoveeBTTempLogger/releases/tag/v2.20220111-1), and I left the service with defaults, so TIME should have defaulted to 60 seconds. I also tried setting the TIME environmental variable to 60 and restarted the service. However, both ways don't seem to "stick":

2022-03-16 02:13:01	26.02	27.4	100
2022-03-16 02:13:15	26.05	27.4	100
2022-03-16 02:13:17	26.05	27.4	100
2022-03-16 02:13:23	26.02	27.42	100
2022-03-16 02:13:25	26.02	27.42	100
2022-03-16 02:13:39	26.04	27.41	100
2022-03-16 02:13:41	26.04	27.41	100
2022-03-16 02:13:55	26.04	27.43	100
2022-03-16 02:13:57	26.04	27.43	100

Any ideas? Thanks!

wcbonner, 2022-03-16:
I should document the time parameter more accurately. What it's doing is the frequency of writes to the log file. 

The timestamps that are actually written inside the log file depend on when the Bluetooth announcements arrive. I'd put the time parameter in place because I wanted to reduce wear on the filesystem.

timmerk, 2022-03-16:
Ahh, thank you! I'm trying to avoid wearing down the filesystem, too (it's an SD card.) What do you mean by frequency? Should I just set it to some very high number? Thank you!

wcbonner, 2022-03-16:
The default is 60 seconds. When I was originally using the program, I'd be running a second instance every five minutes that opened the log file in read only mode to generate a graph for MRTG. That means that if I used a longer write period I'd be missing data in my graph generation. I'm not running it that method anymore, as I generate SVG graphs from data held in ram every five minutes. you could set the time value to any number you want, though I wouldn't recommend more than 300, just because there's always the possibility of losing whatever hasn't been written to disk. 

When I started this project I knew a lot less about the underlying micro sd flash cards. Since then, I've realized that they all have some wear leveling logic build into the firmware, and I've moved to using larger micro sd cards, so I'm running with a lot more free space on the drive, allowing the wear leveling to extend the life of the card itself.

jamesderlin, 2022-06-10:
I think it would be useful to have an ability to throttle the data to not write *all* announcements.  Writing temperature/humidity every few seconds seems excessive for most use cases.  A thermometer that is nearby could generate a few hundred MB of log data per year.  Processing all of that extra data to generate the SVG graphs also seems like a waste.

wcbonner, 2022-06-10:
I can tell you that the largest individual data file I have in over two years of running this service is 46,256,625. The total size of all of my log files over the time is 4,637,050,750.  I have seven thermometers locally, including one in my freezer and one in my refrigerator. I also have several remote thermometers that have data rsynced to my log directory hourly. 

4 gigabytes does not seem like an unreasonable amount over a few years.  If it was, I'd implement a separate log rotation and log compression rule using the standard logrotate tool in linux. I don't want to throw away data the thermometer reports. 

I've found that the thermometer broadcasts data more often in my relatively noisy Bluetooth environment than it does at either my friends home or my boat. I live in an apartment in Seattle with people living above, below, and to both sides of me.

closed 2022-06-10 by wcbonner

jamesderlin, 2022-06-10:
Heh, I guess we have different definitions of reasonable. =)

I consider 4 GB of data for 7 devices over 2 years to be a lot.  I certainly don't need temperature/humidity data with a resolution in seconds.

All that said, I suppose it would not be much trouble to write a separate script to prune the log files afterward as desired.  I'll probably do that instead.