wcbonner/GoveeBTTempLogger #28: Add some ability to make filenames friendlier

jamesderlin, 2022-06-04:
Currently GoveeBTTempLogger generates files named after the devices' Bluetooth address.  This makes inspecting the `.log` file for a particular device rather awkward if there are many devices with similar addresses.   It would be nice if there were a mechanism to change the filenames.

Proposal:
1. Allow `gvh-titlemap.txt` to live in `/usr/local/etc` (possibly with a different filename, e.g. `goveetemplogger-devices.conf`).  For backward compatibility, GoveeBTTempLogger could additionally check for `gvh-titlemap.txt` in the SVG output directory first.
2. Add a command-line argument (e.g. `--map-filenames`) to enable naming files using the mapping file. (I'd personally prefer enabling it by default, but forcing users to opt-in is okay with me for the sake of compatibility.)
3. To avoid filename collisions, the actual filenames would be of the form `TITLE (ADDRESS)`.  For example: `Outside (E35ECC215C0F).{log,svg}`.

If the above sounds good and if you don't foresee any major obstacles, I'm willing to try to implement it.

wcbonner, 2022-06-05:
I don't have time to look at this before Monday. I'll see what it looks like then..

wcbonner, 2022-06-09:
I'm still thinking about the logfile name friendliness issue. 

I've lamented that I chose gvh507x_ as the prefix for quite a while now, as gv_ would have been much nicer as I've learned about more devices. 

Having run the program for over two years with way too many devices, I've got 272 files in my log directory. Initial startup takes quite a while on my Pi4 as it reads all the old data into memory.

One of the advantages of the current naming scheme is that all filenames are the same length, and the Bluetooth address occupies the same position in each filename, as does the year and month.

wcbonner, 2022-06-09:
There's a couple of other issues with trying to make the filenames friendlier via a map file. 

The current filenames are known to be filesystem friendly. It would be very easy to mistakenly put characters in a custom title that work fine for an SVG title, but don't work in a filesystem. Worse, sometimes those names can be created in a filesystem easily, and manipulated from a C program just fine, but shell command line expansion causes funny things to happen, such as making a file nearly impossible to delete, or mistakenly deleting all files in a directory.

If input from a titlemap was included in the filename, a single character change in the map would mean that the data would be logged to an entirely new file. merging the data after the fact is a headache. (Part of the reason I've been living with the prefix I'm no longer happy with for such a long time.)

jamesderlin, 2022-06-09:
> One of the advantages of the current naming scheme is that all filenames are the same length, and the Bluetooth address occupies the same position in each filename, as does the year and month.

While that might be a nice property, is it a *useful* property?  And if it is useful, does that usefulness outweigh utility from being able to find a desired log, especially if the user explicitly opted in to using a custom filename?

> The current filenames are known to be filesystem friendly. It would be very easy to mistakenly put characters in a custom title that work fine for an SVG title, but don't work in a filesystem. [...]

I agree that's a potential problem, but again, what I'm suggesting here is a mechanism for users to opt-in to using custom filenames.  An argument that custom filenames shouldn't be allowed because users *might* give their files problematic filenames also would apply to any other program (including the shell) that allows users to name files.  Ultimately users can't be protected from themselves.

If it's still a concern, I think that 99% of problems could be addressed by:
* Requiring that filenames have a fixed prefix to prevent them from starting with a `-`.
* Disallowing all whitespace that isn't `' '` (0x20).

> If input from a titlemap was included in the filename, a single character change in the map would mean that the data would be logged to an entirely new file. merging the data after the fact is a headache.

That's a good point, but IMO that shouldn't necessarily be GoveeBTTempLogger's responsibility to deal with.  If a user chooses to rename their device, then I think that creating a new set of log files, while not ideal, would not be unreasonable.  Users with sufficient technical aptitude and desire could write scripts to rename their existing files.

Alternatively: instead of renaming the existing files, create symlinks with friendlier names.  I don't particularly like the idea of cluttering directories by doubling the number of files though.  Maybe a satisfactory compromise could be to create friendly symlinks to just the latest log file or perhaps storing all of the symlinks in a separate directory.

jamesderlin, 2022-07-03:
Since I expect backward compatibility to be a sticking point, I've instead written a Python script to make it easier to view the log files I want: https://github.com/jamesderlin/gv-log-tools

closed 2022-07-03 by jamesderlin