wcbonner/GoveeBTTempLogger #34: Checking on H5179, is it supported?

otanim, 2022-07-03:
Can you please confirm if [GoveeBTTempLogger](https://github.com/wcbonner/GoveeBTTempLogger) supports the H5179 device?

It seems like you've listed it in models, but the actual algorithm to decrypt the data is not done. Just wanted to confirm if my assumptions are correct?

Please note that the decryption algorithm is available in another project:
https://github.com/Thrilleratplay/GoveeWatcher/blob/master/debugger/govee_advertisement.py#L108-L114

wcbonner, 2022-07-03:
I see where I added the type name in the code, but I don't think I added support for that device.  I don't think it's listed in the Readme.md file as being supported.

I generally haven't tried to claim the code I created worked for a device if I haven't seen it working, which requires my having one of the devices. If someone with the particular device wants to create a patch supporting the H5179 and test it, I'd be willing to accept the patch into the master branch, but right now I have more thermometers than I need sitting around and don't feel the need to spend money on more.

otanim, 2022-07-03:
@wcbonner, understandable, thanks for the speedy reply, I appreciate that.

Unfortunately, I don't own the device as well, otherwise, I would've given it a shot. It's just noticed some possible advantages over H5074, which's battery life is approximately 12 months, while the H5179 is advertised by Govee as a "500 days" (12 + 11 months) battery life.

closed 2022-07-03 by otanim

wcbonner, 2022-07-03:
I think the 5074 was the small one with the button battery, and I've been incredibly impressed with the battery life. 

I've had one running since I started this project, zip tied to the balcony on my apartment in Seattle, and it's still reporting 74% battery.  I've also got one in my fridge and one in my freezer and they are holding up surprisingly well considering they are constantly going below freezing. 

[image]

otanim, 2022-07-03:
> I think the 5074 was the small one with the button battery, and I've been incredibly impressed with the battery life.

Yes, it's that one, the `H5074` it's using 1x [CR2477](https://www.farnell.com/datasheets/1496886.pdf) battery which is 3V, while the `H5179` is using 3x [AAA](https://www.farnell.com/datasheets/2626067.pdf) batteries (4.5v together). Considering its batteries are quite common to find in stores and it has a longer battery life, makes it worth some specific scenarios, but of course, it depends on the use case and relevancy.

> I've had one running since I started this project, zip tied to the balcony on my apartment in Seattle, and it's still reporting 74% battery. I've also got one in my fridge and one in my freezer and they are holding up surprisingly well considering they are constantly going below freezing.

Regarding battery life, that's impressive, I somehow didn't though zip-tying sensors, but so doing that you'll have "incorrect" results of humidity of what would've been expected.
I have a sensor hanging on my balcony too (since March 12th), but it somehow keeps showing battery life 100%, and now I'm not sure if it's working that great, or it's just reporting wrong data or it's the decryptor that's deciphering the data incorrectly.

Regarding doing testing inside the freezer/fridge, I've considered using [SensorPush HTP.xw](https://www.amazon.com/gp/product/B08PKVRJJX) which is:
- advertised as 24-month battery life,
- has a better temperature range (-40°C - 60°C (-40°F - 140°F)),
- and as advertised can work under low temperature very well (without of zipping).

Unfortunately, this device is:
- way expensive,
- there is no decypher found in open source,
- Sampling Interval is 1m (I believe Govee ones are 2s)
- and currently, the only way to receive data on your computer is to purchase a separate gateway device, which needs to connect to internet and store the data on seller's servers, from where you can retrieve the data using APIs.

wcbonner, 2022-07-04:
The humidity issues in the graph I attribute mostly to unrealistic expectations when I created the particular graph and included min/max. Each data point is an entire day, and humidity is relative humidity. It's quite often that at some point in the 24 hour period it's going to reach a maximum relative humidity of 100%, whether it's cold and hits the dew point, or raining, making the graph itself nearly useless. 

The battery life in the units initially led me to believe that I was interpreting the data incorrectly. What I've really found is that they seem to be incredibly efficient as long as they are only broadcasting data. If you use the phone app to query the data, it goes into a connected Bluetooth protocol that seems to use significantly more power than the BTLE announcements. It's especially true of the units with a display.  

The unit in my freezer dropped it's estimated battery quickly, then didn't drop much more, making me think that the voltage that the device is using to monitor battery state hasn't changed much, but obviously was lower than the expected and calibrated factory voltage should be. I also have a feeling that they are going to work until at some point the voltage drops rapidly and the unit just stops.

[image]

The H5177 unit has a much larger display than the H5075 units and takes three batteries instead of two, but seems to last much more than 150% of the time of the smaller unit.

otanim, 2022-07-04:
> The H5177 unit has a much larger display than the H5075 units and takes three batteries instead of two, but seems to last much more than 150% of the time of the smaller unit.

H5075's battery life is quite bad.. it doesn't even last a month in my case, mostly 20-25 days...

> The humidity issues in the graph I attribute mostly to unrealistic expectations when I created the particular graph and included min/max. Each data point is an entire day, and humidity is relative [...]

Yeah, that's a problem, I would suggest you to try-out interactive graphs too, it can be web-based, i.e. [Chart.js](https://www.chartjs.org/), you can set dynamic min-max, pick your desired duration, zoom-in-out (you need to attach a plugin for this), and even export it as an image file too.

> The battery life in the units initially led me to believe that I was interpreting the data incorrectly. What I've really found is that they seem to be incredibly efficient as long as they are only [...]

oof, yeah, that's a problem.. it would've been so nice to finally figure out how to get the historical data from the Govee devices.. so there won't be a necessity to continuously monitor LE beacons constantly..