wcbonner/GoveeBTTempLogger #35: new argument "--no-logging" to disable the local file logging

otanim, 2022-07-05:
Can you please add an option to disable the logging in the filesystem?

wcbonner, 2022-07-05:
I'm trying to understand the usefulness of no logging. 

On the other hand, since the normal method I see this run is as a service with the log directory specified in the unit file, changing the behavior so that if the log file is not specified, no logs are created. 

Would that satisfy the same need as the option you were looking for?

Wim

otanim, 2022-07-05:
> I'm trying to understand the usefulness of no logging.

I'm running an external app that is a wrapper for `goveebttemplogger` program, it basically constantly reads the output stream and makes its own table in a different structure.

I'm basically monitoring 8 different sensors live.

wcbonner, 2022-07-05:
> > I'm trying to understand the usefulness of no logging.
> 
> 
> 
> I'm running an external app which is a wrapper for `goveebttemplogger` program, it basically constantly reads the output stream and makes its own table in a different structure.
> 

Interesting. I can probably get the change so it doesn't do logging if a log directory isn't specified done tomorrow. I'm driving home from my long weekend today and just checking in from my phone periodically. 

Wim

otanim, 2022-07-05:
You're very generous, thanks @wcbonner, I really appreciate your help :pray:

Take your time :beach_umbrella:

closed 2022-07-07 by otanim

wcbonner, 2022-07-07:
Cool Map. Glad to be able to help. 

I've got 7 thermometers in my studio apartment. Three are on different walls of the main room, in part because I was interested in knowing if it made a difference being closer to the outside wall, It is currently 2 degrees F warmer at the outside wall and desk than the inside wall nearer my bed.

otanim, 2022-07-07:
Oh, an interesting experiment :) But please consider that there might be various other factors as well: water pipes, electricity cables, height, etc.

 I can suggest to you a better solution for this, I mean surely it's an unnecessary countermeasure, but still might interest you:
 You can use an infrared thermal imaging camera (I use [HT-301](https://www.amazon.com/dp/B07Z7J6LXD)), which makes visualization of a thermal map way better, i.e. in my case I didn't even know that the fridge is warming up my 15cm wall from the other side this much :)
[image]