wcbonner/GoveeBTTempLogger #36: H5179?

alec2dabreen, 2022-09-16:
Does this work with H5179? I'm not getting any BLE readings, so was just wondering.

wcbonner, 2022-09-16:
> Does this work with H5179? I'm not getting any BLE readings, so was just wondering.

No. I never bought a H5179 so didn't have the raw data in a format to be sure I was decoding it properly. I've got too many h5074 and h5075 sitting around to justify spending more on another device.

My first 5074 stopped responding after two years. My decoding still thought it had 58% battery. I bought a ten pack of batteries for $10, replaced its button battery and it's working fine again, reporting battery at 100%.

thisgarryhas2rs, 2022-10-28:
I have a H5181 (like the H5182 but only one probe). Can I provide raw data to help decipher the information for that model?

thisgarryhas2rs, 2022-10-28:
It actually looks like it is working but just shows the second temp as 0. 
```
2022-10-28 03:18:31     33      0       100     5183    -0.01
2022-10-28 03:18:34     33      0       100     5183    -0.01
2022-10-28 03:18:36     33      0       100     5183    -0.01
2022-10-28 03:23:38     29      0       100     5183    -0.01
2022-10-28 03:23:39     29      0       100     5183    -0.01
2022-10-28 03:23:42     29      0       100     5183    -0.01
2022-10-28 03:28:45     32      0       100     5183    -0.01
2022-10-28 03:28:48     32      0       100     5183    -0.01
2022-10-28 03:28:49     32      0       100     5183    -0.01
```

wcbonner, 2022-10-28:
> No. I never bought a H5179 so didn't have the raw data in a format to be sure I was decoding it properly. I've got too many h5074 and h5075 sitting around to justify spending more on another [...]

@alec2dabreen I might be able to add support for the H5179 if you can supply some data in a comment by running the program so it's only listening to the Bluetooth address of your device similar to the following example:

```
wim@WimPi4-Dev:~/GoveeBTTempLogger $ sudo /home/visualstudio/projects/GoveeBTTempLogger/bin/ARM/Debug/GoveeBTTempLogger.out  -v 3 -o A4:C1:38:EC:0B:03
[2022-10-28T17:35:33] GoveeBTTempLogger Version 2.20221007-5 Built on: Oct 28 2022 at 10:31:54
[                   ]      log: 
[                   ]      svg: 
[                   ]  battery: 0
[                   ]   minmax: 0
[                   ]  celsius: false
[                   ] titlemap: 
[                   ]     time: 60
[                   ]  average: 5
[                   ] download: false
[2022-10-28T17:35:33] LocalName: WimPi4-Dev
[2022-10-28T17:35:33] BlueTooth Address Filter: [A4:C1:38:EC:0B:03]
[2022-10-28T17:35:33] Scanning...
[2022-10-28T17:35:43] 46 [A4:C1:38:EC:0B:03] (bdaddr_type) 00 (evt_type) 00 (Name) GVH5075_0B03 (UUID) 88EC (Flags) 05 (Manu) 88EC0002F4146300 (Temp) 19.3556°C (Humidity) 55.6% (Battery) 99% (GVH5075)
[2022-10-28T17:35:45] 46 [A4:C1:38:EC:0B:03] (bdaddr_type) 00 (evt_type) 00 (Name) GVH5075_0B03 (UUID) 88EC (Flags) 05 (Manu) 88EC0002F4146300 (Temp) 19.3556°C (Humidity) 55.6% (Battery) 99% (GVH5075)
[2022-10-28T17:35:45] 42 [A4:C1:38:EC:0B:03] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FFC2 (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) C2
[2022-10-28T17:35:47] 46 [A4:C1:38:EC:0B:03] (bdaddr_type) 00 (evt_type) 00 (Name) GVH5075_0B03 (UUID) 88EC (Flags) 05 (Manu) 88EC0002F4146300 (Temp) 19.3556°C (Humidity) 55.6% (Battery) 99% (GVH5075)
[2022-10-28T17:35:47] 42 [A4:C1:38:EC:0B:03] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FFC2 (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) C2
[2022-10-28T17:35:49] 46 [A4:C1:38:EC:0B:03] (bdaddr_type) 00 (evt_type) 00 (Name) GVH5075_0B03 (UUID) 88EC (Flags) 05 (Manu) 88EC0002F4146300 (Temp) 19.3556°C (Humidity) 55.6% (Battery) 99% (GVH5075)
^C***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
GoveeBTTempLogger Version 2.20221007-5 Built on: Oct 28 2022 at 10:31:54 (exiting)
```

mc-fu, 2023-07-29:
@wcbonner Not sure if it helps, but as I have a H5179 lying around, I can provide you with the output of the device:

```
root@lisa:/tmp# /usr/local/bin/goveebttemplogger -v 3 -o 18:1F:68:D3:00:B7
[2023-07-29T08:54:32] GoveeBTTempLogger Version 2.20230320-1 Built on: Mar 20 2023 at 11:12:52
[                   ]      log:
[                   ]      svg:
[                   ]  battery: 0
[                   ]   minmax: 0
[                   ]  celsius: false
[                   ] titlemap:
[                   ]     time: 60
[                   ]  average: 5
[                   ] download: 0
[2023-07-29T08:54:32] LocalName: lisa
[2023-07-29T08:54:32] BlueTooth Address Filter: [18:1F:68:D3:00:B7]
[2023-07-29T08:54:32] Scanning Stopped.
[2023-07-29T08:54:32] Scanning Started. ScanInterval(40 msec) ScanWindow(30 msec) ScanType(1)
[2023-07-29T08:54:33] 44 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5179_00B7
[2023-07-29T08:54:33] 28 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 04 (Manu) 0188EC000101EC09621664
[2023-07-29T08:54:34] 44 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5179_00B7
[2023-07-29T08:54:34] 28 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 04 (Manu) 0188EC000101EC09581664
[2023-07-29T08:54:35] 44 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5179_00B7
[2023-07-29T08:54:35] 28 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 04 (Manu) 0188EC000101EC09581664
[2023-07-29T08:54:36] 44 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5179_00B7
[2023-07-29T08:54:36] 28 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 04 (Manu) 0188EC000101EC09581664
[2023-07-29T08:54:37] 44 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5179_00B7
[2023-07-29T08:54:37] 28 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 04 (Manu) 0188EC000101EC09581664
[2023-07-29T08:54:38] 44 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5179_00B7
[2023-07-29T08:54:38] 28 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 04 (Manu) 0188EC000101EC09581664
[2023-07-29T08:54:39] 44 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5179_00B7
[2023-07-29T08:54:39] 28 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 04 (Manu) 0188EC000101EC09581664
^C***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
[2023-07-29T08:54:39] Scanning Stopped.
GoveeBTTempLogger Version 2.20230320-1 Built on: Mar 20 2023 at 11:12:52 (exiting)
```

Let me know if I should test something else. Would be happy to see support for the H5179.

wcbonner, 2023-07-29:
> @wcbonner Not sure if it helps, but as I have a H5179 lying around, I can provide you with the output of the device:
>
> ```
> root@lisa:/tmp# /usr/local/bin/goveebttemplogger -v 3 -o 18:1F:68:D3:00:B7 [...]

@mc-fu That same log with details of what the thermometer was reading in °C and % relative humidity would be useful. I might be able to figure it out without the confirmation of the actual temperature, but having the details would be good.

mc-fu, 2023-08-01:
@wcbonner:

```
[2023-08-01T18:59:16] 44 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5179_00B7
[2023-08-01T18:59:16] 28 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 04 (Manu) 0188EC0001012008121B64
[2023-08-01T18:59:17] 44 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5179_00B7
[2023-08-01T18:59:17] 28 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 04 (Manu) 0188EC0001012008121B64
[2023-08-01T18:59:18] 44 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5179_00B7
[2023-08-01T18:59:18] 28 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 04 (Manu) 0188EC0001012008121B64
```

should be something like 20.8C 69.3%RH

wcbonner, 2023-08-01:
@mc-fu I believe I just added support for the H5179. I'd appreciate if you can pull the current code, run the same command as before and post the results.

mc-fu, 2023-08-02:
@wcbonner

looks good:

```
[2023-08-02T16:25:14] 44 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5179_00B7
[2023-08-02T16:25:14] 28 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 04 (Manu) 0188EC0001013C0A1E1964 (Temp) 26.2°C (Humidity) 65% (Battery) 100% (GVH5179)
```

app says 26,2C and 64,3%RH - so this matches. thanks for the work!

wcbonner, 2023-08-02:
@mc-fu I made one more change related to the humidity on the 5179. I was mistakenly using the battery value as the least significant digits. If you can send me a log of the code I just pushed I'd appreciate it. (I would like to have it in the code comment)

mc-fu, 2023-08-02:
@wcbonner 

[2023-08-02T16:50:40] 44 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5179_00B7
[2023-08-02T16:50:40] 28 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 04 (Manu) 0188EC0001010A0AB01864 (Temp) 25.7°C (Humidity) 63.2% (Battery) 100% (GVH5179)

Perfekt, now it exactly matches the value within the App. Thanks!

wcbonner, 2024-08-14:
> @wcbonner
>
>
>
> [2023-08-02T16:50:40] 44 [18:1F:68:D3:00:B7] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5179_00B7
>
> [2023-08-02T16:50:40] 28 [18:1F:68:D3:00:B7] [...]

@mc-fu I don't know if you are still using this program or still have a Govee 5179 thermometer. If you still have time, I'd appreciate a test of my BlueZ-DBus code branch to see if it decodes the 5179 thermometer data. (It's one thermometer that I've never purchased) 

Thanks, Wim