wcbonner/GoveeBTTempLogger #38: Abort on Default SystemD Unit on Ubuntu 20.04 and Raspberry Pi OS (Latest)

sl33nyc, 2022-10-07:
```
@raspberrypi:~/GoveeBTTempLogger/GoveeBTTempLogger/usr/local/bin $ sudo ./goveebttemplogger -v 1 --log /var/log/goveebttemplogger --time 60 --svg /var/www/html/goveebttemplogger/ --battery 8 --minmax 8 --titlemap /var/www/html/goveebttemplogger/gvh-titlemap.txt
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::erase: __pos (which is 47) > this->size() (which is 32)
Aborted
```
It seems like the trailing `/` on the `--svg` directory is what causes the crash.  I'm documenting here and will try putting my C++ hat on to submit a follow up pull request.  (It's been awhile since I've written C++, so bear with me... :grin:)

sl33nyc, 2022-10-07:
> Is that the only output you got?

Yes

> If you removed the trailing / on the --svg option it doesn't crash the same way?

Yes

As a side-effect of removing the `/`, it attempts to output SVG's to the path, `/var/www/html/goveebttemploggergvh-XXXXXXXX.svg`.

wcbonner, 2022-10-07:
@sl33nyc Thanks for pointing this out. I think I introduced the problem a couple of months ago and am not positive how my system has been correctly working. I just pushed new code that should fix it. 

I also included a validated parameter dump if you use --verbose 3 or higher. It was especially useful because the machine I was testing didn't have those specific paths, so my validation routine didn't use either one initially.

If this now works properly, you can close the issue. ;-)

sl33nyc, 2022-10-12:
I think it works great now!  Thanks for the quick fix!

closed 2022-10-12 by sl33nyc