wcbonner/GoveeBTTempLogger #39: No rule to make target 'deb'

billjuv, 2022-10-23:
I'm unable to complete the install of goveebttemplogger.service as you can see below. 
On my Raspberry Pi 4 Model B Rev 1.5  
I'm running:
**Linux HomePi 5.15.61-v7l+ #1579 SMP Fri Aug 26 11:13:03 BST 2022 armv7l GNU/Linux**

pi@HomePi:~ $ sudo apt-get install bluetooth bluez libbluetooth-dev
make deb
sudo make install-deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
bluetooth is already the newest version (5.55-3.1+rpt2).
bluez is already the newest version (5.55-3.1+rpt2).
libbluetooth-dev is already the newest version (5.55-3.1+rpt2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
make: *** No rule to make target 'deb'.  Stop.
make: *** No rule to make target 'install-deb'.  Stop.

Not sure where to go from here... Thanks

wcbonner, 2022-10-24:
did you do this?
```
git https://github.com/wcbonner/GoveeBTTempLogger
cd GoveeBTTempLogger
make deb
```

billjuv, 2022-10-24:
No, that did it. New concept till last night; it never dawned on me that I could/should clone a repository to my Pi. I'd only used GitHub from my computer while programming microcontrollers.

wcbonner, 2022-10-24:
Glad it was an easy fix. 😉

closed 2022-10-24 by wcbonner