wcbonner/GoveeBTTempLogger #40: H5181 support? (like the H5182 but only one probe)

wcbonner, 2022-10-28:
I have a H5181 (like the H5182 but only one probe). Can I provide raw data to help decipher the information for that model?

_Originally posted by @thisgarryhas2rs in https://github.com/wcbonner/GoveeBTTempLogger/issues/36#issuecomment-1294399973_

wcbonner, 2022-10-28:
@thisgarryhas2rs If you can run the program and capture some data I might be able to recognize the difference in format between the two devices.

I just ran the program on my local machine looking for a particular device I knew was reporting. If you run a similar command on your local machine and substitute the Bluetooth address of your device, it would help.

```
wim@WimPi4-Dev:~/GoveeBTTempLogger $ sudo /home/visualstudio/projects/GoveeBTTempLogger/bin/ARM/Debug/GoveeBTTempLogger.out  -v 3 -o A4:C1:38:EC:0B:03
[2022-10-28T17:35:33] GoveeBTTempLogger Version 2.20221007-5 Built on: Oct 28 2022 at 10:31:54
[                   ]      log: 
[                   ]      svg: 
[                   ]  battery: 0
[                   ]   minmax: 0
[                   ]  celsius: false
[                   ] titlemap: 
[                   ]     time: 60
[                   ]  average: 5
[                   ] download: false
[2022-10-28T17:35:33] LocalName: WimPi4-Dev
[2022-10-28T17:35:33] BlueTooth Address Filter: [A4:C1:38:EC:0B:03]
[2022-10-28T17:35:33] Scanning...
[2022-10-28T17:35:43] 46 [A4:C1:38:EC:0B:03] (bdaddr_type) 00 (evt_type) 00 (Name) GVH5075_0B03 (UUID) 88EC (Flags) 05 (Manu) 88EC0002F4146300 (Temp) 19.3556°C (Humidity) 55.6% (Battery) 99% (GVH5075)
[2022-10-28T17:35:45] 46 [A4:C1:38:EC:0B:03] (bdaddr_type) 00 (evt_type) 00 (Name) GVH5075_0B03 (UUID) 88EC (Flags) 05 (Manu) 88EC0002F4146300 (Temp) 19.3556°C (Humidity) 55.6% (Battery) 99% (GVH5075)
[2022-10-28T17:35:45] 42 [A4:C1:38:EC:0B:03] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FFC2 (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) C2
[2022-10-28T17:35:47] 46 [A4:C1:38:EC:0B:03] (bdaddr_type) 00 (evt_type) 00 (Name) GVH5075_0B03 (UUID) 88EC (Flags) 05 (Manu) 88EC0002F4146300 (Temp) 19.3556°C (Humidity) 55.6% (Battery) 99% (GVH5075)
[2022-10-28T17:35:47] 42 [A4:C1:38:EC:0B:03] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FFC2 (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) C2
[2022-10-28T17:35:49] 46 [A4:C1:38:EC:0B:03] (bdaddr_type) 00 (evt_type) 00 (Name) GVH5075_0B03 (UUID) 88EC (Flags) 05 (Manu) 88EC0002F4146300 (Temp) 19.3556°C (Humidity) 55.6% (Battery) 99% (GVH5075)
^C***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
GoveeBTTempLogger Version 2.20221007-5 Built on: Oct 28 2022 at 10:31:54 (exiting)
```

thisgarryhas2rs, 2022-10-29:
```
server:~/git/GoveeBTTempLogger/GoveeBTTempLogger/usr/local/bin$ sudo ./goveebttemplogger -v 3 -o a4:c1:38:5c:c9:36
[2022-10-29T16:34:37] GoveeBTTempLogger Version 2.20221007-5 Built on: Oct 27 2022 at 20:09:55
[                   ]      log: 
[                   ]      svg: 
[                   ]  battery: 0
[                   ]   minmax: 0
[                   ]  celsius: false
[                   ] titlemap: 
[                   ]     time: 60
[                   ]  average: 5
[                   ] download: false
[2022-10-29T16:34:37] LocalName: server
[2022-10-29T16:34:37] BlueTooth Address Filter: [A4:C1:38:5C:C9:36]
[2022-10-29T16:34:37] Scanning...
[2022-10-29T16:34:37] 46 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 00 (UUID) 5181 (Flags) 05 (Manu) 5CC93601000101E4018F0898FFFF0000 (Temp) 22°C (Temp) -0.01°C (Battery) 100% (GVH5183) (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: CB) 
[2022-10-29T16:34:37] 42 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
[2022-10-29T16:34:38] shuffling week 2022-10-29 09:30:00
[2022-10-29T16:34:38] 46 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 00 (UUID) 5181 (Flags) 05 (Manu) 5CC93601000101E4008F0898FFFF0000 (Temp) 22°C (Temp) -0.01°C (Battery) 100% (GVH5183) (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: D1) 
[2022-10-29T16:34:38] 42 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
[2022-10-29T16:34:39] 46 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 00 (UUID) 5181 (Flags) 05 (Manu) 5CC93601000101E4018F0898FFFF0000 (Temp) 22°C (Temp) -0.01°C (Battery) 100% (GVH5183) (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: D1) 
[2022-10-29T16:34:39] 42 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
[2022-10-29T16:34:40] 46 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 00 (UUID) 5181 (Flags) 05 (Manu) 5CC93601000101E4008F0898FFFF0000 (Temp) 22°C (Temp) -0.01°C (Battery) 100% (GVH5183) (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: D0) 
[2022-10-29T16:34:40] 42 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
[2022-10-29T16:34:42] 46 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 00 (UUID) 5181 (Flags) 05 (Manu) 5CC93601000101E4008F0898FFFF0000 (Temp) 22°C (Temp) -0.01°C (Battery) 100% (GVH5183) (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: D1) 
[2022-10-29T16:34:42] 42 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
[2022-10-29T16:34:43] 46 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 00 (UUID) 5181 (Flags) 05 (Manu) 5CC93601000101E4018F0898FFFF0000 (Temp) 22°C (Temp) -0.01°C (Battery) 100% (GVH5183) (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: CD) 
[2022-10-29T16:34:43] 42 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
[2022-10-29T16:34:44] 46 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 00 (UUID) 5181 (Flags) 05 (Manu) 5CC93601000101E4008F0898FFFF0000 (Temp) 22°C (Temp) -0.01°C (Battery) 100% (GVH5183) (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: CC) 
[2022-10-29T16:34:44] 42 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
[2022-10-29T16:34:47] 46 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 00 (UUID) 5181 (Flags) 05 (Manu) 5CC93601000101E4018F0898FFFF0000 (Temp) 22°C (Temp) -0.01°C (Battery) 100% (GVH5183) (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: CC) 
[2022-10-29T16:34:47] 42 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
```

wcbonner, 2022-10-29:
@thisgarryhas2rs I believe that my latest code in the repository should now recognize the H5181 and label it as such in the output log files. If you can update and let me know, I'd appreciate it. (GoveeBTTempLogger Version 2.20221029-1)

thisgarryhas2rs, 2022-10-29:
Awesome. Thanks!

wcbonner, 2022-10-29:
> Awesome. Thanks!

@thisgarryhas2rs If you could do both a quick run with the` -v 3 -o a4:c1:38:5c:c9:36` options, as well as a logfile output I'd appreciate confirmation that all looks good to me. Thanks.

thisgarryhas2rs, 2022-10-30:
```
[2022-10-30T03:23:50] GoveeBTTempLogger Version 2.20221029-1 Built on: Oct 29 2022 at 15:41:55
[                   ]      log:
[                   ]      svg:
[                   ]  battery: 0
[                   ]   minmax: 0
[                   ]  celsius: false
[                   ] titlemap:
[                   ]     time: 60
[                   ]  average: 5
[                   ] download: false
[2022-10-30T03:23:50] LocalName: server
[2022-10-30T03:23:50] BlueTooth Address Filter: [A4:C1:38:5C:C9:36]
[2022-10-30T03:23:50] Scanning...
[2022-10-30T03:23:51] 46 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 00 (UUID) 5181 (Flags) 05 (Manu) 5CC93601000101E400860A281CD40000 (Temp) 26°C (Temp) 73.8°C (Battery) 100% (GVH5183) (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: CE)
[2022-10-30T03:23:51] 42 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
[2022-10-30T03:23:52] 46 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 00 (UUID) 5181 (Flags) 05 (Manu) 5CC93601000101E401860A281CD40000 (Temp) 26°C (Temp) 73.8°C (Battery) 100% (GVH5183) (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: C9)
[2022-10-30T03:23:52] 42 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
[2022-10-30T03:23:54] 46 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 00 (UUID) 5181 (Flags) 05 (Manu) 5CC93601000101E401860A281CD40000 (Temp) 26°C (Temp) 73.8°C (Battery) 100% (GVH5183) (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: CC)
[2022-10-30T03:23:54] 42 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
```

gvh507x_A4C1385CC936-2022-10.txt
```
2022-10-30 03:26:32     26      0       100     5183    73.8
2022-10-30 03:26:33     26      0       100     5183    73.8
2022-10-30 03:26:34     26      0       100     5183    73.8
2022-10-30 03:26:38     26      0       100     5183    73.8
2022-10-30 03:26:39     26      0       100     5183    73.8
2022-10-30 03:26:41     26      0       100     5183    73.8
2022-10-30 03:26:45     26      0       100     5183    73.8
2022-10-30 03:26:46     26      0       100     5183    73.8
2022-10-30 03:26:47     26      0       100     5183    73.8
2022-10-30 03:26:48     26      0       100     5183    73.8
2022-10-30 03:26:50     26      0       100     5183    73.8
2022-10-30 03:26:51     26      0       100     5183    73.8
2022-10-30 03:26:52     26      0       100     5183    73.8
2022-10-30 03:26:53     26      0       100     5183    73.8
2022-10-30 03:26:54     26      0       100     5183    73.8
2022-10-30 03:26:57     26      0       100     5183    73.8
2022-10-30 03:26:58     26      0       100     5183    73.8
2022-10-30 03:26:59     26      0       100     5183    73.8
2022-10-30 03:27:00     26      0       100     5183    73.8
2022-10-30 03:27:01     26      0       100     5183    73.8
2022-10-30 03:27:02     26      0       100     5183    73.8
```

thisgarryhas2rs, 2022-10-30:
`GoveeBTTempLogger Version 2.20221029-1 Built on: Oct 29 2022 at 15:41:55`

wcbonner, 2022-10-30:
> `GoveeBTTempLogger Version 2.20221029-1 Built on: Oct 29 2022 at 15:41:55`

@thisgarryhas2rs thanks for running that. 

I could have sworn that it should recognize the UUID and set the thermometer type correctly, but it's obviously still printing the wrong type and logging the wrong number of temperatures per device. 

I probably won't have time to look at this again before Monday.

wcbonner, 2022-10-31:
> @thisgarryhas2rs thanks for running that.
>
> I could have sworn that it should recognize the UUID and set the thermometer type correctly, but it's obviously still printing the wrong type and logging [...]

@thisgarryhas2rs I think I just fixed the recognition of the H5181. My brain was not in gear when I was looking at the output yesterday, and I think I was doing everything correct except for overwriting the model type. I think the number in the last column of output after the 5183 is the temperature that you've set your alarm to. If you can verify it, I'd appreciate it.

Wim.

thisgarryhas2rs, 2022-11-01:
Looks like the data being sent has both 5181 `(UUID) 5181` and 5183 `GVH5183` names in the packet (last line). I checked the physical cover and it does say `Mode: H5181`. 

```
sudo /usr/local/bin/goveebttemplogger -o A4:C1:38:5C:C9:36 -v 3

[2022-11-01T04:08:59] GoveeBTTempLogger Version 2.20221030-1 Built on: Oct 31 2022 at 20:52:08
[2022-11-01T04:08:59] LocalName: server
[2022-11-01T04:08:59] BlueTooth Address Filter: [A4:C1:38:5C:C9:36]
[2022-11-01T04:08:59] Scanning...
[2022-11-01T04:09:00] 46 [A4:C1:38:5C:C9:36] (Temp) 26°C (Temp) 73.8°C (Battery) 100% (GVH5183) (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: C7)
[2022-11-01T04:09:00] 42 [A4:C1:38:5C:C9:36] (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
[2022-11-01T04:09:01] 46 [A4:C1:38:5C:C9:36] (UUID) 5181 (Flags) 05 (Manu) 5CC93601000101E401860A281CD40000 (Temp) 26°C (Temp) 73.8°C (Battery) 100% (GVH5183) (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: C7)
```

```
gvh507x_A4C1385CC936-2022-11.txt

2022-11-01 04:05:07     36      0       100     5183    73.8
2022-11-01 04:05:08     36      0       100     5183    73.8
2022-11-01 04:05:09     36      0       100     5183    73.8
2022-11-01 04:05:11     36      0       100     5183    73.8
```

wcbonner, 2022-11-01:
> Looks like the data being sent has both 5181 `(UUID) 5181` and 5183 `GVH5183` names in the packet (last line). I checked the physical cover and it does say `Mode: H5181`.
>
> ```
> sudo [...]

I'll have to look into this again tomorrow.  Since the UUID is broadcast in the same message as the temperature data in this device and broadcast first it should be caught and correctly identified in line 2136, Then in line 2191 it decodes the manufacturer specific data that has the temperatures. I'm obviously net recognizing the UUID and the type is still unknown when it gets decoded, so I default set it to type 5183.

Do you have your alarm set to 73.8 degrees C or 165 degrees F?

thisgarryhas2rs, 2022-11-01:
Alarm is 165F. Temps should be in F but they seem to be in Celsius from the logs.

wcbonner, 2022-11-02:
> Alarm is 165F. Temps should be in F but they seem to be in Celsius from the logs.

The devices communicate in Celsius. I do all the logging in Celsius to keep things consistent, I have code that converts to Fahrenheit in my SVG creation routines. I figure that's pretty simple on any platform and being able to compare the raw data with the logged data is a more useful detail when debugging. 

@thisgarryhas2rs : I think I finally have the recognition of the 5181 working correctly by its UUID. I had a big/little endian problem and fixed it by simply hard coding the numbers in the reverse byte order.

thisgarryhas2rs, 2022-11-03:
Looks good. -0.01 is when the probe was not plugged in.

```
2022-11-03 01:06:40     -0.01   0       100     5181    73.8
2022-11-03 01:06:41     -0.01   0       100     5181    73.8
2022-11-03 01:06:42     -0.01   0       100     5181    73.8
2022-11-03 01:06:44     26      0       100     5181    73.8
2022-11-03 01:06:45     26      0       100     5181    73.8
2022-11-03 01:06:46     26      0       100     5181    73.8
```

```
server:~$ sudo goveebttemplogger -o A4:C1:38:5C:C9:36 -v 3
[2022-11-03T01:08:38] GoveeBTTempLogger Version 2.20221101-1 Built on: Nov  2 2022 at 17:56:28
[                   ]      log:
[                   ]      svg:
[                   ]  battery: 0
[                   ]   minmax: 0
[                   ]  celsius: false
[                   ] titlemap:
[                   ]     time: 60
[                   ]  average: 5
[                   ] download: false
[2022-11-03T01:08:38] LocalName: server
[2022-11-03T01:08:38] BlueTooth Address Filter: [A4:C1:38:5C:C9:36]
[2022-11-03T01:08:38] Scanning...
[2022-11-03T01:08:38] 46 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 00 (UUID) 5181 (Flags) 05 (Manu) 5CC93601000101E4018609C41CD40000 (Temp) 25°C (Temp) 73.8°C (Battery) 100% (GVH5181) (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: 00)  (Other: D1)
[2022-11-03T01:08:38] 42 [A4:C1:38:5C:C9:36] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
```

closed 2022-11-12 by wcbonner