wcbonner/GoveeBTTempLogger #41: Single output option flag

thisgarryhas2rs, 2022-10-29:
Background: I'm trying to control my electric smoker with a smart plug depending on the temperature of the meat probe. Getting a single reading output would be handy for this.

Question: Is there a way to program an option flag to just output the current temperature? Or maybe an array of temperatures if there are multiple probes?  

Thoughts: `-r (--readings)`  Outputs a json key-value pairs for each reading

wcbonner, 2022-10-29:
I don't really see this as easy to implement because of the uncertain nature of waiting for Bluetooth announcements. 

My original development version of the program ran with two methods, one listened for Bluetooth announcements and logged them, the second read the log files to be able to respond with recent data for MRTG input purposes. 

I've moved away from using MRTG to poll the program, choosing instead to write similar sized SVG graphic files. 

I've tried to keep the logging file as little changed as possible to make them easy to parse by other programs. Monitoring log files and retrieving the last line periodically might be the easiest way to achieve what you are looking for in a polling fashion.

thisgarryhas2rs, 2022-10-29:
Understood.

closed 2022-10-29 by thisgarryhas2rs