wcbonner/GoveeBTTempLogger #42: Question: goveebttemplogger.service doesn't seem to accept '-c' and '--celsius'

Orsim27, 2022-12-13:
Hey there!

I ran into the following issue today while setting everything up on my Raspberry Zero. The .svg files that are created automatically in /var/www/html don't seem to follow the '-c' argument if allpied in goveebttemplogger.service.

I tried the following:
1. "sudo systemctl edit goveebttemplogger.service" with both '--celsius' and '-c', I also tried the EXTRAARGS:
```
Environment="SVGARGS=--svg /var/www/html/goveebttemplogger/ --celsius --battery 8 --minmax 8 --titlemap /var/www/html/goveebttemplogger/gvh-titlemap.txt"
ExecStart=/usr/local/bin/goveebttemplogger  -v ${VERBOSITY} --log ${LOGDIR} --time ${TIME} $SVGARGS $EXTRAARGS
```
3. `sudo systemctl restart goveebttemplogger`

But the .svg files in /var/www/html/goveebttemplogger/  are still showing Fahrenheit instead of Celsius. Am I doing something wrong or can't I edit that part of the service? I can get Celsius .svg when manually starting /usr/local/bin/goveebttemplogger  after all and the logs are in Celsius as well.

Thanks in advance for your help!

Orsim27, 2022-12-13:
Hey, thanks for your quick response and help! It works with `sudo systemctl edit --full goveebttemplogger.service` but not with out the `--full` eventhough that should also work? 

The file override.conf under /etc/systemd/system/goveebttemplogger.service.d also contains my `--celsius`, it just doesn't seem to actually override anything

closed 2022-12-13 by Orsim27

wcbonner, 2022-12-13:
> Hey, thanks for your quick response and help! It works with `sudo systemctl edit --full goveebttemplogger.service` but not with out the `--full` eventhough that should also work?
>
> The file [...]

The systemctl edit features are a bit of a mystery to me. I admit to being a bit of a script kiddie when it comes to using what works that I've seen elsewhere and not really understanding the details as far as systemctl and dpkg commands. it's possible the --full difference has somethign to do with the fact that the service file itself lives in /usr/local/lib/systemd/system/goveebttemplogger.service and systemctl puts a link in the appropriate service location to start it.