wcbonner/GoveeBTTempLogger #43: Failing on ubuntu 22.04 (linux mint 21) - LE Set Scan Parameters  Command Disallowed

yduf, 2023-02-04:
Hello,

I am facing trouble trying to make goveebttemplogger work, may be because of latest change in bluetooth stack.
The issue happens on first call of [`hci_le_set_scan_parameters(device_handle, 0x01, htobs(0x0012), htobs(0x0012), 0x01, 0x00, 1000)`](https://github.com/wcbonner/GoveeBTTempLogger/blob/master/goveebttemplogger.cpp#L2072)
and it is failing with error :` Error: Failed to set scan parameters: Input/output error` due to the action being disallowed (see btmon trace below).

On the other side I am able to complete a manual le scan with `bluetoothctl` on the very same host and see my Govee devices.

So to me it looks very much like this [issue where bluetoothctl works but not hcitools](https://unix.stackexchange.com/a/96639/192991).

It seems people recommend to do a hci_reset before see [1](https://community.infineon.com/t5/Smart-Bluetooth/CYBT-413034-02-quot-LE-Set-Scan-Parameters-Command-quot-returns-quot-COMMAND/td-p/236681) & [2](https://github.com/noble/noble/issues/864).
Yet I don't understand how the code should be change to have a succesfull bluetooth initialisation - I tried my own code, but it fail the same.  I am completely new to bluetooth stack, so can help much more.

Regards,

Yves

PS:
btmon output
```
@ RAW Open: goveebttemplogg (privileged) version 2.22                                          {0x0002} 1140.953661
@ RAW Close: goveebttemplogg                                                                   {0x0002} 1140.953673
@ RAW Open: goveebttemplogg (privileged) version 2.22                                   {0x0002} [hci0] 1140.953687
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                       #15 [hci0] 1140.953714
        Scanning: Disabled (0x00)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                                  #16 [hci0] 1141.077301
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Command Disallowed (0x0c)
< HCI Command: Read Local Name (0x03|0x0014) plen 0                                          #17 [hci0] 1141.077566
> HCI Event: Command Complete (0x0e) plen 252                                                #18 [hci0] 1141.082287
      Read Local Name (0x03|0x0014) ncmd 1
        Status: Success (0x00)
        Name: yves-huv
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                                   #19 [hci0] 1141.082712
        Type: Active (0x01)
        Interval: 11.250 msec (0x0012)
        Window: 11.250 msec (0x0012)
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                  #20 [hci0] 1141.084237
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Command Disallowed (0x0c)
@ RAW Close: goveebttemplogg 
```

wcbonner, 2023-02-04:
I'll have to look into this more. I've not been using btmon, but instead use Wireshark for most of my Bluetooth debugging. It looks like we are using the same version of Bluetooth. Comparing what's happening on your machine to mine might be easier if the sequence numbers matched up. You can see that I start receiving advertisements by sequence 11. 

Were there a set of hci commands 1-14 in your log?

Is GoveeBTTempLogger exiting on its own after sequence 20 on your output?

I can't remember why I first set the scanning parameters to 11.25msec and then issue a second scan parameters command to 500msec. It was probably something I saw happening in my pcap log from the android device. 

I'm running the program in a different window with the command: `sudo ../visualstudio/projects/GoveeBTTempLogger/bin/ARM/Debug/GoveeBTTempLogger.out`

```
wim@WimPi4-Dev:~ $ uname -a
Linux WimPi4-Dev 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
wim@WimPi4-Dev:~ $ sudo btmon
Bluetooth monitor ver 5.55
= Note: Linux version 5.15.84-v8+ (aarch64)                                                0.813857
= Note: Bluetooth subsystem version 2.22                                                   0.813871
= New Index: DC:A6:32:B9:23:06 (Primary,UART,hci0)                                  [hci0] 0.813877
= Open Index: DC:A6:32:B9:23:06                                                     [hci0] 0.813881
= Index Info: DC:A6:32:B9:23:06 (Cypress Semiconductor)                             [hci0] 0.813890
@ MGMT Open: bluetoothd (privileged) version 1.21                                 {0x0001} 0.813909
@ RAW Open: GoveeBTTempLogg (privileged) version 2.22                             {0x0002} 6.262036
@ RAW Close: GoveeBTTempLogg                                                      {0x0002} 6.262078
@ RAW Open: GoveeBTTempLogg (privileged) version 2.22                      {0x0002} [hci0] 6.262124
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                           #1 [hci0] 6.262228
        Scanning: Disabled (0x00)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                      #2 [hci0] 6.262570
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Command Disallowed (0x0c)
< HCI Command: Read Local Name (0x03|0x0014) plen 0                              #3 [hci0] 6.262645
> HCI Event: Command Complete (0x0e) plen 252                                    #4 [hci0] 6.263871
      Read Local Name (0x03|0x0014) ncmd 1
        Status: Success (0x00)
        Name: WimPi4-Dev
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                       #5 [hci0] 6.264075
        Type: Active (0x01)
        Interval: 11.250 msec (0x0012)
        Window: 11.250 msec (0x0012)
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                      #6 [hci0] 6.264386
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                       #7 [hci0] 6.264487
        Type: Active (0x01)
        Interval: 5000.000 msec (0x1f40)
        Window: 5000.000 msec (0x1f40)
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                      #8 [hci0] 6.264767
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                           #9 [hci0] 6.264859
        Scanning: Enabled (0x01)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                     #10 [hci0] 6.265226
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 43                                       #11 [hci0] 6.292477
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Non connectable undirected - ADV_NONCONN_IND (0x03)
        Address type: Random (0x01)
        Address: 1E:AC:39:2F:70:8A (Non-Resolvable)
        Data length: 31
        Company: Microsoft (6)
          Data: 01092022034c28c4c5089b92b5eb6074fa7eca293700d1fdf943ce
        RSSI: -72 dBm (0xb8)
> HCI Event: LE Meta Event (0x3e) plen 43                                       #12 [hci0] 6.304756
[... 43 lines cut ...]
        Address: A4:C1:38:13:AE:36 (Telink Semiconductor (Taipei) Co. Ltd.)
        Data length: 31
        Name (complete): GVH5075_AE36
        16-bit Service UUIDs (complete): 1 entry
          Unknown (0xec88)
        Flags: 0x05
          LE Limited Discoverable Mode
          BR/EDR Not Supported
        Company: not assigned (60552)
          Data: 0002d08e2600
        RSSI: -55 dBm (0xc9)
> HCI Event: LE Meta Event (0x3e) plen 39                                      #20 [hci0] 13.731291
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Scan response - SCAN_RSP (0x04)
        Address type: Public (0x00)
        Address: A4:C1:38:13:AE:36 (Telink Semiconductor (Taipei) Co. Ltd.)
        Data length: 27
        Company: Apple, Inc. (76)
          Type: iBeacon (2)
          UUID: 57485f53-4b43-4f52-5f49-4c4c45544e49
          Version: 30032.65522
          TX power: -62 dB
        RSSI: -55 dBm (0xc9)
```

yduf, 2023-02-04:
Sorry, I didn't think of restarting btmon between my attemps.

I updated my first message with a new trace starting at #1:
GoveeBTTempLogger exit on its own, when it encounter the first error at #5, during initialization (at [goveebttemplogger.cpp#L2072](https://github.com/wcbonner/GoveeBTTempLogger/blob/master/goveebttemplogger.cpp#L2072) in the code. I get that from the debugger.

Nothing was retrieved from the bluetooth network at this point, since it is way before the `while (bRun) {...}` loop.

wcbonner, 2023-02-04:
Ok. And you are definitely running GoveeBTTempLogger with full root privileges? Just checking more possibilities. 

```
wim@WimPi4-Dev:~ $ ../visualstudio/projects/GoveeBTTempLogger/bin/ARM/Debug/GoveeBTTempLogger.out
[2023-02-04T18:20:05] GoveeBTTempLogger Version 2.20221220-1 Built on: Feb  4 2023 at 04:47:19
[2023-02-04T18:20:05] LocalName: WimPi4-Dev
[2023-02-04T18:20:05] No BlueTooth Address Filter: [00:00:00:00:00:00]
[                   ] Error: Failed to set scan parameters: Operation not permitted
GoveeBTTempLogger Version 2.20221220-1 Built on: Feb  4 2023 at 04:47:19 (exiting)
```
GoveeBTTempLogger  without the sudo produces:
```
wim@WimPi4-Dev:~ $ sudo btmon
Bluetooth monitor ver 5.55
= Note: Linux version 5.15.84-v8+ (aarch64)                                                0.471531
= Note: Bluetooth subsystem version 2.22                                                   0.471545
= New Index: DC:A6:32:B9:23:06 (Primary,UART,hci0)                                  [hci0] 0.471551
= Open Index: DC:A6:32:B9:23:06                                                     [hci0] 0.471554
= Index Info: DC:A6:32:B9:23:06 (Cypress Semiconductor)                             [hci0] 0.471560
@ MGMT Open: bluetoothd (privileged) version 1.21                                 {0x0001} 0.471566
@ RAW Open: GoveeBTTempLogg version 2.22                                         {0x0002} 15.474829
@ RAW Close: GoveeBTTempLogg                                                     {0x0002} 15.474858
@ RAW Open: GoveeBTTempLogg version 2.22                                  {0x0002} [hci0] 15.474902
< HCI Command: Read Local Name (0x03|0x0014) plen 0                             #1 [hci0] 15.474976
> HCI Event: Command Complete (0x0e) plen 252                                   #2 [hci0] 15.476220
      Read Local Name (0x03|0x0014) ncmd 1
        Status: Success (0x00)
        Name: WimPi4-Dev
@ RAW Close: GoveeBTTempLogg                                              {0x0002} [hci0] 15.476790
```

I've been digging through the library headers and there's not a simple reset command I can call for the HCI, so it's going to take longer to figure out how to add that command in the code.

[hci_send_cmd](http://gbrault.github.io/gattclient/hci_8c.html#a30a626b20fb93c9463b73e29bd2e8881) is promising, but I don't really have experience sending direct hardware control interface commands.
HCIDEVRESET is defined [here](http://gbrault.github.io/gattclient/hci_8h.html#a0e0aab9fb1ba78a570592fb9952e7965), but I'm not sure how to use it.

wcbonner, 2023-02-04:
@yduf If you run bluetoothctl does it go into scanning mode properly? 
```
wim@WimPi4-Dev:~ $ sudo bluetoothctl
Agent registered
[bluetooth]# agent on
Agent is already registered
[bluetooth]# default-agent
Default agent request successful
[bluetooth]# scan on
Discovery started
[CHG] Controller DC:A6:32:B9:23:06 Discovering: yes
[NEW] Device A4:C1:38:EC:0B:03 GVH5075_0B03
[bluetooth]# scan off
Discovery stopped
[CHG] Device A4:C1:38:EC:0B:03 RSSI is nil
[CHG] Controller DC:A6:32:B9:23:06 Discovering: no
[bluetooth]# agent off
Agent unregistered
[bluetooth]# exit
```
produces this output:
```
wim@WimPi4-Dev:~ $ sudo btmon
Bluetooth monitor ver 5.55
= Note: Linux version 5.15.84-v8+ (aarch64)                                                0.332971
= Note: Bluetooth subsystem version 2.22                                                   0.332983
= New Index: DC:A6:32:B9:23:06 (Primary,UART,hci0)                                  [hci0] 0.332987
= Open Index: DC:A6:32:B9:23:06                                                     [hci0] 0.332990
= Index Info: DC:A6:32:B9:23:06 (Cypress Semiconductor)                             [hci0] 0.332992
@ MGMT Open: bluetoothd (privileged) version 1.21                                 {0x0001} 0.332996
@ MGMT Command: Set IO Capability (0x0018) plen 1                         {0x0001} [hci0] 57.478932
        Capability: KeyboardDisplay (0x04)
@ MGMT Event: Command Complete (0x0001) plen 3                            {0x0001} [hci0] 57.478959
      Set IO Capability (0x0018) plen 0
        Status: Success (0x00)
@ MGMT Command: Start Discovery (0x0023) plen 1                           {0x0001} [hci0] 67.990918
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
< HCI Command: LE Set Random Address (0x08|0x0005) plen 6                       #1 [hci0] 67.991328
        Address: 2D:1D:D9:1F:CC:EB (Non-Resolvable)
> HCI Event: Command Complete (0x0e) plen 4                                     #2 [hci0] 67.991679
      LE Set Random Address (0x08|0x0005) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                      #3 [hci0] 67.991742
        Type: Active (0x01)
        Interval: 11.250 msec (0x0012)
        Window: 11.250 msec (0x0012)
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                     #4 [hci0] 67.992134
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                          #5 [hci0] 67.992198
        Scanning: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                     #6 [hci0] 67.992580
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                            {0x0001} [hci0] 67.992657
      Start Discovery (0x0023) plen 1
        Status: Success (0x00)
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
@ MGMT Event: Discovering (0x0013) plen 2                                 {0x0001} [hci0] 67.992676
        Address type: 0x07
          BR/EDR
[... 275 lines cut ...]
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                         #21 [hci0] 70.407203
        Scanning: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                    #22 [hci0] 70.408342
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                            {0x0001} [hci0] 70.408683
      Stop Discovery (0x0024) plen 1
        Status: Success (0x00)
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
@ MGMT Event: Discovering (0x0013) plen 2                                 {0x0001} [hci0] 70.408720
        Address type: 0x07
          BR/EDR
          LE Public
          LE Random
        Discovery: Disabled (0x00)
@ MGMT Command: Set IO Capability (0x0018) plen 1                         {0x0001} [hci0] 79.207502
        Capability: DisplayYesNo (0x01)
@ MGMT Event: Command Complete (0x0001) plen 3                            {0x0001} [hci0] 79.207534
      Set IO Capability (0x0018) plen 0
        Status: Success (0x00)
```

yduf, 2023-02-04:
Yes I am running GoveeBTTempLogger with full root privileges.
But the funny things is that it is not necessary to use sudo for bluetoothctl.

Not sure What you wanted to compare with the above trace.
The way I have used bluetoothctl to scan my govee devices:

bluetoothctl (no sudo needed)

```
$ bluetoothctl
Agent registered
[bluetooth]# menu scan
Menu scan:
Available commands:
... # remove menu displayed
[bluetooth]# transport le
[bluetooth]# duplicate-data on
[bluetooth]# back
Menu main:
... # remove menu displayed
[bluetooth]# scan on
SetDiscoveryFilter success
Discovery started
[CHG] Controller CC:D9:AC:C3:14:A4 Discovering: yes
[NEW] Device A4:C1:38:32:15:9E GVH5075_159E
[NEW] Device A4:C1:38:12:7D:CD GVH5075_7DCD
[NEW] Device A4:C1:38:E8:44:16 GVH5075_4416
[CHG] Device A4:C1:38:32:15:9E RSSI: -74
[CHG] Device A4:C1:38:32:15:9E ManufacturerData Key: 0xec88
[CHG] Device A4:C1:38:32:15:9E ManufacturerData Value:
  00 03 22 2f 49 00                                .."/I.          
[CHG] Device A4:C1:38:32:15:9E ManufacturerData Key: 0x004c
[CHG] Device A4:C1:38:32:15:9E ManufacturerData Value:
  02 15 49 4e 54 45 4c 4c 49 5f 52 4f 43 4b 53 5f  ..INTELLI_ROCKS_
  48 57 50 75 f2 ff c2                             HWPu...         
[CHG] Device A4:C1:38:12:7D:CD RSSI: -67
[CHG] Device A4:C1:38:12:7D:CD ManufacturerData Key: 0xec88
[CHG] Device A4:C1:38:12:7D:CD ManufacturerData Value:
  00 02 e3 9e 43 00                                ....C.          
[CHG] Device A4:C1:38:12:7D:CD ManufacturerData Key: 0x004c
[CHG] Device A4:C1:38:12:7D:CD ManufacturerData Value:
  02 15 49 4e 54 45 4c 4c 49 5f 52 4f 43 4b 53 5f  ..INTELLI_ROCKS_
  48 57 50 75 f2 ff c2                             HWPu...         
[CHG] Device A4:C1:38:E8:44:16 RSSI: -63
[CHG] Device A4:C1:38:E8:44:16 ManufacturerData Key: 0xec88
[CHG] Device A4:C1:38:E8:44:16 ManufacturerData Value:
  00 02 ef 5e 43 00                                ...^C.          
[CHG] Device A4:C1:38:E8:44:16 ManufacturerData Key: 0x004c
[CHG] Device A4:C1:38:E8:44:16 ManufacturerData Value:
  02 15 49 4e 54 45 4c 4c 49 5f 52 4f 43 4b 53 5f  ..INTELLI_ROCKS_
  48 57 50 75 f2 ff c2                             HWPu...         
[CHG] Device A4:C1:38:32:15:9E RSSI: -74
[CHG] Device A4:C1:38:32:15:9E ManufacturerData Key: 0xec88
[CHG] Device A4:C1:38:32:15:9E ManufacturerData Value:
  00 03 22 2f 49 00                                .."/I.          
[CHG] Device A4:C1:38:32:15:9E ManufacturerData Key: 0x004c
[CHG] Device A4:C1:38:32:15:9E ManufacturerData Value:
  02 15 49 4e 54 45 4c 4c 49 5f 52 4f 43 4b 53 5f  ..INTELLI_ROCKS_
  48 57 50 75 f2 ff c2                             HWPu...         
[CHG] Device A4:C1:38:12:7D:CD RSSI: -67
[CHG] Device A4:C1:38:12:7D:CD ManufacturerData Key: 0xec88
[CHG] Device A4:C1:38:12:7D:CD ManufacturerData Value:
  00 02 e3 9e 43 00                                ....C.          
[CHG] Device A4:C1:38:12:7D:CD ManufacturerData Key: 0x004c
[CHG] Device A4:C1:38:12:7D:CD ManufacturerData Value:
  02 15 49 4e 54 45 4c 4c 49 5f 52 4f 43 4b 53 5f  ..INTELLI_ROCKS_
  48 57 50 75 f2 ff c2                             HWPu...         
[CHG] Device A4:C1:38:E8:44:16 RSSI: -68
[CHG] Device A4:C1:38:E8:44:16 ManufacturerData Key: 0xec88
[CHG] Device A4:C1:38:E8:44:16 ManufacturerData Value:
[... 1121 lines cut ...]
          TX power: -62 dB
@ MGMT Command: Stop Discovery (0x0024) plen 1                          {0x0001} [hci0] 51.480435
        Address type: 0x06
          LE Public
          LE Random
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6              #69 [hci0] 51.480576
        Extended scan: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                  #70 [hci0] 51.555921
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                          {0x0001} [hci0] 51.556134
      Stop Discovery (0x0024) plen 1
        Status: Success (0x00)
        Address type: 0x06
          LE Public
          LE Random
@ MGMT Event: Discovering (0x0013) plen 2                               {0x0001} [hci0] 51.556162
        Address type: 0x06
          LE Public
          LE Random
        Discovery: Disabled (0x00)
```

wcbonner, 2023-02-04:
@yduf I don't really know what I was trying to see, mainly if a known program that could scan on your machine operated differently on my machine. 

As part of my visual studio build process, I run this command: `sudo setcap 'cap_net_raw,cap_net_admin+eip' $(RemoteTargetPath)` which allows the bluetooth stuff without requiring root access each time.

I've now pushed a new update to my code that sends an HCI Reset, and it doesn't seem to have any detrimental issues on my development machine. Let me know how it changes the btmon output on your machine.
```
wim@WimPi4-Dev:~ $ sudo btmon
Bluetooth monitor ver 5.55
= Note: Linux version 5.15.84-v8+ (aarch64)                                                0.161640
= Note: Bluetooth subsystem version 2.22                                                   0.161647
= New Index: DC:A6:32:B9:23:06 (Primary,UART,hci0)                                  [hci0] 0.161650
= Open Index: DC:A6:32:B9:23:06                                                     [hci0] 0.161651
= Index Info: DC:A6:32:B9:23:06 (Cypress Semiconductor)                             [hci0] 0.161653
@ MGMT Open: bluetoothd (privileged) version 1.21                                 {0x0001} 0.161658
@ RAW Open: GoveeBTTempLogg (privileged) version 2.22                             {0x0002} 8.213041
@ RAW Close: GoveeBTTempLogg                                                      {0x0002} 8.213097
@ RAW Open: GoveeBTTempLogg (privileged) version 2.22                      {0x0002} [hci0] 8.213181
< HCI Command: Reset (0x03|0x0003) plen 0                                        #1 [hci0] 8.213307
> HCI Event: Command Complete (0x0e) plen 4                                      #2 [hci0] 8.216433
      Reset (0x03|0x0003) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                           #3 [hci0] 8.216888
        Scanning: Disabled (0x00)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                      #4 [hci0] 8.217351
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Command Disallowed (0x0c)
< HCI Command: Read Local Name (0x03|0x0014) plen 0                              #5 [hci0] 8.217863
> HCI Event: Command Complete (0x0e) plen 252                                    #6 [hci0] 8.219193
      Read Local Name (0x03|0x0014) ncmd 1
        Status: Success (0x00)
        Name: WimPi4-Dev
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                       #7 [hci0] 8.220408
        Type: Active (0x01)
        Interval: 11.250 msec (0x0012)
        Window: 11.250 msec (0x0012)
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                      #8 [hci0] 8.220835
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                       #9 [hci0] 8.221113
        Type: Active (0x01)
        Interval: 5000.000 msec (0x1f40)
        Window: 5000.000 msec (0x1f40)
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                     #10 [hci0] 8.221572
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                          #11 [hci0] 8.221831
        Scanning: Enabled (0x01)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                     #12 [hci0] 8.222287
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 43                                       #13 [hci0] 8.229686
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Random (0x01)
        Address: E7:E7:B4:AB:4A:1F (Static)
        Data length: 31
        Flags: 0x04
          BR/EDR Not Supported
        16-bit Service UUIDs (complete): 1 entry
          Sonos, Inc. (0xfe07)
        Company: Sonos Inc (1447)
          Data: 051001000000000000012200ca
        Appearance: Unknown (0x0000)
        TX power: 0 dBm
        RSSI: -70 dBm (0xba)
```

yduf, 2023-02-05:
I get the update from master branch and tested it. It goes a bit farther, failing in hci_le_set_scan_enable:

$ sudo ./GoveeBTTempLogger/usr/local/bin/goveebttemplogger
```
[2023-02-05T09:13:18] GoveeBTTempLogger Version 2.20230204-1 Built on: Feb  5 2023 at 10:09:20
[2023-02-05T09:13:18] LocalName: 
[2023-02-05T09:13:18] No BlueTooth Address Filter: [00:00:00:00:00:00]
[                   ] Error: Failed to enable scan: Input/output error
GoveeBTTempLogger Version 2.20230204-1 Built on: Feb  5 2023 at 10:09:20 (exiting)
```

$ sudo btmon
```
[sudo] password for yves:        
Bluetooth monitor ver 5.64
= Note: Linux version 5.19.0-28-generic (x86_64)                                                           0.717271
= Note: Bluetooth subsystem version 2.22                                                                   0.717278
= New Index: CC:D9:AC:C3:14:A4 (Primary,USB,hci0)                                                   [hci0] 0.717279
= Open Index: CC:D9:AC:C3:14:A4                                                                     [hci0] 0.717280
= Index Info: CC:D9:AC:C3:14:A4 (Intel Corp.)                                                       [hci0] 0.717281
@ MGMT Open: bluetoothd (privileged) version 1.22                                                 {0x0001} 0.717283
@ RAW Open: goveebttemplogg (privileged) version 2.22                                             {0x0002} 4.775527
@ RAW Close: goveebttemplogg                                                                      {0x0002} 4.775550
@ RAW Open: goveebttemplogg (privileged) version 2.22                                      {0x0002} [hci0] 4.775579
< HCI Command: Reset (0x03|0x0003) plen 0                                                        #1 [hci0] 4.775676
> HCI Event: Command Complete (0x0e) plen 4                                                      #2 [hci0] 4.902212
      Reset (0x03|0x0003) ncmd 2
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                           #3 [hci0] 4.902539
        Scanning: Disabled (0x00)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                                      #4 [hci0] 4.904233
      LE Set Scan Enable (0x08|0x000c) ncmd 2
        Status: Success (0x00)
< HCI Command: Read Local Name (0x03|0x0014) plen 0                                              #5 [hci0] 4.904560
> HCI Event: Command Complete (0x0e) plen 252                                                    #6 [hci0] 4.908141
      Read Local Name (0x03|0x0014) ncmd 1
        Status: Success (0x00)
        Name: 
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                                       #7 [hci0] 4.908633
        Type: Active (0x01)
        Interval: 11.250 msec (0x0012)
        Window: 11.250 msec (0x0012)
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                      #8 [hci0] 4.909064
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                                       #9 [hci0] 4.909247
        Type: Active (0x01)
        Interval: 5000.000 msec (0x1f40)
        Window: 5000.000 msec (0x1f40)
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                     #10 [hci0] 4.910114
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                          #11 [hci0] 4.910341
        Scanning: Enabled (0x01)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                     #12 [hci0] 4.911062
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Invalid HCI Command Parameters (0x12)
@ RAW Close: goveebttemplogg    
```

yduf, 2023-02-05:
(delete previous post to clarify issue)
Got latest version from master, and tested it. It goes a bit farther but fail in [hci_le_set_scan_enable](https://github.com/wcbonner/GoveeBTTempLogger/blob/master/goveebttemplogger.cpp#L2096)

But when doing so, it's breaking bluetooth on my system, to the point that restarting bluetooth service is not sufficient,
and the only solution I have for now is to restart the computer.
When this happens launching bluetoothctl after goveebttemplogger gives me
`Failed to start discovery: org.bluez.Error.InProgress`

1) bluetoothctl working fine

2) $ ./GoveeBTTempLogger/usr/local/bin/goveebttemplogger => failure is expected here

```
[2023-02-05T10:08:54] GoveeBTTempLogger Version 2.20230204-1 Built on: Feb  5 2023 at 10:09:20
[                   ] Error: Could not reset host controller: 
[2023-02-05T10:08:54] LocalName: yves-huv
[2023-02-05T10:08:54] No BlueTooth Address Filter: [00:00:00:00:00:00]
[                   ] Error: Failed to set scan parameters: Operation not permitted
GoveeBTTempLogger Version 2.20230204-1 Built on: Feb  5 2023 at 10:09:20 (exiting)
```

Notice that LocalName: is **yves-huv**

3) $ sudo GoveeBTTempLogger/usr/local/bin/goveebttemplogger 

```
$ sudo ./GoveeBTTempLogger/usr/local/bin/goveebttemplogger 
[sudo] password for yves:        
[2023-02-05T10:11:50] GoveeBTTempLogger Version 2.20230204-1 Built on: Feb  5 2023 at 10:09:20
[2023-02-05T10:11:51] LocalName: 
[2023-02-05T10:11:51] No BlueTooth Address Filter: [00:00:00:00:00:00]
[                   ] Error: Failed to enable scan: Input/output error
GoveeBTTempLogger Version 2.20230204-1 Built on: Feb  5 2023 at 10:09:20 (exiting)
```

Notice that LocalName: is **empty now**

$ sudo btmon
```
Bluetooth monitor ver 5.64
= Note: Linux version 5.19.0-28-generic (x86_64)                                                           0.786920
= Note: Bluetooth subsystem version 2.22                                                                   0.786923
= New Index: CC:D9:AC:C3:14:A4 (Primary,USB,hci0)                                                   [hci0] 0.786923
= Open Index: CC:D9:AC:C3:14:A4                                                                     [hci0] 0.786924
= Index Info: CC:D9:AC:C3:14:A4 (Intel Corp.)                                                       [hci0] 0.786924
@ MGMT Open: bluetoothd (privileged) version 1.22                                                 {0x0001} 0.786925
@ RAW Open: goveebttemplogg (privileged) version 2.22                                             {0x0002} 8.981914
@ RAW Close: goveebttemplogg                                                                      {0x0002} 8.981921
@ RAW Open: goveebttemplogg (privileged) version 2.22                                      {0x0002} [hci0] 8.981929
< HCI Command: Reset (0x03|0x0003) plen 0                                                        #1 [hci0] 8.981975
> HCI Event: Command Complete (0x0e) plen 4                                                      #2 [hci0] 9.109878
      Reset (0x03|0x0003) ncmd 2
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                           #3 [hci0] 9.110150
        Scanning: Disabled (0x00)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                                      #4 [hci0] 9.110785
      LE Set Scan Enable (0x08|0x000c) ncmd 2
        Status: Success (0x00)
< HCI Command: Read Local Name (0x03|0x0014) plen 0                                              #5 [hci0] 9.110991
> HCI Event: Command Complete (0x0e) plen 252                                                    #6 [hci0] 9.114779
      Read Local Name (0x03|0x0014) ncmd 1
        Status: Success (0x00)
        Name: 
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                                       #7 [hci0] 9.115225
        Type: Active (0x01)
        Interval: 11.250 msec (0x0012)
        Window: 11.250 msec (0x0012)
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                      #8 [hci0] 9.115724
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                                       #9 [hci0] 9.115976
        Type: Active (0x01)
        Interval: 5000.000 msec (0x1f40)
        Window: 5000.000 msec (0x1f40)
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                     #10 [hci0] 9.116738
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                          #11 [hci0] 9.116904
        Scanning: Enabled (0x01)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                     #12 [hci0] 9.117778
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Invalid HCI Command Parameters (0x12)
@ RAW Close: goveebttemplogg  
```

4) bluetooth is broken now

$ bluetoothctl
```
Agent registered
[CHG] Controller CC:D9:AC:C3:14:A4 Pairable: yes
[bluetooth]# scan on
Failed to start discovery: org.bluez.Error.InProgress
[bluetooth]# 
```
and dmesg keep telling:
`[  742.315277] Bluetooth: hci0: Opcode 0x2039 failed: -16`

restarting bluetooth service won't fix that.

wcbonner, 2023-02-05:
@yduf Can you look at /usr/include/bluetooth/hci.h and see if these entries are the same as in my version:
```
#define OCF_LE_SET_SCAN_PARAMETERS		0x000B
typedef struct {
	uint8_t		type;
	uint16_t	interval;
	uint16_t	window;
	uint8_t		own_bdaddr_type;
	uint8_t		filter;
} __attribute__ ((packed)) le_set_scan_parameters_cp;
#define LE_SET_SCAN_PARAMETERS_CP_SIZE 7

#define OCF_LE_SET_SCAN_ENABLE			0x000C
typedef struct {
	uint8_t		enable;
	uint8_t		filter_dup;
} __attribute__ ((packed)) le_set_scan_enable_cp;
#define LE_SET_SCAN_ENABLE_CP_SIZE 2
```
the fact that I'm calling higher level functions in hci_lib.h should make any changes there transparent, but I'm grasping at differences now. 
```
int hci_le_set_scan_enable(int dev_id, uint8_t enable, uint8_t filter_dup, int to);
int hci_le_set_scan_parameters(int dev_id, uint8_t type, uint16_t interval,
					uint16_t window, uint8_t own_type,
					uint8_t filter, int to);
```
The master branch has the HCI_RESET removed from compiling now.

yduf, 2023-02-05:
I have check and they are exactly the same, this is are mine:
```
#define OCF_LE_SET_SCAN_PARAMETERS		0x000B
typedef struct {
	uint8_t		type;
	uint16_t	interval;
	uint16_t	window;
	uint8_t		own_bdaddr_type;
	uint8_t		filter;
} __attribute__ ((packed)) le_set_scan_parameters_cp;
#define LE_SET_SCAN_PARAMETERS_CP_SIZE 7

#define OCF_LE_SET_SCAN_ENABLE			0x000C
typedef struct {
	uint8_t		enable;
	uint8_t		filter_dup;
} __attribute__ ((packed)) le_set_scan_enable_cp;
#define LE_SET_SCAN_ENABLE_CP_SIZE 2

int hci_le_set_scan_enable(int dev_id, uint8_t enable, uint8_t filter_dup, int to);
int hci_le_set_scan_parameters(int dev_id, uint8_t type, uint16_t interval,
					uint16_t window, uint8_t own_type,
					uint8_t filter, int to);
```

yduf, 2023-02-05:
@wcbonner with your reset implementation and looking at [bluez](https://github.com/bluez/bluez/tree/e6b7d537d728ff390b76cb3f147e7411de61150b)/[lib](https://github.com/bluez/bluez/tree/e6b7d537d728ff390b76cb3f147e7411de61150b/lib)/hci.c I got a better understanding of how to issue hci command directly.
To the point that I successfully implemented my own version that mimic the blutetoothctl sequence captured by btmon.
As I was hoping, doing the same seq, allows to do trigger the le scan without failure.

While there is not much difference in the args used, it seems that using _extended_ operation that are not avaible in bluetooth/hci_lib.h make it works. And no need to call reset at all.

I did not do a full implementation, just the init code, so it ends on its own at step #6, just after having started the scan.

$ sudo btmon
```
Bluetooth monitor ver 5.64
= Note: Linux version 5.19.0-28-generic (x86_64)                                                           0.318185
= Note: Bluetooth subsystem version 2.22                                                                   0.318189
= New Index: CC:D9:AC:C3:14:A4 (Primary,USB,hci0)                                                   [hci0] 0.318192
= Open Index: CC:D9:AC:C3:14:A4                                                                     [hci0] 0.318192
= Index Info: CC:D9:AC:C3:14:A4 (Intel Corp.)                                                       [hci0] 0.318193
@ MGMT Open: bluetoothd (privileged) version 1.22                                                 {0x0001} 0.318195
@ RAW Open: goveescan (privileged) version 2.22                                                   {0x0002} 4.500438
@ RAW Close: goveescan                                                                            {0x0002} 4.500448
@ RAW Open: goveescan (privileged) version 2.22                                            {0x0002} [hci0] 4.500457
< HCI Command: LE Set Random Address (0x08|0x0005) plen 6                                        #1 [hci0] 4.500507
        Address: 00:00:00:00:00:00 (Non-Resolvable)
> HCI Event: Command Complete (0x0e) plen 4                                                      #2 [hci0] 4.616967
      LE Set Random Address (0x08|0x0005) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Parameters (0x08|0x0041) plen 8                              #3 [hci0] 4.617246
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
        PHYs: 0x01
        Entry 0: LE 1M
          Type: Active (0x01)
          Interval: 11.250 msec (0x0012)
          Window: 11.250 msec (0x0012)
> HCI Event: Command Complete (0x0e) plen 4                                                      #4 [hci0] 4.619020
      LE Set Extended Scan Parameters (0x08|0x0041) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                  #5 [hci0] 4.619251
        Extended scan: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                      #6 [hci0] 4.621010
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
@ RAW Close: goveescan                                                                     {0x0002} [hci0] 4.621181
> HCI Event: LE Meta Event (0x3e) plen 57                                                        #7 [hci0] 4.824196
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0013
            Props: 0x0013
              Connectable
              Scannable
              Use legacy advertising PDUs
            Data status: Complete
          Legacy PDU Type: ADV_IND (0x0013)
          Address type: Public (0x00)
          Address: A4:C1:38:E8:44:16 (Telink Semiconductor (Taipei) Co. Ltd.)
          Primary PHY: LE 1M
          Secondary PHY: No packets
          SID: no ADI field (0xff)
          TX power: 127 dBm
          RSSI: -71 dBm (0xb9)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x1f
        0d 09 47 56 48 35 30 37 35 5f 34 34 31 36 03 03  ..GVH5075_4416..
        88 ec 02 01 05 09 ff 88 ec 00 02 db ee 42 00     .............B. 
        Name (complete): GVH5075_4416
[... 162 lines cut ...]
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0013
            Props: 0x0013
              Connectable
              Scannable
              Use legacy advertising PDUs
            Data status: Complete
          Legacy PDU Type: ADV_IND (0x0013)
          Address type: Random (0x01)
          Address: 7B:F0:AC:CE:29:32 (Resolvable)
          Primary PHY: LE 1M
          Secondary PHY: No packets
          SID: no ADI field (0xff)
          TX power: 127 dBm
          RSSI: -85 dBm (0xab)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x03
        02 01 02                                         ...             
        Flags: 0x02
          LE General Discoverable Mode
```

wcbonner, 2023-02-05:
@yduf Can you supply the code snippet you used for the two function calls:

> ```
> < HCI Command: LE Set Extended Scan Parameters (0x08|0x0041) plen 8                              #3 [hci0] 4.617246
>         Own address type: Random (0x01)
>         Filter policy: Accept all [...]

I was looking at the function [hci_le_set_scan_enable](https://github.com/bluez/bluez/blob/d84ce72a543e090665a33ecac64b604805d2ec4c/lib/hci.c#L2917) as a model for issuing the extended scan parameters but didn't see the structure values in an easy format to know how to fill the [bt_hci_cmd_le_set_ext_scan_params](https://github.com/bluez/bluez/blob/d84ce72a543e090665a33ecac64b604805d2ec4c/monitor/bt.h#L2460) structure.

yduf, 2023-02-06:
Sure:
< HCI Command: LE Set Extended Scan Parameters (0x08|0x0041) plen 8                              #3 [hci0] 4.617246

```
static 
int set_extended( int dd, uint8_t type,
                uint16_t interval, uint16_t window,
                uint8_t own_type, uint8_t filter, int to) {

    #define BT_HCI_CMD_LE_SET_EXT_SCAN_PARAMS		0x2041
    struct bt_hci_cmd_le_set_ext_scan_params {
        uint8_t  own_addr_type;
        uint8_t  filter_policy;
        uint8_t  num_phys;
        uint8_t  type;
        uint16_t interval;
        uint16_t window;
    } __attribute__ ((packed));

    struct hci_request rq;
    bt_hci_cmd_le_set_ext_scan_params param_cp;
    uint8_t status;

    memset(&param_cp, 0, sizeof(param_cp));
    param_cp.type = type;
    param_cp.interval = interval;
    param_cp.window = window;
    param_cp.own_addr_type = own_type;
    param_cp.filter_policy = filter;
    param_cp.num_phys = 1;

    memset(&rq, 0, sizeof(rq));
    rq.ogf = OGF_LE_CTL;
    rq.ocf = BT_HCI_CMD_LE_SET_EXT_SCAN_PARAMS;
    rq.cparam = &param_cp;
    rq.clen = sizeof(bt_hci_cmd_le_set_ext_scan_params);
    rq.rparam = &status;
    rq.rlen = 1;

    if (hci_send_req( dd, &rq, to) < 0)
        return -1;

    if (status) {
        errno = EIO;
        return -1;
    }

    return 0;
}
```

< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6

```
static
int set_ext_scan_enable( int dd, uint8_t enable, uint8_t filter_dup, int to) {
    #define BT_HCI_CMD_LE_SET_EXT_SCAN_ENABLE		0x2042
    struct bt_hci_cmd_le_set_ext_scan_enable {
        uint8_t  enable;
        uint8_t  filter_dup;
        uint16_t duration;
        uint16_t period;
    } __attribute__ ((packed));

    struct hci_request rq;
    bt_hci_cmd_le_set_ext_scan_enable scan_cp;
    uint8_t status;

    memset(&scan_cp, 0, sizeof(scan_cp));
    scan_cp.enable = enable;
    scan_cp.filter_dup = filter_dup;

    memset(&rq, 0, sizeof(rq));
    rq.ogf = OGF_LE_CTL;
    rq.ocf = BT_HCI_CMD_LE_SET_EXT_SCAN_ENABLE;
    rq.cparam = &scan_cp;
    rq.clen = sizeof(scan_cp);
    rq.rparam = &status;
    rq.rlen = 1;

    if (hci_send_req( dd, &rq, to) < 0)
        return -1;

    if (status) {
        errno = EIO;
        return -1;
    }

    return 0;
}
```

called like this:
```
if (set_extended( socket, 0x01, htobs(0x0012), htobs(0x0012), LE_RANDOM_ADDRESS, 0x00, 1000) < 0)
      throw Error( "failed");

if (set_ext_scan_enable( socket, 0x01, 0x01, 1000) < 0)
      throw Error( "failed");
```

wcbonner, 2023-02-06:
@yduf I've pushed a new version of the code that first attempts the regular le_scan and if that fails attempts the extended le_scan. I was not able to test the extended scan, and don't know if the advertisements will be recognized by my program when requested in this fashion. 

I initially tried replacing the standard calls with the extended calls, and that didn't work for me.
```
wim@WimPi4-Dev:~ $ sudo btmon
Bluetooth monitor ver 5.55
= Note: Linux version 5.15.84-v8+ (aarch64)                                                              0.553332
= Note: Bluetooth subsystem version 2.22                                                                 0.553340
= New Index: DC:A6:32:B9:23:06 (Primary,UART,hci0)                                                [hci0] 0.553342
= Open Index: DC:A6:32:B9:23:06                                                                   [hci0] 0.553344
= Index Info: DC:A6:32:B9:23:06 (Cypress Semiconductor)                                           [hci0] 0.553347
@ MGMT Open: bluetoothd (privileged) version 1.21                                               {0x0001} 0.553352
@ RAW Open: GoveeBTTempLogg (privileged) version 2.22                                           {0x0002} 8.307641
@ RAW Close: GoveeBTTempLogg                                                                    {0x0002} 8.307683
@ RAW Open: GoveeBTTempLogg (privileged) version 2.22                                    {0x0002} [hci0] 8.307737
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                         #1 [hci0] 8.307813
        Scanning: Disabled (0x00)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                                    #2 [hci0] 8.308140
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Command Disallowed (0x0c)
< HCI Command: Read Local Name (0x03|0x0014) plen 0                                            #3 [hci0] 8.308224
> HCI Event: Command Complete (0x0e) plen 252                                                  #4 [hci0] 8.309462
      Read Local Name (0x03|0x0014) ncmd 1
        Status: Success (0x00)
        Name: WimPi4-Dev
< HCI Command: LE Set Extended Scan Parameters (0x08|0x0041) plen 8                            #5 [hci0] 8.309798
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
        PHYs: 0x01
        Entry 0: LE 1M
          Type: Active (0x01)
          Interval: 11.250 msec (0x0012)
          Window: 11.250 msec (0x0012)
> HCI Event: Command Complete (0x0e) plen 4                                                    #6 [hci0] 8.310160
      LE Set Extended Scan Parameters (0x08|0x0041) ncmd 1
        Status: Unknown HCI Command (0x01)
@ RAW Close: GoveeBTTempLogg                                                             {0x0002} [hci0] 8.310319
```

I found the definition for [BT_HCI_CMD_LE_SET_EXT_SCAN_PARAMS in the monitor program](https://github.com/bluez/bluez/blob/master/monitor/bt.h#L2460), but not in the Bluetooth development headers to be able to easily recognize if it's defined at compile time. Similarly, [BT_HCI_CMD_LE_SET_EXT_SCAN_ENABLE is in the monitor code](https://github.com/bluez/bluez/blob/master/monitor/bt.h#L2473).

I don't really like that I'm defining the constants in my code and then basing the calls on whether they are defined. If you can find them defined in the standard headers on your machine, I'd appreciate knowing in which files they are defined.

I'd also like to know if you receive LE Advertising Report (0x02) or if you now receive an extended advertising report?

yduf, 2023-02-06:
So far the same on my side, found [BT_HCI_CMD_LE_SET_EXT_SCAN_PARAMS only in the monitor program](https://github.com/bluez/bluez/blob/master/monitor/bt.h#L2460).

Your modification solve my bluetooth issue, as far as scanning is concerned => now scanning is succesfull.

But now, Ctr-C stop the GoveeBTTempLogger (signal is caught), but it doesn't stop scanning.
sudo btmon, show that scan continue while GoveeBTTempLogger is terminated.

And I see following event continuing to come through btmon:
```
> HCI Event: LE Meta Event (0x3e) plen 57                                                     #117 [hci0] 11.808571
      LE Extended Advertising Report (0x0d)

```

wcbonner, 2023-02-06:
@yduf New code push that hopefully uses the extended call to stop scanning properly. 

The default service file passes the parameter `--svg /var/www/html/goveebttemplogger/` so that's the directory that SVG files should be written unless you've overridden it to go elsewhere.

yduf, 2023-02-07:
This one solves all bluetooth issue. Many thanks!

closed 2023-02-07 by yduf

yduf, 2023-03-03:
Just a quick followup for completeness: 
Performing extended scan, produce event of type [bt_hci_le_ext_adv_report](https://github.com/bluez/bluez/blob/d84ce72a543e090665a33ecac64b604805d2ec4c/monitor/bt.h#L3498): (private type again), that are incompatible in layout with le_advertising_info. 

So in the end, the current fix allow a smooth bt  scan with no errors on my kernel & system. But events gathered are not recognized because of the different binary layout. That's the reason why I was not seeing any log or svg files.

Yet I am not reopening the issue:  it seems that currently I am the only one facing the original problem.

With the knowledge gained from this issue, I created my own [simple monitor](https://github.com/yduf/Govee-monitor).
It's use the extended parameters described above, and cover my use case so far.

Thanks again for your help!

PS:
This are the [extended event reported](https://github.com/bluez/bluez/blob/d84ce72a543e090665a33ecac64b604805d2ec4c/monitor/bt.h#L3498):
```c
struct bt_hci_le_ext_adv_report {
	uint16_t event_type;
	uint8_t  addr_type;
	uint8_t  addr[6];
	uint8_t  primary_phy;
	uint8_t  secondary_phy;
	uint8_t  sid;
	uint8_t  tx_power;
	int8_t   rssi;
	uint16_t interval;
	uint8_t  direct_addr_type;
	uint8_t  direct_addr[6];
	uint8_t  data_len;
	uint8_t  data[0];
} __attribute__ ((packed));
```

vs the "regular" one

```c
typedef struct {
	uint8_t		evt_type;
	uint8_t		bdaddr_type;
	bdaddr_t	bdaddr;
	uint8_t		length;
	uint8_t		data[];
} __attribute__ ((packed)) le_advertising_info;
```

wcbonner, 2023-03-03:
> Just a quick followup for completeness: Performing extended scan, produce event of type [bt_hci_le_ext_adv_report](https://github.com/bluez/bluez/blob/d84ce72a543e090665a33ecac64b604805d2ec4c/monitor/bt.h#L3498): (private type again), that are incompatible in layout with le_advertising_info.
> 
> So in the end, the current fix allow a smooth bt scan with no errors on my kernel & system. But events gathered are not recognized because of the different binary layout. That's the reason why I was not seeing any log or svg files.
> 
> Yet I am not reopening the issue: it seems that currently I am the only one facing the original problem.
> 
> With the knowledge gained from this issue, I created my own [simple monitor](https://github.com/yduf/Govee-monitor). It's use the extended parameters described above, and cover my use case so far.
> 
> Thanks again for your help!
> 
> PS: This is the type of the [extended event reported](https://github.com/bluez/bluez/blob/d84ce72a543e090665a33ecac64b604805d2ec4c/monitor/bt.h#L3498):
> 
> ```c
> struct bt_hci_le_ext_adv_report {
> 	uint16_t event_type;
> 	uint8_t  addr_type;
> 	uint8_t  addr[6];
> 	uint8_t  primary_phy;
> 	uint8_t  secondary_phy;
> 	uint8_t  sid;
> 	uint8_t  tx_power;
> 	int8_t   rssi;
> 	uint16_t interval;
> 	uint8_t  direct_addr_type;
> 	uint8_t  direct_addr[6];
> 	uint8_t  data_len;
> 	uint8_t  data[0];
> } __attribute__ ((packed));
> ```
> 
> vs the "regular" one
> 
> ```c
> typedef struct {
> 	uint8_t		evt_type;
> 	uint8_t		bdaddr_type;
> 	bdaddr_t	bdaddr;
> 	uint8_t		length;
> 	uint8_t		data[];
> } __attribute__ ((packed)) le_advertising_info;
> ```

@yduf This is interesting to me, and I've mostly completed my obsession with downloading historical data. 

I'd originally wondered if enabling extended scanning might return extended results, but without firsthand knowledge I've been hesitant to try to add the support for an unknown packet.  I was just looking at https://ubuntu.com/download/raspberry-pi and it looks like I should be able to create an Ubuntu Pi on a spare sd card pretty easily.  

Your original post described the code you were running on and processor as:
```
uname -a
 Linux yves-huv 5.19.0-28-generic #29~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Dec 15 12:05:40 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

Do you know what release you are running?

yduf, 2023-03-03:
Sure it's a close derivative of ubuntu:
OS: Linux Mint 21.1 x86_64 (issue was on 21.0 but I upgraded recently)
Kernel: 5.19.0-35-generic 
It should correspond to 
linux mint 21 (vanessa) <-> ubuntu Jammy 22.04 LTS

wcbonner, 2023-03-03:
I spent a while getting Ubuntu up and running on my Pi4 and it seems to use the standard commands instead of the extended commands. Looks like I'm not going to have a system that I can test extended events anytime soon.

In one window I ran the command `./GoveeBTTempLogger/usr/local/bin/goveebttemplogger --log /home/wim --svg /home/wim --only FF:FF:FF:FF:FF:FF`

While the other window I logged in, ran uname, and then btmon:
```
Welcome to Ubuntu 22.10 (GNU/Linux 5.19.0-1013-raspi aarch64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

0 updates can be applied immediately.

Last login: Fri Mar  3 14:52:04 2023 from 2604:4080:1304:8010:540a:6fa1:97a5:9947
wim@WimPi4-Ubuntu:~$ uname -a
Linux WimPi4-Ubuntu 5.19.0-1013-raspi #20-Ubuntu SMP PREEMPT Wed Jan 18 14:37:07 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux
wim@WimPi4-Ubuntu:~$ sudo btmon
[sudo] password for wim:
Bluetooth monitor ver 5.65
= Note: Linux version 5.19.0-1013-raspi (aarch64)                                                                  0.796618
= Note: Bluetooth subsystem version 2.22                                                                           0.796626
= New Index: DC:A6:32:B9:23:06 (Primary,UART,hci0)                                                          [hci0] 0.796628
= Open Index: DC:A6:32:B9:23:06                                                                             [hci0] 0.796630
= Index Info: DC:A6:32:B9:23:06 (Cypress Semiconductor)                                                     [hci0] 0.796631
@ MGMT Open: bluetoothd (privileged) version 1.22                                                         {0x0001} 0.796634
@ RAW Open: goveebttemplogg (privileged) version 2.22                                                    {0x0002} 44.099275
@ RAW Close: goveebttemplogg                                                                             {0x0002} 44.099317
@ RAW Open: goveebttemplogg (privileged) version 2.22                                             {0x0002} [hci0] 44.099401
< HCI Command: Read Local Name (0x03|0x0014) plen 0                                                     #1 [hci0] 44.099496
> HCI Event: Command Complete (0x0e) plen 252                                                           #2 [hci0] 44.100775
      Read Local Name (0x03|0x0014) ncmd 1
        Status: Success (0x00)
        Name: WimPi4-Ubuntu
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                                  #3 [hci0] 44.101096
        Scanning: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                             #4 [hci0] 44.101385
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Command Disallowed (0x0c)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                         #5 [hci0] 44.101486
        Extended scan: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                             #6 [hci0] 44.101760
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 1
        Status: Unknown HCI Command (0x01)
< HCI Command: LE Clear Accept List (0x08|0x0010) plen 0                                                #7 [hci0] 44.101859
> HCI Event: Command Complete (0x0e) plen 4                                                             #8 [hci0] 44.102204
      LE Clear Accept List (0x08|0x0010) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Add Device To Accept List (0x08|0x0011) plen 7                                        #9 [hci0] 44.102336
        Address type: Public (0x00)
        Address: A4:C1:38:05:C7:A1 (Telink Semiconductor (Taipei) Co. Ltd.)
> HCI Event: Command Complete (0x0e) plen 4                                                            #10 [hci0] 44.102755
      LE Add Device To Accept List (0x08|0x0011) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Add Device To Accept List (0x08|0x0011) plen 7                                       #11 [hci0] 44.102867
        Address type: Public (0x00)
        Address: A4:C1:38:0D:3B:10 (Telink Semiconductor (Taipei) Co. Ltd.)
> HCI Event: Command Complete (0x0e) plen 4                                                            #12 [hci0] 44.103254
      LE Add Device To Accept List (0x08|0x0011) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Add Device To Accept List (0x08|0x0011) plen 7                                       #13 [hci0] 44.103351
        Address type: Public (0x00)
        Address: A4:C1:38:13:AE:36 (Telink Semiconductor (Taipei) Co. Ltd.)
> HCI Event: Command Complete (0x0e) plen 4                                                            #14 [hci0] 44.103817
      LE Add Device To Accept List (0x08|0x0011) ncmd 1
        Status: Success (0x00)
[... 100 lines cut ...]
        Num reports: 1
        Event type: Scan response - SCAN_RSP (0x04)
        Address type: Public (0x00)
        Address: A4:C1:38:05:C7:A1 (Telink Semiconductor (Taipei) Co. Ltd.)
        Data length: 11
        Company: not assigned (60552)
          Data: 004d0045095902
        RSSI: -77 dBm (0xb3)
> HCI Event: LE Meta Event (0x3e) plen 43                                                              #32 [hci0] 46.028358
[... cut ...]
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Public (0x00)
        Address: A4:C1:38:DC:CC:3D (Telink Semiconductor (Taipei) Co. Ltd.)
        Data length: 31
        Name (complete): GVH5174_CC3D
        16-bit Service UUIDs (complete): 1 entry
          Unknown (0xec88)
        Flags: 0x05
          LE Limited Discoverable Mode
          BR/EDR Not Supported
        Company: Nokia Mobile Phones (1)
          Data: 010102a58a60
        RSSI: -45 dBm (0xd3)
```