wcbonner/GoveeBTTempLogger #47: Very new user looking to try the code out

Lakonaphoto, 2023-05-19:
Hello I am trying to use the templogger, and I am new to all of this. 
I am trying to run this on my Raspberry pi 3B+

Started by verifying my Kernel version: 6.1.21-v7+
Then I made sure my Raspberry pi was updated and upgraded: Sudo apt update and sudo apt upgrade

Then installed libbluetooth-dev:  sudo apt install bluetooth pi-bluetooth bluez blueman

Then I tried: git clone https://github.com/wcbonner/GoveeBTTempLogger.git

and: sudo apt-get install bluetooth bluez libbluetooth-dev dpkg-dev 
make deb
sudo make install-deb

At this point I reread the readme and tried to verify the files were created. I was unable to find them in /var/log or /var/www/html 

Thank you for putting this all together and if anyone could help me get this running I would appreciate it.

wcbonner, 2023-05-19:
Did the make deb command complete without errors? 

I realized a change I made recently using the C++17 filesystem functions limits the newer code to building on updated bullseye systems.

Lakonaphoto, 2023-05-21:
No, it said: 
make: *** No rule to make target 'deb'.  Stop.

wcbonner, 2023-05-21:
> No, it said: make: *** No rule to make target 'deb'. Stop.

@Lakonaphoto It sounds like you didn't change directory into the cloned software directory. I just updated the readme to include that step just now.  I also updated the requirements to mention a problem building on older editions of Debian.

wcbonner, 2023-05-31:
> That was the trick, thank you very much! I can see the files being updated in the folder. What do you use to view the output besides navigating to the folder and opening each?

@Lakonaphoto I run the program with just the log file and index options and it will scan the log files and create an html index file you can load if you are running a web server. 

```
--i (--index) HTML index file for SVG files, must be paired with log directory. HTML file is a fully qualified name. This is meant as a one time run option just to create a simple index of all the SVG files. i (--index) HTML index file for SVG files, must be paired with log directory. HTML file is a fully qualified name. This is meant as a one time run option just to create a simple index of all the SVG files.
```

closed 2023-05-31 by wcbonner