wcbonner/GoveeBTTempLogger #48: Checkout for <dirent.h> ..?

eGit, 2023-05-25:
What git command should I use to check out and compile the latest version that still uses <dirent.h> ..?

wcbonner, 2023-05-27:
> What git command should I use to check out and compile the latest version that still uses <dirent.h> ..?

@eGit : I was able to build the version prior to switching to <filesystem> using the following commands:
```
git clone https://github.com/wcbonner/GoveeBTTempLogger.git
cd GoveeBTTempLogger
git reset --hard 6c72d30d0e7b40753fe9bbca94f8dc361c46d0c1
make deb
```
You can see my full output below.
```
wim@FlightTracker:~/foo/GoveeBTTempLogger $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
wim@FlightTracker:~/foo $ git clone https://github.com/wcbonner/GoveeBTTempLogger.git
Cloning into 'GoveeBTTempLogger'...
remote: Enumerating objects: 1424, done.
remote: Counting objects: 100% (386/386), done.
remote: Compressing objects: 100% (189/189), done.
remote: Total 1424 (delta 223), reused 339 (delta 191), pack-reused 1038
Receiving objects: 100% (1424/1424), 2.62 MiB | 1.68 MiB/s, done.
Resolving deltas: 100% (731/731), done.
wim@FlightTracker:~/foo $ cd GoveeBTTempLogger/
wim@FlightTracker:~/foo/GoveeBTTempLogger $ git reset --hard 6c72d30d0e7b40753fe9bbca94f8dc361c46d0c1
HEAD is now at 6c72d30 added optional passive BTLE scanning
wim@FlightTracker:~/foo/GoveeBTTempLogger $ make deb
cc    -c -o uuid.o uuid.c
mkdir -p GoveeBTTempLogger/usr/local/bin
g++ -Wno-psabi -O3 -std=c++17 goveebttemplogger.cpp uuid.o -oGoveeBTTempLogger/usr/local/bin/goveebttemplogger -lbluetooth
sudo setcap 'cap_net_raw,cap_net_admin+eip' GoveeBTTempLogger/usr/local/bin/goveebttemplogger
g++ -Wno-psabi -O3 -std=c++17 gvh-organizelogs.cpp -oGoveeBTTempLogger/usr/local/bin/gvh-organizelogs -lbluetooth
# Set architecture for the resulting .deb to the actually built architecture
sed -i "s/Architecture: .*/Architecture: armhf/" GoveeBTTempLogger/DEBIAN/control
chmod a+x GoveeBTTempLogger/DEBIAN/postinst GoveeBTTempLogger/DEBIAN/postrm GoveeBTTempLogger/DEBIAN/prerm
dpkg-deb --build GoveeBTTempLogger
dpkg-deb: building package 'goveebttemplogger' in 'GoveeBTTempLogger.deb'.
dpkg-name --overwrite GoveeBTTempLogger.deb
dpkg-name: info: moved 'GoveeBTTempLogger.deb' to './GoveeBTTempLogger_2.20230414-1_armhf.deb'
dpkg-deb --build GoveeBTTempLogger
dpkg-deb: building package 'goveebttemplogger' in 'GoveeBTTempLogger.deb'.
wim@FlightTracker:~/foo/GoveeBTTempLogger $ 
```

eGit, 2023-05-27:
Thanks. Seems to work.

What doesn't work is writing SVGs with Celsius unit: sudo goveebttemplogger -c true

Always prints Fahrenheid.

wcbonner, 2023-05-27:
> Thanks. Seems to work.
> 
> What doesn't work is writing SVGs with Celsius unit: sudo goveebttemplogger -c true
> 
> Always prints Fahrenheid.

Try using the c option with no extra parameters..  

I just looked at the code that parses the parameters:
```
		case 'c':
			SVGFahrenheit = false;
			break;
```

If that doesn't work, it'll take me a lot longer to look into it because I'm traveling for the next couple of weeks.

closed 2023-08-02 by wcbonner