wcbonner/GoveeBTTempLogger #49: Temperature for H5075 slightly wrong

dirk754, 2023-06-08:
The temperature for the H5075 is not correct, but the error is less than 0.1 °C. The problem is that the humidity is included in the temperature. You can see it in the example in the code: "88ec00 03519e 64 00 Temp: 21.7502°C Temp: 71.1504°F Humidity: 50.2%"

The last three digits in the temperature (Celsius) are the same digits as in the humidity. It seems to me that the description in  https://github.com/Thrilleratplay/GoveeWatcher is not correct. To get correct values I changed
"Temperature[0] = float(iTemp) / 10000.0;" to "Temperature[0] = float(iTemp/1000) / 10.0;"

wcbonner, 2023-06-08:
> The temperature for the H5075 is not correct, but the error is less than 0.1 °C. The problem is that the humidity is included in the temperature. You can see it in the example in the code: "88ec00 [...]

I'm going to think on this one since I'm not near a real computer for another couple days. 

Im pretty sure the claimed accuracy of any of these thermometers is only 0.2°C, so I might want to round to those increments and truncate. 

Due to the nature of real number storage in computers I chose to do the conversion to floating point first and keep the extra digits created.

dirk754, 2023-06-08:
Hello William,

maybe I was not not clear enough. It is not about rounding or about the
accuracy of the sensors! After converting the hexadecimal number into
decimal the first three digits are the temperature and the last three
digits are the humidity. So "03519e" converts to "217502" which means 21.7
°C and 50.2% humidity without any rounding.

BTW, thanks for your software. You are the reason I bought 12 of this
sensors. :-)

Bye Dirk

wcbonner, 2023-06-08:
> maybe I was not not clear enough. It is not about rounding or about the
> accuracy of the sensors! After converting the hexadecimal number into
> decimal the first three digits are the temperature and [...]

Ah. I've been on a sailboat for the last two weeks and my brain is a little foggy. I'm not even sure what day of the week it is without switching to the calendar app. 

Each of the different thermometers uses a slightly different data packing system, and your follow up clarification was greatly helpful. (I'll probably put it in the code comments when I get home and have a chance to look at things on a real computer)

It's always great to have appreciation for the code. I seem to have fallen into a rabbit hole of buying and playing with these devices. I've got them in refrigerators and freezers and have learned that different refrigerators use completely different cycling algorithms to keep their interior cool. Also having them outside produces almost useless relative humidity values, at least in Seattle. 

Wim

wcbonner, 2023-06-11:
@dirk754 I swapped in your equation and pushed the fix

closed 2023-06-11 by wcbonner