wcbonner/GoveeBTTempLogger #50: Input/Output Error Starting GoveeBTTempLogger

blackketter, 2023-09-01:
Not sure if this is something I missed, but following the instructions with my Pi ZeroW, I'm seeing this error in the syslog:

```
Sep  1 10:34:07 picam systemd[1]: Starting GoveeBTTempLogger service...
Sep  1 10:34:07 picam systemd[1]: Started GoveeBTTempLogger service.
Sep  1 10:34:07 picam goveebttemplogger[2763]: GoveeBTTempLogger Version 2.20230831-1 Built on: Sep  1 2023 at 10:21:02 (starting)
Sep  1 10:34:07 picam goveebttemplogger[2763]: [                   ] Error: Failed to enable scan: Input/output error (5)
Sep  1 10:34:07 picam goveebttemplogger[2763]: GoveeBTTempLogger Version 2.20230831-1 Built on: Sep  1 2023 at 10:21:02 (exiting)
Sep  1 10:34:07 picam systemd[1]: goveebttemplogger.service: Succeeded.
```

hciconfig seems to work:
```
/var/log $ hciconfig
hci0:	Type: Primary  Bus: UART
	BD Address: B8:27:EB:97:50:9E  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING
	RX bytes:5850 acl:0 sco:0 events:180 errors:0
	TX bytes:3245 acl:0 sco:0 commands:180 errors:0
```

and

```
sudo hcitool lescan
LE Scan ...
40:8F:ED:C6:8F:21 (unknown)
6C:D2:AC:27:A8:1E (unknown)
7E:E9:C2:B5:1D:A5 (unknown)
A4:C1:38:E4:3F:C8 GVH5075_3FC8
A4:C1:38:E4:3F:C8 (unknown)
67:59:9D:3B:C6:DE (unknown)
67:59:9D:3B:C6:DE (unknown)
71:9C:12:F9:16:AB (unknown)
5E:92:F2:D6:2B:C6 (unknown)
2F:CC:2E:CC:29:37 (unknown)
6F:29:8A:E4:DF:9C (unknown)
6F:29:8A:E4:DF:9C (unknown)
```

Not sure what the IO error is, though I noticed that hcitool lescan fails with Operation Not Permitted, if that's relevant...

Any suggestions?

blackketter, 2023-09-01:
Followup... I found the line in the cpp file that generates this error and the warning below it.  Ran the suggested command 
```
sudo setcap 'cap_net_raw,cap_net_admin+eip' goveebttemplogger
```
but that didn't help

wcbonner, 2023-09-01:
@blackketter It's been a while since I've run this on a PiZero. It used to work, and I've not tested it recently.

I happen to have a Raspberry Pi Zero 2 W that arrived a couple of weeks ago and I got running but didn't do anything more with. I just ran the following commands and seem to duplicate your problem. 

I will look into fixing it over the next few days.
```
sudo apt install git bluetooth bluez libbluetooth-dev dpkg-dev
git clone https://github.com/wcbonner/GoveeBTTempLogger.git
cd GoveeBTTempLogger
make deb
sudo apt install ./GoveeBTTempLogger_2.20230901-1_armhf.deb
```
then I checked on the status
```
wim@WimPiZero2W:~/GoveeBTTempLogger $ sudo systemctl status goveebttemplogger.service
● goveebttemplogger.service - GoveeBTTempLogger service
     Loaded: loaded (/usr/local/lib/systemd/system/goveebttemplogger.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) since Fri 2023-09-01 13:09:18 PDT; 4s ago
    Process: 30133 ExecStartPre=/bin/mkdir -p ${LOGDIR} (code=exited, status=0/SUCCESS)
    Process: 30134 ExecStart=/usr/local/bin/goveebttemplogger -v ${VERBOSITY} --log ${LOGDIR} --time ${TIME} $SVGARGS $EXTRAARGS (code=exited, status=0/SUCCESS)
   Main PID: 30134 (code=exited, status=0/SUCCESS)
        CPU: 28ms
wim@WimPiZero2W:~/GoveeBTTempLogger $ tail -f /var/log/syslog
Sep  1 13:09:23 WimPiZero2W systemd[1]: Started GoveeBTTempLogger service.
Sep  1 13:09:23 WimPiZero2W goveebttemplogger[30148]: GoveeBTTempLogger Version 2.20230901-1 Built on: Sep  1 2023 at 13:05:32 (starting)
Sep  1 13:09:23 WimPiZero2W goveebttemplogger[30148]: [                   ] Error: Failed to enable scan: Input/output error (5)
Sep  1 13:09:23 WimPiZero2W goveebttemplogger[30148]: GoveeBTTempLogger Version 2.20230901-1 Built on: Sep  1 2023 at 13:05:32 (exiting)
Sep  1 13:09:23 WimPiZero2W systemd[1]: goveebttemplogger.service: Succeeded.
Sep  1 13:09:28 WimPiZero2W systemd[1]: goveebttemplogger.service: Scheduled restart job, restart counter is at 5.
Sep  1 13:09:28 WimPiZero2W systemd[1]: Stopped GoveeBTTempLogger service.
Sep  1 13:09:28 WimPiZero2W systemd[1]: goveebttemplogger.service: Start request repeated too quickly.
Sep  1 13:09:28 WimPiZero2W systemd[1]: goveebttemplogger.service: Failed with result 'start-limit-hit'.
Sep  1 13:09:28 WimPiZero2W systemd[1]: Failed to start GoveeBTTempLogger service.

wim@WimPiZero2W:~/GoveeBTTempLogger $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
wim@WimPiZero2W:~/GoveeBTTempLogger $ uname -a
Linux WimPiZero2W 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux

```

wcbonner, 2023-09-01:
I just went to an older PiZeroW I have access to. I built and ran the program without installing it. No problems there. Running it with the verbosity set to 2 let it display all of the messages it was receiving on the console. On the non-working Pi running bullseye the extended verbosity wasn't reaching a point in the code where there was any more useful information to display.

```
pi@WimPiZeroW-Brett:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
pi@WimPiZeroW-Brett:~ $ uname -a
Linux WimPiZeroW-Brett 5.10.103+ #1529 Tue Mar 8 12:19:18 GMT 2022 armv6l GNU/Linux
pi@WimPiZeroW-Brett:~ $ GoveeBTTempLogger/GoveeBTTempLogger/usr/local/bin/goveebttemplogger
[2023-09-01T21:12:00] GoveeBTTempLogger Version 2.20230901-1 Built on: Sep  1 2023 at 13:58:09
[2023-09-01T21:12:00] LocalName: WimPiZeroW-Brett
[2023-09-01T21:12:00] No BlueTooth Address Filter
[2023-09-01T21:12:00] Scanning Stopped.
[2023-09-01T21:12:00] Scanning Started. ScanInterval(40 msec) ScanWindow(30 msec) ScanType(1)
^C***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
[2023-09-01T21:12:16] Scanning Stopped.
GoveeBTTempLogger Version 2.20230901-1 Built on: Sep  1 2023 at 13:58:09 (exiting)
pi@WimPiZeroW-Brett:~ $ GoveeBTTempLogger/GoveeBTTempLogger/usr/local/bin/goveebttemplogger -v 2
[2023-09-01T21:12:34] GoveeBTTempLogger Version 2.20230901-1 Built on: Sep  1 2023 at 13:58:09
[                   ]      log: ""
[                   ]      svg: ""
[                   ]  battery: 0
[                   ]   minmax: 0
[                   ]  celsius: false
[                   ] titlemap: ""
[                   ]     time: 60
[                   ]  average: 5
[                   ] download: 0
[                   ]  passive: false
[                   ] no-bluetooth: false
[2023-09-01T21:12:34] LocalName: WimPiZeroW-Brett
[2023-09-01T21:12:34] No BlueTooth Address Filter
[2023-09-01T21:12:34] Scanning Stopped.
[2023-09-01T21:12:34] Scanning Started. ScanInterval(40 msec) ScanWindow(30 msec) ScanType(1)
[2023-09-01T21:12:34] 40 [48:AD:5A:F3:A9:2D] (Flags) 06 (Name) PBL-056BFB0 (Tx Power) 03 (Other: 12) 00010002
[2023-09-01T21:12:34] 46 [44:A6:E5:72:FF:4C] (Flags) 06 (UUID) 0010 (Manu) 30313335303130330000000000000000000000000000
[2023-09-01T21:12:34] 33 [70:24:4B:64:71:7C] (Flags) 1A (Tx Power) 07 (Manu) 4C001006331E0EC5F3A5 (Apple) 1006331E0EC5F3
[2023-09-01T21:12:34] 32 [42:09:52:B8:54:09] (Flags) 1A (Tx Power) 04 (Manu) 4C001005031C5DB9F9 (Apple) 1005031C5DB9
[2023-09-01T21:12:34] 40 [48:AD:5A:F3:A9:2D] (Flags) 06 (Name) PBL-056BFB0 (Tx Power) 03 (Other: 12) 00010002
[2023-09-01T21:12:34] 15 [48:AD:5A:F3:A9:2D]
[2023-09-01T21:12:34] 32 [6C:08:F6:FD:D5:87] (Flags) 1A (Tx Power) 0C (Manu) 4C001005031C80BF0F (Apple) 1005031C80BF
[2023-09-01T21:12:34] 15 [6C:08:F6:FD:D5:87]
[2023-09-01T21:12:34] 46 [44:A6:E5:72:FF:4C] (Flags) 06 (UUID) 0010 (Manu) 30313335303130330000000000000000000000000000
[2023-09-01T21:12:34] 44 [44:A6:E5:72:FF:4C] (Name) BLE#0x44A6E572FF4C (Other: 12) 08001000 (Tx Power) 9F
[2023-09-01T21:12:34] 46 [44:A6:E5:72:FF:4C] (Flags) 06 (UUID) 0010 (Manu) 30313335303130330000000000000000000000000000
[2023-09-01T21:12:34] 40 [48:AD:5A:F3:A9:2D] (Flags) 06 (Name) PBL-056BFB0 (Tx Power) 03 (Other: 12) 00010002
[2023-09-01T21:12:34] 33 [70:24:4B:64:71:7C] (Flags) 1A (Tx Power) 07 (Manu) 4C001006331E0EC5F3A5 (Apple) 1006331E0EC5F3
[2023-09-01T21:12:34] 32 [6C:08:F6:FD:D5:87] (Flags) 1A (Tx Power) 0C (Manu) 4C001005031C80BF0F (Apple) 1005031C80BF
[2023-09-01T21:12:34] 40 [48:AD:5A:F3:A9:2D] (Flags) 06 (Name) PBL-056BFB0 (Tx Power) 03 (Other: 12) 00010002
[2023-09-01T21:12:34] 46 [44:A6:E5:72:FF:4C] (Flags) 06 (UUID) 0010 (Manu) 30313335303130330000000000000000000000000000
[2023-09-01T21:12:34] 46 [44:A6:E5:72:FF:4C] (Flags) 06 (UUID) 0010 (Manu) 30313335303130330000000000000000000000000000
[2023-09-01T21:12:34] 32 [42:09:52:B8:54:09] (Flags) 1A (Tx Power) 04 (Manu) 4C001005031C5DB9F9 (Apple) 1005031C5DB9
[2023-09-01T21:12:34] 15 [42:09:52:B8:54:09]
[2023-09-01T21:12:34] 32 [6C:08:F6:FD:D5:87] (Flags) 1A (Tx Power) 0C (Manu) 4C001005031C80BF0F (Apple) 1005031C80BF
[2023-09-01T21:12:34] 46 [44:A6:E5:72:FF:4C] (Flags) 06 (UUID) 0010 (Manu) 30313335303130330000000000000000000000000000
[2023-09-01T21:12:34] 46 [44:A6:E5:72:FF:4C] (Flags) 06 (UUID) 0010 (Manu) 30313335303130330000000000000000000000000000
[2023-09-01T21:12:34] 44 [44:A6:E5:72:FF:4C] (Name) BLE#0x44A6E572FF4C (Other: 12) 08001000 (Tx Power) 9F
[2023-09-01T21:12:35] 33 [70:24:4B:64:71:7C] (Flags) 1A (Tx Power) 07 (Manu) 4C001006331E0EC5F3A5 (Apple) 1006331E0EC5F3
[2023-09-01T21:12:35] 32 [42:09:52:B8:54:09] (Flags) 1A (Tx Power) 04 (Manu) 4C001005031C5DB9F9 (Apple) 1005031C5DB9
[2023-09-01T21:12:35] 46 [44:A6:E5:72:FF:4C] (Flags) 06 (UUID) 0010 (Manu) 30313335303130330000000000000000000000000000
[2023-09-01T21:12:35] 46 [44:A6:E5:72:FF:4C] (Flags) 06 (UUID) 0010 (Manu) 30313335303130330000000000000000000000000000
[2023-09-01T21:12:35] 40 [48:AD:5A:F3:A9:2D] (Flags) 06 (Name) PBL-056BFB0 (Tx Power) 03 (Other: 12) 00010002
[2023-09-01T21:12:35] 46 [44:A6:E5:72:FF:4C] (Flags) 06 (UUID) 0010 (Manu) 30313335303130330000000000000000000000000000
[2023-09-01T21:12:35] 44 [44:A6:E5:72:FF:4C] (Name) BLE#0x44A6E572FF4C (Other: 12) 08001000 (Tx Power) 9F
[... cut ...]
[2023-09-01T21:12:35] 15 [42:09:52:B8:54:09]
[2023-09-01T21:12:35] 32 [6C:08:F6:FD:D5:87] (Flags) 1A (Tx Power) 0C (Manu) 4C001005031C80BF0F (Apple) 1005031C80BF
[2023-09-01T21:12:35] 15 [6C:08:F6:FD:D5:87]
[2023-09-01T21:12:35] 40 [48:AD:5A:F3:A9:2D] (Flags) 06 (Name) PBL-056BFB0 (Tx Power) 03 (Other: 12) 00010002
^C***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
[2023-09-01T21:12:35] Scanning Stopped.
GoveeBTTempLogger Version 2.20230901-1 Built on: Sep  1 2023 at 13:58:09 (exiting)
```

wcbonner, 2023-09-01:
@blackketter If you can find any web references as to what has changed with Bluetooth on the PiZero between buster and bullseye I'd appreciate any help. 

I have searched but not come up with anything constructive. 

I'd not had any issue with the transition on the Pi4 or Pi3 units I'm running in most places, and the PiZero units that I have weren't running this software anymore.

blackketter, 2023-09-02:
Thanks, @wcbonner!  I'll look around.

To be clear, I have a Zero 2 W running Bullseye.  

Do you believe that the issue is related to Bullseye vs Buster or Zero2W vs ZeroW?

wcbonner, 2023-09-02:
> To be clear, I have a Zero 2 W running Bullseye.
> 
> Do you believe that the issue is related to Bullseye vs Buster or Zero2W vs ZeroW?

I didn't catch that you were on a Zero 2 W. That's the same model I had failng here. 

I'll see if I can find my Zero and test on it. It would be good to isolate where the problem lies.

wcbonner, 2023-09-02:
@blackketter The problem definitely seems to be local to the Zero2W. Anything that you can find out about Bluetooth differences on the PiZero2W would be appreciated.

I just built a clean install on a brand new PiZeroW. 

```
wim@WimPiZeroW-4:~ $ history
    1  sudo apt update && sudo apt -y upgrade && sudo apt -y dist-upgrade && sudo apt autoremove -y
    2  sudo apt install bluetooth bluez libbluetooth-dev dpkg-dev bluetooth bluez libbluetooth-dev git
    3  git clone https://github.com/wcbonner/GoveeBTTempLogger.git
    4  cd GoveeBTTempLogger/
    5  make deb
    6  ~/GoveeBTTempLogger/GoveeBTTempLogger/usr/local/bin/goveebttemplogger
    7  history
```

and it ran perfectly.

```
wim@WimPiZeroW-4:~/GoveeBTTempLogger $ ~/GoveeBTTempLogger/GoveeBTTempLogger/usr/local/bin/goveebttemplogger
[2023-09-02T05:32:07] GoveeBTTempLogger Version 2.20230901-1 Built on: Sep  1 2023 at 22:27:22
[2023-09-02T05:32:07] LocalName: WimPiZeroW-4
[2023-09-02T05:32:07] No BlueTooth Address Filter
[2023-09-02T05:32:07] Scanning Stopped.
[2023-09-02T05:32:07] Scanning Started. ScanInterval(40 msec) ScanWindow(30 msec) ScanType(1)
[2023-09-02T05:32:07] 26 [E3:8E:C8:C1:98:9A] (Temp) 20.31°C (Humidity) 61.45% (Battery) 100% (GVH5074)
[2023-09-02T05:32:08] 46 [C2:35:33:30:25:50] (Temp) 22.1555°C (Humidity) 55.5% (Battery) 100% (GVH5100)
[2023-09-02T05:32:08] 26 [E3:60:59:23:14:7D] (Temp) 10.5°C (Humidity) 70.36% (Battery) 100% (GVH5074)
[2023-09-02T05:32:08] 46 [A4:C1:38:0D:3B:10] (Temp) 19.064°C (Humidity) 64% (Battery) 47% (GVH5177)
[2023-09-02T05:32:08] 26 [E3:60:59:21:80:65] (Temp) -7.1°C (Humidity) 42.07% (Battery) 94% (GVH5074)
[2023-09-02T05:32:08] 44 [E3:8E:C8:C1:98:9A] (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_989A
[2023-09-02T05:32:08] 42 [E3:8E:C8:C1:98:9A] (Manu) 4C000215494E54454C4C495F524F434B535F485750749A98C2 (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5074 (Minor) 9A98 (RSSI) C2
[2023-09-02T05:32:09] 44 [E3:60:59:21:80:65] (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_8065
[2023-09-02T05:32:09] 42 [E3:60:59:21:80:65] (Manu) 4C000215494E54454C4C495F524F434B535F485750746580C2 (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5074 (Minor) 6580 (RSSI) C2
[2023-09-02T05:32:09] 44 [E3:8E:C8:C1:98:9A] (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_989A
[2023-09-02T05:32:09] 26 [E3:8E:C8:C1:98:9A] (Manu) 88EC00F00704186402 (Temp) 20.32°C (Humidity) 61.48% (Battery) 100% (GVH5074)
[2023-09-02T05:32:09] 26 [A4:C1:38:05:C7:A1] (Temp) 19.28°C (Humidity) 64.92% (Battery) 94% (GVH5074)
^C***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
[2023-09-02T05:32:09] Scanning Stopped.
GoveeBTTempLogger Version 2.20230901-1 Built on: Sep  1 2023 at 22:27:22 (exiting)
```

Just for reference, I've run this command:
```
wim@WimPiZeroW-4:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 000000009cd77483
Model           : Raspberry Pi Zero W Rev 1.1
```

and on the zero2W
```
wim@WimPiZero2W:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : 902120
Serial          : 000000002e3f5718
Model           : Raspberry Pi Zero 2 W Rev 1.0
```

blackketter, 2023-09-02:
Ok, great, thanks.

I built an sd card with buster and installed the latest from git am not seeing the problem.  Working well (at least for the last 5 minutes. :)

So it appears to be specific to Zero2W on Bullseye, I guess.

blackketter, 2023-09-02:
> So it appears to be specific to Zero2W on Bullseye, I guess.

Confirmed.  Found an old ZeroW in a drawer, GoveeBTTempLogger works great on Bullseye on that hardware.

blackketter, 2023-09-02:
Golly, this just happened:
```
dean@bullseye:~ $ sudo hcitool lescan --duplicates
Set scan parameters failed: Input/output error
```
This is on bullseye on a ZeroW (not 2), after running successfully for a while.  
Tried stopping the goveebttemplogger service, which didn't help.
Rebooted and it started working again.

wcbonner, 2023-09-15:
@blackketter I was doing something else and came across the information that the Raspberry Pi Zero 2 W will run the 64 bit kernel. Since I wasn't doing anything with my Zero2W I reinstalled with 64bit to see if that made a difference, unfortunately not.

```
wim@WimPiZero2W:~/GoveeBTTempLogger $ ~/GoveeBTTempLogger/GoveeBTTempLogger/usr/local/bin/goveebttemplogger
[2023-09-15T23:04:15] GoveeBTTempLogger Version 2.20230906-1 Built on: Sep 15 2023 at 16:02:41
[2023-09-15T23:04:15] LocalName: WimPiZero2W
[2023-09-15T23:04:15] No BlueTooth Address Filter
[2023-09-15T23:04:15] Scanning Stopped.
[                   ] Error: Failed to enable scan: Input/output error (5)
GoveeBTTempLogger Version 2.20230906-1 Built on: Sep 15 2023 at 16:02:41 (exiting)
wim@WimPiZero2W:~/GoveeBTTempLogger $ cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : 902120
Serial          : 000000002e3f5718
Model           : Raspberry Pi Zero 2 W Rev 1.0
wim@WimPiZero2W:~/GoveeBTTempLogger $ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
wim@WimPiZero2W:~/GoveeBTTempLogger $ uname -a
Linux WimPiZero2W 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

wcbonner, 2023-10-11:
I was really hoping that bookworm would fix the issue on the PiZero2W. No such luck. I just built a new install, built the code and got the same results.

```
wim@WimPiZero2W:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
wim@WimPiZero2W:~ $ uname -a
Linux WimPiZero2W 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux

wim@WimPiZero2W:~ $ ~/GoveeBTTempLogger/build/goveebttemplogger -v 9
[2023-10-11T17:20:18] GoveeBTTempLogger Version 2.20231009.2 Built on: Oct 11 2023 at 10:17:58
[                   ]      log: ""
[                   ]    cache: ""
[                   ]      svg: ""
[                   ]  battery: 0
[                   ]   minmax: 0
[                   ]  celsius: false
[                   ] titlemap: ""
[                   ]     time: 60
[                   ]  average: 5
[                   ] download: 0
[                   ]  passive: false
[                   ] no-bluetooth: false
[2023-10-11T17:20:18] LocalName: WimPiZero2W
[2023-10-11T17:20:18] No BlueTooth Address Filter
[2023-10-11T17:20:18] Scanning Stopped.
[                   ] Error: Failed to enable scan: Input/output error (5)
GoveeBTTempLogger Version 2.20231009.2 Built on: Oct 11 2023 at 10:17:58 (exiting)

wim@WimPiZero2W:~ $ history
    1  lsb_release -a
    2  sudo apt update && sudo apt -y upgrade && sudo apt -y dist-upgrade && sudo apt autoremove -y
    3  sudo apt install build-essentials cmake git libbluetooth-dev
    4  sudo apt install build-essential cmake git libbluetooth-dev
    5  git clone https://github.com/wcbonner/GoveeBTTempLogger.git
    6  cmake -S GoveeBTTempLogger -B GoveeBTTempLogger/build
    7  cmake --build GoveeBTTempLogger/build
    8  pushd GoveeBTTempLogger/build && cpack . && popd
    9  ~/GoveeBTTempLogger/build/goveebttemplogger
   10  ~/GoveeBTTempLogger/build/goveebttemplogger -v 9
   11  history
```

wcbonner, 2023-11-25:
@blackketter I was testing something related to another issue and just realized I am successfully running the program on the Pi Zero 2 W now. If you're still in a position to test it on your machine I'd love to know if either the current code just runs for you or if it will work after running other Bluetooth commands. Here's what suddenly worked:

After rebooting the Pi my program still reported the same error as before, then I ran bluetoothctl with a short scan and my program was able to enable scanning after that. 

I'm traveling and will have to think about how to run Wireshark remotely to see what may be happening in the bluetoothctl command.

```
wim@WimPiZero2W:~ $ bluetoothctl
Agent registered
[CHG] Controller B8:27:EB:95:FD:B2 Pairable: yes
[bluetooth]# scan on
Discovery started
[CHG] Controller B8:27:EB:95:FD:B2 Discovering: yes
[NEW] Device A4:C1:38:D5:A3:3B Govee_H5074_A33B
[NEW] Device E3:5E:CC:21:5C:0F Govee_H5074_5C0F
[NEW] Device E3:60:59:23:14:7D Govee_H5074_147D
[NEW] Device A4:C1:38:05:C7:A1 Govee_H5074_C7A1
[NEW] Device E3:8E:C8:C1:98:9A Govee_H5074_989A
[NEW] Device E3:60:59:21:80:65 Govee_H5074_8065
[NEW] Device A4:C1:38:37:BC:AE GVH5075_BCAE
[NEW] Device 16:0F:92:27:CC:C3 16-0F-92-27-CC-C3
[bluetooth]# scan off
Discovery stopped
[CHG] Device 16:0F:92:27:CC:C3 RSSI is nil
[CHG] Device A4:C1:38:37:BC:AE RSSI is nil
[CHG] Device E3:60:59:21:80:65 RSSI is nil
[CHG] Device E3:8E:C8:C1:98:9A RSSI is nil
[CHG] Device A4:C1:38:05:C7:A1 RSSI is nil
[CHG] Device E3:60:59:23:14:7D RSSI is nil
[CHG] Device E3:5E:CC:21:5C:0F RSSI is nil
[CHG] Device A4:C1:38:D5:A3:3B RSSI is nil
[CHG] Controller B8:27:EB:95:FD:B2 Discovering: no
[DEL] Device A4:C1:38:D5:A3:3B Govee_H5074_A33B
[DEL] Device E3:5E:CC:21:5C:0F Govee_H5074_5C0F
[DEL] Device E3:60:59:23:14:7D Govee_H5074_147D
[DEL] Device A4:C1:38:05:C7:A1 Govee_H5074_C7A1
[DEL] Device E3:8E:C8:C1:98:9A Govee_H5074_989A
[DEL] Device E3:60:59:21:80:65 Govee_H5074_8065
[DEL] Device A4:C1:38:37:BC:AE GVH5075_BCAE
[DEL] Device 16:0F:92:27:CC:C3 16-0F-92-27-CC-C3
[bluetooth]# exit
wim@WimPiZero2W:~ $ ~/GoveeBTTempLogger/build/goveebttemplogger -v 3 --controller B8:27:EB:95:FD:B2
[2023-11-25T21:34:28] GoveeBTTempLogger Version 2.20231109.2 Built on: Nov  9 2023 at 12:17:44
[                   ]      log: ""
[                   ]    cache: ""
[                   ]      svg: ""
[                   ]  battery: 0
[                   ]   minmax: 0
[                   ]  celsius: false
[                   ] titlemap: ""
[                   ]     time: 60
[                   ]  average: 5
[                   ] download: 0
[                   ]  passive: false
[                   ] no-bluetooth: false
[                   ] BlueToothDevice_ID: 0
[                   ] Reset device: hci0. Success(0)
[                   ] DOWN device: hci0. Success(0)
[                   ] UP device: hci0. Success(0)
[2023-11-25T21:34:28] Controller Address: B8:27:EB:95:FD:B2
[2023-11-25T21:34:28] LocalName: WimPiZero2W
[2023-11-25T21:34:28] No BlueTooth Address Filter
[2023-11-25T21:34:28] Scanning Stopped.
[2023-11-25T21:34:28] Scanning Started. ScanInterval(11.25 msec) ScanWindow(11.25 msec) ScanType(1)
[2023-11-25T21:34:28] 44 [E3:5E:CC:21:5C:0F] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_5C0F
[2023-11-25T21:34:28] 42 [E3:5E:CC:21:5C:0F] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F485750740F5CC2 (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5074 (Minor) 0F5C (RSSI) C2
[2023-11-25T21:34:29] 44 [E3:60:59:23:14:7D] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_147D
[2023-11-25T21:34:29] 44 [E3:8E:C8:C1:98:9A] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_989A
[2023-11-25T21:34:29] 44 [E3:5E:CC:21:5C:0F] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_5C0F
[2023-11-25T21:34:30] 46 [C2:35:33:30:25:50] (bdaddr_type) 01 (evt_type) 00 (Flags) 06 (Name) GVH5100_2550 (UUID) 88EC (Manu) 01000101022C863D (Temp) 14.247°C (Humidity) 47% (Battery) 61% (GVH5100)
[... 56 lines cut ...]
[2023-11-25T21:36:23] Scanning Started. ScanInterval(11.25 msec) ScanWindow(11.25 msec) ScanType(1)
[2023-11-25T21:36:23] 44 [E3:5E:CC:21:5C:0F] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_5C0F
[2023-11-25T21:36:23] 36 [44:AC:63:0D:20:F1] (bdaddr_type) 01 (evt_type) 00 (Flags) 1A (Tx Power) 0C (Manu) 4C000F0590004EDD1810020604 (Apple) 0F0590004EDD18100206
[2023-11-25T21:36:23] 39 [2D:91:60:E1:FB:B3] (bdaddr_type) 01 (evt_type) 03 (Flags) 1A (Manu) 4C00090603FDA9FE89E713071261644E7AB32E (Apple) 090603FDA9FE89E713071261644E7AB3
[2023-11-25T21:36:24] 39 [16:0F:92:27:CC:C3] (bdaddr_type) 01 (evt_type) 03 (Flags) 1A (Manu) 4C00090603A8A9FEA7E613071215DE2C518A37 (Apple) 090603A8A9FEA7E613071215DE2C518A
[2023-11-25T21:36:24] 36 [50:B8:4E:A1:61:E5] (bdaddr_type) 01 (evt_type) 00 (Flags) 1A (Tx Power) 0C (Manu) 4C000F0590008EBEBC10020004 (Apple) 0F0590008EBEBC100200
[2023-11-25T21:36:24] 46 [C2:35:33:30:25:50] (bdaddr_type) 01 (evt_type) 00 (Flags) 06 (Name) GVH5100_2550 (UUID) 88EC (Manu) 01000101022C863B (Temp) 14.247°C (Humidity) 47% (Battery) 59% (GVH5100)
[2023-11-25T21:36:24] 36 [44:AC:63:0D:20:F1] (bdaddr_type) 01 (evt_type) 00 (Flags) 1A (Tx Power) 0C (Manu) 4C000F0590004EDD1810020604 (Apple) 0F0590004EDD18100206
[... cut ...]
[2023-11-25T21:36:25] 44 [E3:60:59:23:14:7D] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_147D
^C***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
[2023-11-25T21:36:25] 26 [E3:60:59:23:14:7D] (bdaddr_type) 00 (evt_type) 04 (Manu) 88EC000D0027125F02 (Temp) 0.13°C (Humidity) 46.47% (Battery) 95% (GVH5074)
[2023-11-25T21:36:25] Scanning Stopped.
GoveeBTTempLogger Version 2.20231109.2 Built on: Nov  9 2023 at 12:17:44 (exiting)
```

blackketter, 2023-11-25:
Alas, my Zero 2 W is not available right now, but I'll get my hands on another and give it a try...

wcbonner, 2023-11-25:
> Alas, my Zero 2 W is not available right now, but I'll get my hands on another and give it a try...

@blackketter No hurry. I am just happy to have a lead to follow up. There may be a sequence of commands I can issue that will make it work on the 2W without adversely effecting other platforms. It's probably going to require rebuilding my 2W with a gui so I can run Wireshark on it and capture exactly what's happening. That won't happen till I'm home from my Thanksgiving trip.

closed 2023-11-29 by wcbonner

wcbonner, 2023-11-29:
@blackketter I believe that I have this correctly working on the PiZero2W now!  See https://github.com/wcbonner/GoveeBTTempLogger/commit/537e6211edbb5524dfd15846aa0c817df3566e7a for the function I added. 

I need to do a little more checking to see that it's OK not to create a random bluetooth address of my own, but so far it seems to work.

I expect I'll be able to close this issue in the next day or two.