wcbonner/GoveeBTTempLogger #51: --index {filepath} \ in service config causes service to exit directly after writing index file

TimoGoetze, 2023-10-16:
I would expect to be able to use --index within the service to automatically add new devices to my svg-dashboard.

Instead it restarts after writing the files. It is not listening.

Log V2 with --index in service config:

Oct 16 08:11:38 energy goveebttemplogger[13149]: [2023-10-16T06:11:38] Writing: "/var/www/html/goveebttemplogger/index.html"
Oct 16 08:11:38 energy goveebttemplogger[13149]: [2023-10-16T06:11:38] Writing: A4C1380611DD
Oct 16 08:11:38 energy goveebttemplogger[13149]: [2023-10-16T06:11:38] Done
Oct 16 08:11:41 energy systemd[1]: Stopped GoveeBTTempLogger service.
Oct 16 08:11:41 energy systemd[1]: Starting GoveeBTTempLogger service...
Oct 16 08:11:41 energy systemd[1]: Started GoveeBTTempLogger service.
Oct 16 08:11:41 energy goveebttemplogger[13181]: [2023-10-16T06:11:41] Reading: "/var/log/goveebttemplogger"
Oct 16 08:11:41 energy goveebttemplogger[13181]: [2023-10-16T06:11:41] Writing: "/var/www/html/goveebttemplogger/index.html"
Oct 16 08:11:41 energy goveebttemplogger[13181]: [2023-10-16T06:11:41] Writing: A4C1380611DD
Oct 16 08:11:41 energy goveebttemplogger[13181]: [2023-10-16T06:11:41] Done
Oct 16 08:11:46 energy systemd[1]: goveebttemplogger.service: Service hold-off time over, scheduling restart.
Oct 16 08:11:46 energy systemd[1]: Stopped GoveeBTTempLogger service.
Oct 16 08:11:46 energy systemd[1]: Starting GoveeBTTempLogger service...
Oct 16 08:11:46 energy systemd[1]: Started GoveeBTTempLogger service.
Oct 16 08:11:46 energy goveebttemplogger[13243]: [2023-10-16T06:11:46] Reading: "/var/log/goveebttemplogger"
Oct 16 08:11:46 energy goveebttemplogger[13243]: [2023-10-16T06:11:46] Writing: "/var/www/html/goveebttemplogger/index.html"
Oct 16 08:11:46 energy goveebttemplogger[13243]: [2023-10-16T06:11:46] Writing: A4C1380611DD
Oct 16 08:11:46 energy goveebttemplogger[13243]: [2023-10-16T06:11:46] Done

the index file contens is correct. After removing --index from service config, everything runs properly.

Thanks for this great tool ! - I use it on raspy 3 :-)

wcbonner, 2023-10-16:
> using release arm binary https://github.com/wcbonner/GoveeBTTempLogger/releases/tag/v2.20231001.1.
> 
> I would expect to be able to use --index within the service to automatically add new devices to my svg-dashboard.
> 
> Instead it restarts after writing the files. It is not listening.
> 
> the index file contens is correct. After removing --index from service config, everything runs properly.
> 
> Thanks for this great tool ! - I use it on raspy 3 :-)

Yes, the index option is operating as planned.

https://github.com/wcbonner/GoveeBTTempLogger#minor-update-2022-12-17 shows an example of the way I use the index option.  The program can be run this way without stopping the instance running in the background listening to Bluetooth advertisements and creating log files.

I wanted to create a starting index file when there are a lot of devices that are being monitored. I can then easily manually reorder the sections or add details to the index file in a normal editor.

I should have created the index program as a separate executable but because everything so far has been a monolithic code file, keeping all the file handling in the same place was the safer option. My alternate solution would have been to break the code into multiple header files and code files. 

I've been happy the tool is useful beyond my own obsession.

closed 2023-10-16 by wcbonner

TimoGoetze, 2023-10-16:
Thank you for your fast Answer. Then I only would suggest to add this behaviour to the docs in Readme.md. I did not find and read the other example without advice. 

My suggestion: 

i (--index) **Command line option. Not for usage in service**: HTML index file for SVG files, must be paired with log directory. HTML file is a fully qualified name. This is meant as a one time run option just to create a simple index of all the SVG files.

Cheers.