wcbonner/GoveeBTTempLogger #55: Usage questions: Custom paths and support for H5100

franzos, 2023-11-25:
I'm trying to collect data from 3x H5100 but I have trouble with two points:

1. There doesn't seem to be any log / cache / svg output

```bash
goveebttemplogger \
	--verbose 1 \
	--log /root/goveebttemplogger/log \
	--time 60 \
	--svg /root/goveebttemplogger/html --battery 8 --minmax 8 \
	--cache /root/goveebttemplogger/cache \
	--download
```

I created and checked above dirs, but there's nothing there (`/log`, `/html`, `cache`), even after hours of running, and stdout like this:

```bash
[2023-11-24T22:55:12] 300 seconds or more have passed. Writing SVG Files
[2023-11-24T22:55:12] 60 seconds or more have passed. Writing LOG Files
```

I assume there's nothing, because nothing was found?

---

2. I don't actually seem to pickup anything

```bash
[2023-11-25T15:31:30] GoveeBTTempLogger Version 2.20231001.1 Built on: Jan  1 1970 at 00:00:01
[2023-11-25T15:31:30] LocalName: BlueZ
[2023-11-25T15:31:30] No BlueTooth Address Filter
[2023-11-25T15:31:30] Scanning Stopped.
[2023-11-25T15:31:30] Scanning Started. ScanInterval(40 msec) ScanWindow(30 msec) ScanType(1)
[2023-11-25T15:31:31] 300 seconds or more have passed. Writing SVG Files
[2023-11-25T15:31:31] No recent Bluetooth LE Advertisments! (> 3 Minutes)
[2023-11-25T15:31:31] Scanning Stopped.
...
# just repeats from here on
```

I can discover the devices fine, both from the computer as well as my mobile (which I currently use to collect data). I paired one to see if it makes any difference, but it does not.

As far as I understand, these devices are supposed to broadcast the stats occasionally, but I couldn't pickup any relevant BLE data in nRF Connect on Android. Maybe there's a config I missed?

[image]

I also tried `--only CC:35:34:...` without any luck.

wcbonner, 2023-11-25:
I'd run the program from the command line without logging to see if it's correctly receiving data. Below is an example where I built the latest code from github on my development machine, then ran it with verbosity set to 1. You can see that it saw a couple of govee thermometers pretty quickly, including a  GVH5100.

```
wim@WimPi4-Dev:~/GoveeBTTempLogger $ !1981
cmake -S ~/GoveeBTTempLogger -B ~/GoveeBTTempLogger/build && cmake --build ~/GoveeBTTempLogger/build && pushd ~/GoveeBTTempLogger/build && cpack . && popd
-- Components to pack:
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wim/GoveeBTTempLogger/build
Scanning dependencies of target goveebttemplogger
[ 20%] Building CXX object CMakeFiles/goveebttemplogger.dir/goveebttemplogger.cpp.o
[ 40%] Linking CXX executable goveebttemplogger
Setting Raw Priveleges on $<TARGET_FILE:goveebttemplogger>
[ 60%] Built target goveebttemplogger
[100%] Built target gvh-organizelogs
~/GoveeBTTempLogger/build ~/GoveeBTTempLogger
CPack: Create package using DEB
CPack: Install projects
CPack: - Run preinstall target for: GoveeBTTempLogger
CPack: - Install project: GoveeBTTempLogger []
CPack: Create package
CPackDeb: - Generating dependency list
CPack: - package: /home/wim/GoveeBTTempLogger/build/goveebttemplogger_2.20231109.2-2_arm64.deb generated.
~/GoveeBTTempLogger
wim@WimPi4-Dev:~/GoveeBTTempLogger $ ~/GoveeBTTempLogger/build/goveebttemplogger -v 1
[2023-11-25T16:42:00] GoveeBTTempLogger Version 2.20231109.2 Built on: Nov 25 2023 at 08:41:10
[                   ] BlueToothDevice_ID: 0
[                   ] Reset device: hci0. Success(0)
[                   ] DOWN device: hci0. Success(0)
[                   ] UP device: hci0. Success(0)
[2023-11-25T16:42:00] LocalName: WimPi4-Dev
[2023-11-25T16:42:00] No BlueTooth Address Filter
[2023-11-25T16:42:00] Scanning Stopped.
[2023-11-25T16:42:00] Scanning Started. ScanInterval(11.25 msec) ScanWindow(11.25 msec) ScanType(1)
[2023-11-25T16:42:00] 46 [C2:35:33:30:25:50] (Temp) 9.0497°C (Humidity) 49.7% (Battery) 58% (GVH5100)
[2023-11-25T16:42:02] 46 [C2:35:33:30:25:50] (Flags) 06 (Name) GVH5100_2550 (UUID) 88EC (Manu) 010001010161803A (Temp) 9.0496°C (Humidity) 49.6% (Battery) 58% (GVH5100)
[2023-11-25T16:42:03] 46 [A4:C1:38:0D:3B:10] (Temp) 7.9537°C (Humidity) 53.7% (Battery) 17% (GVH5177)
[2023-11-25T16:42:04] 46 [A4:C1:38:37:BC:AE] (Temp) 10.6°C (Humidity) 45.3% (Battery) 25% (GVH5075)
[2023-11-25T16:42:04] 46 [C2:35:33:30:25:50] (Flags) 06 (Name) GVH5100_2550 (UUID) 88EC (Manu) 010001010161813A (Temp) 9.0497°C (Humidity) 49.7% (Battery) 58% (GVH5100)
[2023-11-25T16:42:06] 46 [C2:35:33:30:25:50] (Flags) 06 (Name) GVH5100_2550 (UUID) 88EC (Manu) 010001010165693A (Temp) 9.1497°C (Humidity) 49.7% (Battery) 58% (GVH5100)
[2023-11-25T16:42:08] 46 [C2:35:33:30:25:50] (Flags) 06 (Name) GVH5100_2550 (UUID) 88EC (Manu) 010001010165693A (Temp) 9.1497°C (Humidity) 49.7% (Battery) 58% (GVH5100)
^C***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
[2023-11-25T16:42:09] Scanning Stopped.
GoveeBTTempLogger Version 2.20231109.2 Built on: Nov 25 2023 at 08:41:10 (exiting)
wim@WimPi4-Dev:~/GoveeBTTempLogger $
```

Increasing the verbosity would confirm that it's getting something from the bluetooth scanning..  

```
wim@WimPi4-Dev:~/GoveeBTTempLogger $ ~/GoveeBTTempLogger/build/goveebttemplogger -v 3
[2023-11-25T16:47:44] GoveeBTTempLogger Version 2.20231109.2 Built on: Nov 25 2023 at 08:41:10
[                   ]      log: ""
[                   ]    cache: ""
[                   ]      svg: ""
[                   ]  battery: 0
[                   ]   minmax: 0
[                   ]  celsius: false
[                   ] titlemap: ""
[                   ]     time: 60
[                   ]  average: 5
[                   ] download: 0
[                   ]  passive: false
[                   ] no-bluetooth: false
[                   ] BlueToothDevice_ID: 0
[                   ] Reset device: hci0. Success(0)
[                   ] DOWN device: hci0. Success(0)
[                   ] UP device: hci0. Success(0)
[2023-11-25T16:47:44] LocalName: WimPi4-Dev
[2023-11-25T16:47:44] No BlueTooth Address Filter
[... 119 lines cut ...]
[                   ] |  |  |  |  |  |  ^------------------------> le_advertising_info.bdaddr_type
[                   ] |  |  |  |  |  ^---------------------------> ??
[                   ] |  |  |  |  ^------------------------------> le_advertising_info.evt_type
[                   ] |  |  |  ^---------------------------------> evt_le_meta_event.subevent = EVT_LE_ADVERTISING_REPORT = 02
[                   ] |  |  ^------------------------------------> ?? length (bytes following)
[                   ] |  ^---------------------------------------> hci_event_hdr.plen = EVT_LE_META_EVENT = 3E
[                   ] ^------------------------------------------> hci_event_hdr.evt = HCI_EVENT_PKT = 04
[2023-11-25T16:50:05] 44 [E3:60:59:23:14:7D] (bdaddr_type) 00 (evt_type) 00 (Flags) [LE General Discoverable Mode][LE General Discoverable Mode] (UUID) 0A18F5FE88EC (Name) Govee_H5074_147D
[2023-11-25T16:50:05] Read: 39 Bytes
[                   ] 04 3E 24 02 01 03 01 7C C2 2C A5 EA 31 18 02 01 1A 14 FF 4C 00 09 06 03 95 A9 FE A7 E6 13 07 12 78 EB 78 85 99 D4 AE
[                   ] ^^ ^^ ^^ ^^ ^^ ^^ ^^ ^^                ^^--> le_advertising_info.length (bytes following)
[... cut ...]
[2023-11-25T16:50:05] 39 [31:EA:A5:2C:C2:7C] (bdaddr_type) 01 (evt_type) 03 (Flags) [LE General Discoverable Mode][Simultaneous LE and BR/EDR (Controller)][Simultaneous LE and BR/EDR (Host)] (Manu) 4C0009060395A9FEA7E613071278EB788599D4 (Apple) 09060395A9FEA7E613071278EB788599
^C***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
[2023-11-25T16:50:05] Scanning Stopped.
GoveeBTTempLogger Version 2.20231109.2 Built on: Nov 25 2023 at 08:41:10 (exiting)
wim@WimPi4-Dev:~/GoveeBTTempLogger $
```

franzos, 2023-11-25:
I appreciate your suggestions. Unfortunately I can't seem to see anything. I put one of the sensors on the Laptop to be sure.

```
goveebttemplogger -v 9
[2023-11-25T17:17:48] GoveeBTTempLogger Version 2.20231001.1 Built on: Jan  1 1970 at 00:00:01
[                   ]      log: ""
[                   ]    cache: ""
[                   ]      svg: ""
[                   ]  battery: 0
[                   ]   minmax: 0
[                   ]  celsius: false
[                   ] titlemap: ""
[                   ]     time: 60
[                   ]  average: 5
[                   ] download: 0
[                   ]  passive: false
[                   ] no-bluetooth: false
[2023-11-25T17:17:48] LocalName: BlueZ
[2023-11-25T17:17:48] No BlueTooth Address Filter
[2023-11-25T17:17:48] Scanning Stopped.
[2023-11-25T17:17:48] Scanning Started. ScanInterval(40 msec) ScanWindow(30 msec) ScanType(1)
[2023-11-25T17:17:50] Read: 60 Bytes
[-------------------] 04 3E 39 0D 01 13 00 01 2A 6F 35 34 35 C8 01 00 FF 7F AE 00 00 00 00 00 00 00 00 00 1F 02 01 06 0D 09 47 56 48 35 31 30 30 5F 36 46 32 41 03 03 88 EC 09 FF 01 00 01 01 03 79 0D 59
[2023-11-25T17:17:50] No recent Bluetooth LE Advertisments! (> 3 Minutes)
[2023-11-25T17:17:50] Scanning Stopped.
[2023-11-25T17:17:50] Scanning Started. ScanInterval(40 msec) ScanWindow(30 msec) ScanType(1)
[2023-11-25T17:17:53] Read: 60 Bytes
[-------------------] 04 3E 39 0D 01 13 00 01 1B 3C 35 34 35 CC 01 00 FF 7F CB 00 00 00 00 00 00 00 00 00 1F 02 01 06 0D 09 47 56 48 35 31 30 30 5F 33 43 31 42 03 03 88 EC 09 FF 01 00 01 01 03 9C 15 58
[2023-11-25T17:17:53] No recent Bluetooth LE Advertisments! (> 3 Minutes)
[2023-11-25T17:17:53] Scanning Stopped.
```

On the same computer, Bluetoothctl quickly finds two devices:

```
scan on
Discovery started
[CHG] Controller 94:E6:... Discovering: yes
...
[NEW] Device C8:35:34:... GVH5100_6F2A
[CHG] Device C8:35:34:... ManufacturerData Key: 0x0001
[CHG] Device C8:35:34:... ManufacturerData Value:
  01 01 03 a7 c6 58                                .....X          
[CHG] Device C8:35:34:... ManufacturerData Key: 0x004c
[CHG] Device C8:35:34:... ManufacturerData Value:
  02 15 49 4e 54 45 4c 4c 49 5f 52 4f 43 4b 53 5f  ..INTELLI_ROCKS_
  48 57 50 75 f2 ff 0c                             HWPu...         
[CHG] Device CC:35:34:... RSSI: -52
[CHG] Device CC:35:34:... ManufacturerData Key: 0x0001
[CHG] Device CC:35:34:... ManufacturerData Value:
  01 01 03 9c 0e 58                                .....X          
[CHG] Device CC:35:34:... ManufacturerData Key: 0x004c
[CHG] Device CC:35:34:... ManufacturerData Value:
  02 15 49 4e 54 45 4c 4c 49 5f 52 4f 43 4b 53 5f  ..INTELLI_ROCKS_
  48 57 50 75 f2 ff 0c                             HWPu...         
[CHG] Device C8:35:34:... RSSI: -43
[CHG] Device C8:35:34:... ManufacturerData Key: 0x0001
[CHG] Device C8:35:34:... ManufacturerData Value:
  01 01 03 a7 c5 58                                .....X          
[CHG] Device C8:35:34:... ManufacturerData Key: 0x004c
[CHG] Device C8:35:34:... ManufacturerData Value:
  02 15 49 4e 54 45 4c 4c 49 5f 52 4f 43 4b 53 5f  ..INTELLI_ROCKS_
  48 57 50 75 f2 ff 0c
```

So the H5100 sends out packages by itself, correct? If so, I should be able to pick it up in nRF Connect too - no? Could it be that for some reason my H5100 don't send out this data? I already looked trough their app - I didn't really change anything.

Here's what I have:
- Firmware: 1.00.14
- Hardware: 3.01.01

wcbonner, 2023-11-25:
> So the H5100 sends out packages by itself, correct? If so, I should be able to pick it up in nRF Connect too - no? Could it be that for some reason my H5100 don't send out this data? I already looked trough their app - I didn't really change anything.
> 

Correct. The Govee thermometers don't seem to store any data from their app locally. They broadcast data using BTLE that anyone nearby can passively receive. The devices can also be connected to and retrieve more data. Most of the devices store two to three weeks temperature/humidity in minute intervals. I'm not positive that the code correctly downloads the history data for the 5100 models. It's the most recent device I purchased and debugging the connected stuff hasn't been a very high priority. 

> 
> EDIT: I should probably mention that I didn't do this: `sudo setcap 'cap_net_raw,cap_net_admin+eip' ...`. As far as I understand, this won't be necessary using `root`.

The setcap command is only important if you want to run the program as a user other than root. Putting the BTLE stack into scanning mode requires privileges most users don't have. Root can do anything.

franzos, 2023-11-25:
> Correct. The Govee thermometers don't seem to store any data from their app locally. They broadcast data using BTLE that anyone nearby can passively receive. 

Excuse my confusion - I just realized that the data is transmitted via manufacturing data; I was looking for something else I guess. So I can confirm that my devices are all publishing the data, but for some reason the application on my Laptop doesn't pick it up.

What bluez version are you using?

```
bluetoothctl --version
bluetoothctl: 5.66
```

I will try this on another device when I get the chance.

wcbonner, 2023-11-25:
I don't know what is different. I've run into a problem on the Raspberry Pi Zero 2 W that doesn't let me put Bluetooth into scanning mode, but it reports the problem and exits, where your machine seems to accept the commands and not receive anything.

Do you only have a single bluetooth device in your linux machine? 

Can you update to the latest code and see if the hci reset commands that were added between 10/01 and 11/09 make the problems change? 

```
wim@WimPi4-Dev:~ $ ~/GoveeBTTempLogger/build/goveebttemplogger -v 3
[2023-11-25T21:01:29] GoveeBTTempLogger Version 2.20231109.2 Built on: Nov 25 2023 at 08:41:10
[                   ]      log: ""
[                   ]    cache: ""
[                   ]      svg: ""
[                   ]  battery: 0
[                   ]   minmax: 0
[                   ]  celsius: false
[                   ] titlemap: ""
[                   ]     time: 60
[                   ]  average: 5
[                   ] download: 0
[                   ]  passive: false
[                   ] no-bluetooth: false
[                   ] BlueToothDevice_ID: 0
[                   ] Reset device: hci0. Success(0)
[                   ] DOWN device: hci0. Success(0)
[                   ] UP device: hci0. Success(0)
[2023-11-25T21:01:29] LocalName: WimPi4-Dev
[2023-11-25T21:01:29] No BlueTooth Address Filter
[2023-11-25T21:01:29] Scanning Stopped.
[2023-11-25T21:01:29] Scanning Started. ScanInterval(11.25 msec) ScanWindow(11.25 msec) ScanType(1)
[2023-11-25T21:01:29] 44 [E3:60:59:21:80:65] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_8065
[2023-11-25T21:01:29] 44 [E3:60:59:23:14:7D] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_147D
[2023-11-25T21:01:29] 44 [E3:8E:C8:C1:98:9A] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_989A
[2023-11-25T21:01:29] 46 [C2:35:33:30:25:50] (bdaddr_type) 01 (evt_type) 00 (Flags) 06 (Name) GVH5100_2550 (UUID) 88EC (Manu) 010001010220CD3C (Temp) 13.9469°C (Humidity) 46.9% (Battery) 60% (GVH5100)
[2023-11-25T21:01:29] 42 [C2:35:33:30:25:50] (bdaddr_type) 01 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
[2023-11-25T21:01:29] 46 [A4:C1:38:37:BC:AE] (bdaddr_type) 00 (evt_type) 00 (Name) GVH5075_BCAE (UUID) 88EC (Flags) 05 (Manu) 88EC000247C21800 (Temp) 14.9°C (Humidity) 44.2% (Battery) 24% (GVH5075)
[2023-11-25T21:01:29] 42 [A4:C1:38:37:BC:AE] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FFC2 (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) C2
[2023-11-25T21:01:30] 44 [E3:5E:CC:21:5C:0F] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_5C0F
[2023-11-25T21:01:30] 26 [E3:5E:CC:21:5C:0F] (bdaddr_type) 00 (evt_type) 04 (Manu) 88EC001504F7224F02 (Temp) 10.45°C (Humidity) 89.51% (Battery) 79% (GVH5074)
[2023-11-25T21:01:30] 39 [01:BA:7D:69:98:55] (bdaddr_type) 01 (evt_type) 03 (Flags) 1A (Manu) 4C00090603FBA9FE89E7130712B08FBB000854 (Apple) 090603FBA9FE89E7130712B08FBB0008
[2023-11-25T21:01:30] 39 [01:BA:7D:69:98:55] (bdaddr_type) 01 (evt_type) 03 (Flags) 1A (Manu) 4C00090603FBA9FE89E7130712B08FBB000854 (Apple) 090603FBA9FE89E7130712B08FBB0008
[2023-11-25T21:01:30] 40 [A4:C1:38:05:C7:A1] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 88EC (Name) Govee_H5074_C7A1
[2023-11-25T21:01:30] 36 [4D:01:0F:59:FF:40] (bdaddr_type) 01 (evt_type) 00 (Flags) 1A (Tx Power) 0C (Manu) 4C000F059000312B4F10020004 (Apple) 0F059000312B4F100200
[2023-11-25T21:01:30] 44 [E3:60:59:21:80:65] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_8065
[2023-11-25T21:01:30] 26 [E3:60:59:21:80:65] (bdaddr_type) 00 (evt_type) 04 (Manu) 88EC009EFA6E0F5702 (Temp) -13.78°C (Humidity) 39.5% (Battery) 87% (GVH5074)
[2023-11-25T21:01:30] 44 [E3:60:59:23:14:7D] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_147D
[2023-11-25T21:01:30] 42 [E3:60:59:23:14:7D] (bdaddr_type) 00 (evt_type) 04 (Manu) 4C000215494E54454C4C495F524F434B535F485750747D14C2 (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5074 (Minor) 7D14 (RSSI) C2
^C***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
[2023-11-25T21:01:30] Scanning Stopped.
GoveeBTTempLogger Version 2.20231109.2 Built on: Nov 25 2023 at 08:41:10 (exiting)
```

```
wim@WimPi4-Dev:~ $ bluetoothctl
Agent registered
[bluetooth]# scan on
Discovery started
[CHG] Controller DC:A6:32:1C:B5:74 Discovering: yes
[NEW] Device E3:60:59:21:80:65 Govee_H5074_8065
[NEW] Device E3:60:59:23:14:7D Govee_H5074_147D
[NEW] Device E3:8E:C8:C1:98:9A Govee_H5074_989A
[NEW] Device A4:C1:38:05:C7:A1 Govee_H5074_C7A1
[NEW] Device E3:5E:CC:21:5C:0F Govee_H5074_5C0F
[NEW] Device C2:35:33:30:25:50 GVH5100_2550
[bluetooth]# scan off
Discovery stopped
[CHG] Device C2:35:33:30:25:50 RSSI is nil
[CHG] Device E3:5E:CC:21:5C:0F RSSI is nil
[CHG] Device A4:C1:38:05:C7:A1 RSSI is nil
[CHG] Device E3:8E:C8:C1:98:9A RSSI is nil
[CHG] Device E3:60:59:23:14:7D RSSI is nil
[... 11 lines cut ...]
```
wim@WimPi4-Dev:~ $ bluetoothctl --version
bluetoothctl: 5.55
wim@WimPi4-Dev:~ $ hcitool dev
Devices:
        hci0    DC:A6:32:1C:B5:74

wim@WimPi4-Dev:~ $ hciconfig -a
hci0:   Type: Primary  Bus: UART
        BD Address: DC:A6:32:1C:B5:74  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING
[... cut ...]
        TX bytes:9583 acl:0 sco:0 commands:492 errors:0
        Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH SNIFF
        Link mode: SLAVE ACCEPT
        Name: 'WimPi4-Dev'
        Class: 0x2c0000
        Service Classes: Rendering, Capturing, Audio
        Device Class: Miscellaneous,
        HCI Version: 5.0 (0x9)  Revision: 0x17e
        LMP Version: 5.0 (0x9)  Subversion: 0x6119
        Manufacturer: Cypress Semiconductor (305)
```

wcbonner, 2023-11-25:
@franzos I'm interested in knowing if specifying the controller after you've run a bluetooth scan using bluetoothctl makes any difference. 

ie
```
wim@WimPi4-Dev:~ $ bluetoothctl
Agent registered
[bluetooth]# scan on
Discovery started
[CHG] Controller DC:A6:32:1C:B5:74 Discovering: yes
[NEW] Device E3:60:59:21:80:65 Govee_H5074_8065
[NEW] Device E3:60:59:23:14:7D Govee_H5074_147D
[NEW] Device E3:8E:C8:C1:98:9A Govee_H5074_989A
[NEW] Device A4:C1:38:05:C7:A1 Govee_H5074_C7A1
[NEW] Device E3:5E:CC:21:5C:0F Govee_H5074_5C0F
[NEW] Device C2:35:33:30:25:50 GVH5100_2550
[bluetooth]# scan off
Discovery stopped
[CHG] Device C2:35:33:30:25:50 RSSI is nil
[CHG] Device E3:5E:CC:21:5C:0F RSSI is nil
[CHG] Device A4:C1:38:05:C7:A1 RSSI is nil
[CHG] Device E3:8E:C8:C1:98:9A RSSI is nil
[CHG] Device E3:60:59:23:14:7D RSSI is nil
[CHG] Device E3:60:59:21:80:65 RSSI is nil
[CHG] Controller DC:A6:32:1C:B5:74 Discovering: no
[DEL] Device E3:60:59:21:80:65 Govee_H5074_8065
[DEL] Device E3:60:59:23:14:7D Govee_H5074_147D
[DEL] Device E3:8E:C8:C1:98:9A Govee_H5074_989A
[DEL] Device A4:C1:38:05:C7:A1 Govee_H5074_C7A1
[DEL] Device E3:5E:CC:21:5C:0F Govee_H5074_5C0F
[DEL] Device C2:35:33:30:25:50 GVH5100_2550
[bluetooth]# exit
wim@WimPi4-Dev:~ $ ~/GoveeBTTempLogger/build/goveebttemplogger -v 3 --controller DC:A6:32:1C:B5:74
[2023-11-25T21:31:44] GoveeBTTempLogger Version 2.20231109.2 Built on: Nov 25 2023 at 08:41:10
[                   ]      log: ""
[                   ]    cache: ""
[                   ]      svg: ""
[                   ]  battery: 0
[                   ]   minmax: 0
[                   ]  celsius: false
[                   ] titlemap: ""
[                   ]     time: 60
[                   ]  average: 5
[                   ] download: 0
[                   ]  passive: false
[                   ] no-bluetooth: false
[                   ] BlueToothDevice_ID: 0
[                   ] Reset device: hci0. Success(0)
[                   ] DOWN device: hci0. Success(0)
[                   ] UP device: hci0. Success(0)
[2023-11-25T21:31:44] Controller Address: DC:A6:32:1C:B5:74
[2023-11-25T21:31:44] LocalName: WimPi4-Dev
[2023-11-25T21:31:44] No BlueTooth Address Filter
[2023-11-25T21:31:44] Scanning Stopped.
[2023-11-25T21:31:44] Scanning Started. ScanInterval(11.25 msec) ScanWindow(11.25 msec) ScanType(1)
[2023-11-25T21:31:44] 44 [E3:5E:CC:21:5C:0F] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_5C0F
[2023-11-25T21:31:45] 44 [E3:60:59:23:14:7D] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_147D
[2023-11-25T21:31:45] 44 [E3:5E:CC:21:5C:0F] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_5C0F
[2023-11-25T21:31:45] 46 [C2:35:33:30:25:50] (bdaddr_type) 01 (evt_type) 00 (Flags) 06 (Name) GVH5100_2550 (UUID) 88EC (Manu) 0100010102289E39 (Temp) 14.147°C (Humidity) 47% (Battery) 57% (GVH5100)
[2023-11-25T21:31:46] 40 [A4:C1:38:D5:A3:3B] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 88EC (Name) Govee_H5074_A33B
[2023-11-25T21:31:46] 44 [E3:60:59:23:14:7D] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_147D
[2023-11-25T21:31:46] 44 [E3:60:59:21:80:65] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_8065
[2023-11-25T21:31:46] 44 [E3:5E:CC:21:5C:0F] (bdaddr_type) 00 (evt_type) 00 (Flags) 06 (UUID) 0A18F5FE88EC (Name) Govee_H5074_5C0F
^C***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
[2023-11-25T21:31:46] Scanning Stopped.
GoveeBTTempLogger Version 2.20231109.2 Built on: Nov 25 2023 at 08:41:10 (exiting)
```

franzos, 2023-11-25:
> Do you only have a single bluetooth device in your linux machine?

Yes, there's only one.

Looks like your bluez version is pretty close too.

> Can you update to the latest code and see if the hci reset commands that were added between 10/01 and 11/09 make the problems change?

Sure, I tried it just now. I use Bluetooth mouse and keyboard - both were disconnected when I ran the app:

```bash
goveebttemplogger -v 9 -p
[2023-11-25T22:31:22] GoveeBTTempLogger Version 2.20231109.2 Built on: Jan  1 1970 at 00:00:01
[                   ]      log: ""
[                   ]    cache: ""
[                   ]      svg: ""
[                   ]  battery: 0
[                   ]   minmax: 0
[                   ]  celsius: false
[                   ] titlemap: ""
[                   ]     time: 60
[                   ]  average: 5
[                   ] download: 0
[                   ]  passive: true
[                   ] no-bluetooth: false
[                   ] BlueToothDevice_ID: 0
[                   ] Reset device: hci0. Success(0)
[                   ] DOWN device: hci0. Success(0)
[                   ] UP device: hci0. Success(0)
[2023-11-25T22:31:22] LocalName: BlueZ
[2023-11-25T22:31:22] No BlueTooth Address Filter
[2023-11-25T22:31:22] Scanning Stopped.
[2023-11-25T22:31:22] Scanning Started. ScanInterval(11.25 msec) ScanWindow(11.25 msec) ScanType(0)
[2023-11-25T22:31:22] Read: 60 Bytes
[-------------------] 04 3E 39 0D 01 13 00 01 2A 6F 35 34 35 C8 01 00 FF 7F C9 00 00 00 00 00 00 00 00 00 1F 02 01 06 0D 09 47 56 48 35 31 30 30 5F 36 46 32 41 03 03 88 EC 09 FF 01 00 01 01 03 03 F3 58
[2023-11-25T22:31:22] No recent Bluetooth LE Advertisments! (> 3 Minutes)
[2023-11-25T22:31:22] Scanning Stopped.
[2023-11-25T22:31:22] Scanning Started. ScanInterval(5000 msec) ScanWindow(500 msec) ScanType(0)
[2023-11-25T22:31:27] Read: 60 Bytes
[-------------------] 04 3E 39 0D 01 13 00 01 1B 3C 35 34 35 CC 01 00 FF 7F C3 00 00 00 00 00 00 00 00 00 1F 02 01 06 0D 09 47 56 48 35 31 30 30 5F 33 43 31 42 03 03 88 EC 09 FF 01 00 01 01 03 0B A5 5D
[2023-11-25T22:31:27] No recent Bluetooth LE Advertisments! (> 3 Minutes)
[2023-11-25T22:31:27] Scanning Stopped.
[2023-11-25T22:31:27] Scanning Started. ScanInterval(5000 msec) ScanWindow(5000 msec) ScanType(0)
[2023-11-25T22:31:28] Read: 60 Bytes
[-------------------] 04 3E 39 0D 01 13 00 01 2A 6F 35 34 35 C8 01 00 FF 7F CA 00 00 00 00 00 00 00 00 00 1F 02 01 06 0D 09 47 56 48 35 31 30 30 5F 36 46 32 41 03 03 88 EC 09 FF 01 00 01 01 03 03 F3 58
[2023-11-25T22:31:28] No recent Bluetooth LE Advertisments! (> 3 Minutes)
[2023-11-25T22:31:28] Scanning Stopped.
[2023-11-25T22:31:28] Scanning Started. ScanInterval(5000 msec) ScanWindow(2000 msec) ScanType(0)
[2023-11-25T22:31:29] Read: 60 Bytes
[-------------------] 04 3E 39 0D 01 13 00 01 3B 16 35 34 35 C8 01 00 FF 7F A2 00 00 00 00 00 00 00 00 00 1F 02 01 06 0D 09 47 56 48 35 31 30 30 5F 31 36 33 42 03 03 88 EC 09 FF 01 00 01 01 03 55 AE 59
[2023-11-25T22:31:29] No recent Bluetooth LE Advertisments! (> 3 Minutes)
[2023-11-25T22:31:29] Scanning Stopped.
[2023-11-25T22:31:29] Scanning Started. ScanInterval(40 msec) ScanWindow(30 msec) ScanType(0)
[2023-11-25T22:31:29] Read: 44 Bytes
...
# no notable changes
```

> I'm interested in knowing if specifying the controller after you've run a bluetooth scan using bluetoothctl makes any difference.

I tried that too, before and after the update but nothing notable

```
goveebttemplogger -v 3 --controller 94:E6:F7:0A:43:F7
[2023-11-25T22:37:57] GoveeBTTempLogger Version 2.20231109.2 Built on: Jan  1 1970 at 00:00:01
[                   ]      log: ""
[                   ]    cache: ""
[                   ]      svg: ""
[                   ]  battery: 0
[                   ]   minmax: 0
[... 21 lines cut ...]
[2023-11-25T22:38:03] No recent Bluetooth LE Advertisments! (> 3 Minutes)
[2023-11-25T22:38:03] Scanning Stopped.
[2023-11-25T22:38:03] Scanning Started. ScanInterval(5000 msec) ScanWindow(5000 msec) ScanType(1)
[-------------------] 04 3E 39 0D 01 13 00 01 1B 3C 35 34 35 CC 01 00 FF 7F C0 00 00 00 00 00 00 00 00 00 1F 02 01 06 0D 09 47 56 48 35 31 30 30 5F 33 43 31 42 03 03 88 EC 09 FF 01 00 01 01 03 07 C0 5A
[2023-11-25T22:38:04] No recent Bluetooth LE Advertisments! (> 3 Minutes)
[2023-11-25T22:38:04] Scanning Stopped.
[2023-11-25T22:38:04] Scanning Started. ScanInterval(5000 msec) ScanWindow(2000 msec) ScanType(1)
[-------------------] 04 3E 2C 0D 01 13 00 01 6B 7D 65 E2 29 7D 01 00 FF 7F A2 00 00 00 00 00 00 00 00 00 12 02 01 1A 02 0A 0C 0B FF 4C 00 10 06 4F 1D 09 15 EA 08
[2023-11-25T22:38:09] No recent Bluetooth LE Advertisments! (> 3 Minutes)
[2023-11-25T22:38:09] Scanning Stopped.
[2023-11-25T22:38:09] Scanning Started. ScanInterval(40 msec) ScanWindow(30 msec) ScanType(1)
[-------------------] 04 3E 29 0D 01 10 00 01 3D BF 15 5F 0F 18 01 00 FF 7F A1 00 00 00 00 00 00 00 00 00 0F 02 01 1A 0B FF 4C 00 09 06 03 C6 C0 A8 01 42
[2023-11-25T22:38:09] No recent Bluetooth LE Advertisments! (> 3 Minutes)
[2023-11-25T22:38:09] Scanning Stopped.
[2023-11-25T22:38:09] Scanning Started. ScanInterval(60 msec) ScanWindow(30 msec) ScanType(1)
[-------------------] 04 3E 2C 0D 01 13 00 01 6B 7D 65 E2 29 7D 01 00 FF 7F A3 00 00 00 00 00 00 00 00 00 12 02 01 1A 02 0A 0C 0B FF 4C 00 10 06 4F 1D 09 15 EA 08
[2023-11-25T22:38:10] No recent Bluetooth LE Advertisments! (> 3 Minutes)
[2023-11-25T22:38:10] Scanning Stopped.
[... cut ...]
[-------------------] 04 3E 29 0D 01 10 00 01 3D BF 15 5F 0F 18 01 00 FF 7F A2 00 00 00 00 00 00 00 00 00 0F 02 01 1A 0B FF 4C 00 09 06 03 C6 C0 A8 01 42
[2023-11-25T22:38:10] No recent Bluetooth LE Advertisments! (> 3 Minutes)
[2023-11-25T22:38:10] Scanning Stopped.
[2023-11-25T22:38:10] Scanning Started. ScanInterval(5000 msec) ScanWindow(500 msec) ScanType(1)
[-------------------] 04 3E 39 0D 01 13 00 01 2A 6F 35 34 35 C8 01 00 FF 7F C9 00 00 00 00 00 00 00 00 00 1F 02 01 06 0D 09 47 56 48 35 31 30 30 5F 36 46 32 41 03 03 88 EC 09 FF 01 00 01 01 03 03 EE 59
```

wcbonner, 2023-11-29:
@franzos I just produced a fix for https://github.com/wcbonner/GoveeBTTempLogger/issues/50 that added a function to the initialization of the Bluetooth command sequence. 

https://github.com/wcbonner/GoveeBTTempLogger/commit/537e6211edbb5524dfd15846aa0c817df3566e7a shows the update. 

If you have time to test the newest code, I'd appreciate knowing if it makes any difference on your platform.

franzos, 2023-11-30:
That's awesome, thanks! Unfortunately no luck yet:

```
[2023-11-30T10:52:25] GoveeBTTempLogger Version 2.20231129.0 Built on: Jan  1 1970 at 00:00:01
[                   ] BlueToothDevice_ID: 0
[                   ] Reset device: hci0. Success(0)
[                   ] DOWN device: hci0. Success(0)
[                   ] UP device: hci0. Success(0)
[2023-11-30T10:52:26] LocalName: BlueZ
[2023-11-30T10:52:26] No BlueTooth Address Filter
[2023-11-30T10:52:26] Scanning Stopped.
[2023-11-30T10:52:26] Scanning Started. ScanInterval(11.25 msec) ScanWindow(11.25 msec) ScanType(1)
[2023-11-30T10:52:27] No recent Bluetooth LE Advertisments! (> 3 Minutes)
[2023-11-30T10:52:27] Scanning Stopped.
[2023-11-30T10:52:27] Scanning Started. ScanInterval(5000 msec) ScanWindow(500 msec) ScanType(1)
[2023-11-30T10:52:32] No recent Bluetooth LE Advertisments! (> 3 Minutes)
[2023-11-30T10:52:32] Scanning Stopped.
[2023-11-30T10:52:32] Scanning Started. ScanInterval(5000 msec) ScanWindow(5000 msec) ScanType(1)
[2023-11-30T10:52:33] No recent Bluetooth LE Advertisments! (> 3 Minutes)
[2023-11-30T10:52:33] Scanning Stopped.
[2023-11-30T10:52:33] Scanning Started. ScanInterval(5000 msec) ScanWindow(2000 msec) ScanType(1)
[2023-11-30T10:52:35] No recent Bluetooth LE Advertisments! (> 3 Minutes)
[2023-11-30T10:52:35] Scanning Stopped.
```

I also collected some dmesg logs; They seem relevant:

```
[65602.901215] Bluetooth: hci0: ACL packet for unknown connection handle 3585
[65602.901243] Bluetooth: hci0: ACL packet for unknown connection handle 3585
[65602.901250] Bluetooth: hci0: ACL packet for unknown connection handle 3585
[65602.901255] Bluetooth: hci0: ACL packet for unknown connection handle 3585
[65602.913266] Bluetooth: hci0: ACL packet for unknown connection handle 3585
[65602.929025] Bluetooth: hci0: ACL packet for unknown connection handle 3585
[65611.267605] filter_write: 191 callbacks suppressed
```

This also still kills all connections (Bluetooth keyboard and mouse stop working, until I restart the Bluetooth service). Maybe that's why it won't find anything (because it just killed Bluetooth)? Could you point me to where exactly you first engage Bluetooth? Maybe I can assemble a minimal reproduction.

wcbonner, 2023-11-30:
I'm not certain about multiple uses of Bluetooth at the same time. I open a socket to the Bluetooth controller with the command ```int BlueToothDevice_Handle = hci_open_dev(BlueToothDevice_ID);``` and maintain exclusive access to that socket until I close it with the command ```hci_close_dev(BlueToothDevice_Handle);```. This is all in the main loop of the program, around line [3230](https://github.com/wcbonner/GoveeBTTempLogger/blob/master/goveebttemplogger.cpp#L3227). 

Can you do a packet capture using either wireshark or tcpdump? (I've been looking at the data in Wireshark, but it was easier to quickly install tcpdump on my headless machine to do a quick capture.)

```
wim@WimPiZero2W:~ $ tcpdump --list-interfaces
1.wlan0 [Up, Running, Wireless]
2.any (Pseudo-device that captures on all interfaces) [Up, Running]
3.lo [Up, Running, Loopback]
4.bluetooth0 (Bluetooth adapter number 0) [Wireless, Association status unknown]
5.bluetooth-monitor (Bluetooth Linux Monitor) [Wireless]
6.nflog (Linux netfilter log (NFLOG) interface) [none]
7.nfqueue (Linux netfilter queue (NFQUEUE) interface) [none]
8.dbus-system (D-Bus system bus) [none]
9.dbus-session (D-Bus session bus) [none]
```
I had to run tcpdump with the sudo command, otherwise it only recorded the packets I was sending to the controller.
```
wim@WimPiZero2W:~ $ sudo tcpdump --interface=bluetooth0 --direction=inout -w bluetooth0_capture_11_30-goveebttemplogger-inout.pcap
tcpdump: listening on bluetooth0, link-type BLUETOOTH_HCI_H4_WITH_PHDR (Bluetooth HCI UART transport layer plus pseudo-header), snapshot length 262144 bytes
^C159 packets captured
1680 packets received by filter
0 packets dropped by kernel
```
In another window I ran the program:
```
wim@WimPiZero2W:~ $ ~/GoveeBTTempLogger/build/goveebttemplogger
[2023-11-30T22:01:30] GoveeBTTempLogger Version 2.20231129.0 Built on: Nov 29 2023 at 15:26:50
[                   ] BlueToothDevice_ID: 0
[                   ] Reset device: hci0. Success(0)
[                   ] DOWN device: hci0. Success(0)
[                   ] UP device: hci0. Success(0)
[2023-11-30T22:01:30] LocalName: WimPiZero2W
[2023-11-30T22:01:30] No BlueTooth Address Filter
[2023-11-30T22:01:30] Scanning Stopped.
[2023-11-30T22:01:30] Scanning Started. ScanInterval(11.25 msec) ScanWindow(11.25 msec) ScanType(1)
[2023-11-30T22:01:41] 26 [E3:5E:CC:21:5C:0F] (Temp) 8.71°C (Humidity) 91.96% (Battery) 78% (GVH5074)
[2023-11-30T22:01:41] 46 [C2:35:33:30:25:50] (Temp) 15.7503°C (Humidity) 50.3% (Battery) 57% (GVH5100)
[2023-11-30T22:01:41] 42 [C2:35:33:30:25:50] (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
[2023-11-30T22:01:41] 26 [E3:8E:C8:C1:98:9A] (Temp) 16.24°C (Humidity) 48.76% (Battery) 100% (GVH5074)
[2023-11-30T22:01:42] 26 [E3:60:59:23:14:7D] (Temp) 2.11°C (Humidity) 47.1% (Battery) 95% (GVH5074)
[2023-11-30T22:01:42] 46 [A4:C1:38:DC:CC:3D] (Temp) 15.5528°C (Humidity) 52.8% (Battery) 84% (GVH5174)
^C***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
[2023-11-30T22:01:42] Scanning Stopped.
GoveeBTTempLogger Version 2.20231129.0 Built on: Nov 29 2023 at 15:26:50 (exiting)
```
After stopping goveebttemplogger with Ctrl-C and then tcpdump with Ctrl-C I was able to load the resulting capture file in Wireshark and see all of the commands that went back and forth to the Bluetooth controller.

wcbonner, 2023-12-02:
> Thank you, I understand better now. I had a look at `hci_open_dev` docs too. How about subscribing to bluez on dbus to listen to events / make connections? This would allow other devices and apps to remain connected / might simplify handling on your end.

That would require a complete rewrite of my Bluetooth code. I wouldn't mind doing that if I were able to come across good documentation and examples of how to do what I'm already doing, but in my experience with this project, BlueZ is extremely poorly documented in both the older interface that I'm using, and the newer interface based on dbus. 

> Edit; Interesting:
> 
> ```
> 1.wlp0s20f3 [Up, Running, Wireless, Associated]
> 2.nebula1 [Up, Running, Connected]
> 3.any (Pseudo-device that captures on all interfaces) [Up, Running]
> 4.lo [Up, Running, Loopback]
> 5.enp0s31f6 [Up, Disconnected]
> 6.docker0 [Up, Disconnected]
> 7.br-6f9b53fcac58 [Up, Disconnected]
> 8.br-b69bd1c0505a [Up, Disconnected]
> 9.br-d400c369a9c4 [Up, Disconnected]
> 10.nflog (Linux netfilter log (NFLOG) interface) [none]
> 11.nfqueue (Linux netfilter queue (NFQUEUE) interface) [none]
> ```

I'm assuming that this was the output of a ```tcpdump --list-interfaces``` command? Interesting that it doesn't list an obvious Bluetooth device.
 
> ```
> lsusb
> Bus 001 Device 005: ID 8087:0aaa Intel Corp. Bluetooth 9460/9560 Jefferson Peak (JfP)
> ```

Here's that output on my machine for comparison:
```
wim@WimPiZero2W:~ $ sudo lsmod | grep bt
btbcm                  24576  1 hci_uart
bluetooth             585728  26 hci_uart,btbcm,bnep

wim@WimPiZero2W:~ $ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

wim@WimPiZero2W:~ $ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
```

franzos, 2023-12-02:
> BlueZ is extremely poorly documented in both the older interface that I'm using, and the newer interface based on dbus.

I guess most people rely on libraries, so this is not as much of a problem; We've recently used BlueZQt; It provides stuff like `BluezQt::Device::serviceDataChanged` to listen to BLE data. Of course it also adds some weight, and potential for breakage.

> I'm assuming that this was the output of a tcpdump --list-interfaces command? Interesting that it doesn't list an obvious Bluetooth device.

Yes that's correct; I also found that interesting. I also tried `sudo tcpdump -i any` but it doesn't include bluetooth either. Here's my current version:

```
$ tcpdump --version
tcpdump version 4.99.4
libpcap version 1.10.1 (with TPACKET_V3)
OpenSSL 3.0.8 7 Feb 2023
```

```bash
lsusb
Bus 001 Device 005: ID 8087:0aaa Intel Corp. Bluetooth 9460/9560 Jefferson Peak (JfP)

lsusb -t                                                                                                                                                                                                                                                                                  
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M                                                                                                                                                                              
    |__ Port 5: Dev 3, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 5: Dev 6, If 0, Class=, Driver=, 480M
```

```
dmesg | grep bt
[    0.017555] ACPI: SSDT 0x000000006DC28000 001983 (v02 LENOVO TbtTypeC 00000000 INTL 20160527)
[   14.227225] usbcore: registered new interface driver btusb
[   14.241516] Bluetooth: hci0: Found device firmware: intel/ibt-17-16-1.sfi
[   16.399037] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-17-16-1.ddc
``

I did find some more logs:

```
Dec  2 11:13:10 localhost bluetoothd[14076]: profiles/audio/vcp.c:vcp_init() D-Bus experimental not enabled
Dec  2 11:13:10 localhost bluetoothd[14076]: src/plugin.c:plugin_init() Failed to init vcp plugin
Dec  2 11:13:10 localhost bluetoothd[14076]: profiles/audio/mcp.c:mcp_init() D-Bus experimental not enabled
Dec  2 11:13:10 localhost bluetoothd[14076]: src/plugin.c:plugin_init() Failed to init mcp plugin
Dec  2 11:13:10 localhost bluetoothd[14076]: profiles/audio/bap.c:bap_init() D-Bus experimental not enabled
Dec  2 11:13:10 localhost bluetoothd[14076]: src/plugin.c:plugin_init() Failed to init bap plugin
Dec  2 11:13:10 localhost bluetoothd[14076]: Bluetooth management interface 1.22 initialized
```

but this appears to be related to `/etc/bluetooth/main.conf` -> `Experimental = false` which I guess is unrelated / not required. These logs are not related to running GoveeBTTempLogger.

Here's my whole bluetooth config:

```
[General]
Name = BlueZ
Class = 0x000000
DiscoverableTimeout = 180
AlwaysPairable = false

PairableTimeout = 0
ReverseServiceDiscovery = true

NameResolving = true
 
DebugKeys = false
 
ControllerMode = dual
MultiProfile = off
FastConnectable = false
 
Privacy = off
JustWorksRepairing = never
TemporaryTimeout = 30
RefreshDiscovery = true
 
Experimental = false

RemoteNameRequestRetryDelay = 300
[BR]
[LE]
AdvMonAllowlistScanDuration = 300
AdvMonNoFilterScanDuration = 500
EnableAdvMonInterleaveScan = 1
[GATT]
Cache = always
KeySize = 0
ExchangeMTU = 517
Channels = 3
[AVDTP]
SessionMode = basic
StreamMode = basic
[Policy]
ReconnectAttempts = 7
ReconnectIntervals = 1,2,4,8,16,32,64
AutoEnable = true
 
ResumeDelay = 2
[AdvMon]
RSSISamplingPeriod = 0xff
```

wcbonner, 2024-08-16:
@franzos I just made a new release that defaults to getting data from BlueZ over DBus and should properly recognize H5100 series devices. https://github.com/wcbonner/GoveeBTTempLogger/releases/tag/v3.20240816.0

Because I'm now using DBus as my interface, I believe that it should coexist with Bluetooth keyboards and mice but have not had a chance to verify it.

closed 2024-08-22 by wcbonner