wcbonner/GoveeBTTempLogger #58: Feature Request: Push Values on to an MQTT Queue

yonz2, 2024-02-20:
Hi,

I'm working on a remote data logger, that logs data from my Govee sensors. I would like have the RPI put the values read at a certaininterval on a MQTT queue.

I'm not a CPP developer, so I can't implement it myself and submit a pull request for a patch.

I did look into it, though, and using the [Eclipse Paho Library](https://github.com/eclipse/paho.mqtt.cpp) and a few new parameters this should be a straight forward thing to do...

### Parameters
```
[MQTT]
MQTT_URI=myserver.athome.com  
MQTT_PORT=1883 
MQTT_Username=myMQTTuser
MQTT_Password=myMQTTpassword
MQTT_TOPIC_PREFIX=goveebttemplogger
MQTT_INTERVAL=300
```

The routine should then every `MQTT_INTERVAL` seconds send the latest reading from each device as a JSON payload to the MQTT Topic:  ` MQTT_TOPIC_PREFIX/<<Mac Address of Device>>` 

### JSON Payload
```json
{
  "device":"<< Mac Address of Decive>>",
  "device_name":"<<Name of Device, if known>>",
  "temperature": 23.20,
  "humidity": 43.70,
  "battery": 100
}
```

On the other end we can then use e.g. [Home Assistant's MQTT Sensor](https://www.home-assistant.io/integrations/sensor.mqtt/) to process the data            

### Example Home Assistant configuration.yaml entry
```yaml
mqtt:
  sensor:
    - name: "Temperature"
      state_topic: "goveebttemplogger/gvh-A4C13876CCXX"
      suggested_display_precision: 1
      unit_of_measurement: "°C"
      value_template: "{{ value_json.temperature }}
    - name: "Humidity"
      state_topic: "goveebttemplogger/gvh-A4C13876CCXX"
      unit_of_measurement: "%"
      value_template: "{{ value_json.humidity }}"
    - name: "Battery"
      state_topic: "goveebttemplogger/gvh-A4C13876CCXX"
      unit_of_measurement: "%"
      value_template: "{{ value_json.battery }}"
```

Just a thought.....

Best regards,
Yonz

wcbonner, 2024-02-20:
I don't see myself adding this to the core program. One of my design principles has been to limit the dependencies as much as possible. 

I believe I'm currently only dependent on c++ std libraries and libbluetooth. 

You could probably achieve what you want by monitoring the log directory for updates, then using a stream text editor (sed?) to manipulate the logged data and submit it to MQTT. E.g. "tail -f /var/log/govee/gvh-xxxxxx.txt" 

I just thought of the ```tail -f``` command as an example of monitoring a text file for updates. I'm sure there are better methods in a language such as Perl that are designed to handle text better than C/C++.

yonz2, 2024-02-27:
Hi,

I see your point. It does make sense to keep the code clean and simple.

I created a script to do exactly what you suggested, with the addition of dynamically create the HA Devices when the [GoveeBTTempLogger](https://github.com/wcbonner/GoveeBTTempLogger) encounters a new device (writes to a new log file)

My script is here: [Datalogger](https://github.com/yonz2/datalogger)

closed 2024-02-27 by yonz2