wcbonner/GoveeBTTempLogger #59: Support for H5101

wcbonner, 2024-03-05:
https://a.co/d/cAmWKGc

I saw that there's a new H5101 device that uses the CR2450 battery with an lcd display on Amazon. 

I'm guessing that its Bluetooth encoding is compatible with the other 510x devices. I don't feel like buying more thermometers right now. If someone running this software can confirm the results, I'll update the software to report the devices properly.

SmurfSlayer, 2024-03-05:
just ordered a pair of 'em, I'll let you know when they are online (should be here toward end of week)

wcbonner, 2024-03-12:
> just ordered a pair of 'em, I'll let you know when they are online (should be here toward end of week)

This is kind of cool. Looks like support should be extremely easy as it's already recognizing the thermometer, just not declaring it's type.

```
wim@FlightTracker-Wim:~/GoveeBTTempLogger $ ~/GoveeBTTempLogger/build/goveebttemplogger --only C4:35:33:30:75:6F
[2024-03-12T21:32:48] GoveeBTTempLogger Version 2.20240216.0 Built on: Feb 20 2024 at 10:35:41
[                   ] BlueToothDevice_ID: 0
[                   ] Reset device: hci0. Success(0)
[                   ] DOWN device: hci0. Success(0)
[                   ] UP device: hci0. Success(0)
[2024-03-12T21:32:48] LocalName: FlightTracker-Wim
[2024-03-12T21:32:48] BlueTooth Address Filter: [C4:35:33:30:75:6F]
[2024-03-12T21:32:48] Scanning Stopped.
[2024-03-12T21:32:48] Scanning Started. ScanInterval(11.25 msec) ScanWindow(11.25 msec) ScanType(1)
[2024-03-12T21:32:49] 46 [C4:35:33:30:75:6F] (Temp) 19.1376°C (Humidity) 37.6% (Battery) 79% (ThermometerType::Unknown)
[2024-03-12T21:32:49] 42 [C4:35:33:30:75:6F] (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
[2024-03-12T21:32:51] 46 [C4:35:33:30:75:6F] (Flags) 06 (Name) GVH5101_756F (UUID) 88EC (Manu) 0100010102EB904F (Temp) 19.1376°C (Humidity) 37.6% (Battery) 79% (ThermometerType::Unknown)
[2024-03-12T21:32:55] 46 [C4:35:33:30:75:6F] (Flags) 06 (Name) GVH5101_756F (UUID) 88EC (Manu) 0100010102EB914F (Temp) 19.1377°C (Humidity) 37.7% (Battery) 79% (ThermometerType::Unknown)
[2024-03-12T21:32:57] 46 [C4:35:33:30:75:6F] (Flags) 06 (Name) GVH5101_756F (UUID) 88EC (Manu) 0100010102EB904F (Temp) 19.1376°C (Humidity) 37.6% (Battery) 79% (ThermometerType::Unknown)
[2024-03-12T21:32:59] 46 [C4:35:33:30:75:6F] (Flags) 06 (Name) GVH5101_756F (UUID) 88EC (Manu) 0100010102EB914F (Temp) 19.1377°C (Humidity) 37.7% (Battery) 79% (ThermometerType::Unknown)
[2024-03-12T21:33:01] 46 [C4:35:33:30:75:6F] (Flags) 06 (Name) GVH5101_756F (UUID) 88EC (Manu) 0100010102EB904F (Temp) 19.1376°C (Humidity) 37.6% (Battery) 79% (ThermometerType::Unknown)
^C***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
[2024-03-12T21:33:02] Scanning Stopped.
GoveeBTTempLogger Version 2.20240216.0 Built on: Feb 20 2024 at 10:35:41 (exiting)
wim@FlightTracker-Wim:~/GoveeBTTempLogger $
```

wcbonner, 2024-03-12:
@SmurfSlayer it looks like https://github.com/wcbonner/GoveeBTTempLogger/commit/f2486aae23003e0ffed500f5f020f078926722d9 was all that was required to properly support the advertisement model of the H5101. 

```
wim@FlightTracker-Wim:~/GoveeBTTempLogger $ ~/GoveeBTTempLogger/build/goveebttemplogger --only C4:35:33:30:75:6F
[2024-03-12T21:58:18] GoveeBTTempLogger Version 2.20240312.0 Built on: Mar 12 2024 at 14:57:30
[                   ] BlueToothDevice_ID: 0
[                   ] Reset device: hci0. Success(0)
[                   ] DOWN device: hci0. Success(0)
[                   ] UP device: hci0. Success(0)
[2024-03-12T21:58:18] LocalName: FlightTracker-Wim
[2024-03-12T21:58:18] BlueTooth Address Filter: [C4:35:33:30:75:6F]
[2024-03-12T21:58:18] Scanning Stopped.
[2024-03-12T21:58:18] Scanning Started. ScanInterval(11.25 msec) ScanWindow(11.25 msec) ScanType(1)
[2024-03-12T21:58:19] 46 [C4:35:33:30:75:6F] (Temp) 19.1377°C (Humidity) 37.7% (Battery) 80% (GVH5101)
[2024-03-12T21:58:19] 42 [C4:35:33:30:75:6F] (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
[2024-03-12T21:58:21] 46 [C4:35:33:30:75:6F] (Flags) 06 (Name) GVH5101_756F (UUID) 88EC (Manu) 0100010102EB9151 (Temp) 19.1377°C (Humidity) 37.7% (Battery) 81% (GVH5101)
[2024-03-12T21:58:23] 46 [C4:35:33:30:75:6F] (Flags) 06 (Name) GVH5101_756F (UUID) 88EC (Manu) 0100010102EB9151 (Temp) 19.1377°C (Humidity) 37.7% (Battery) 81% (GVH5101)
[2024-03-12T21:58:23] 42 [C4:35:33:30:75:6F] (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
[2024-03-12T21:58:25] 46 [C4:35:33:30:75:6F] (Flags) 06 (Name) GVH5101_756F (UUID) 88EC (Manu) 0100010102EB9051 (Temp) 19.1376°C (Humidity) 37.6% (Battery) 81% (GVH5101)
[2024-03-12T21:58:25] 42 [C4:35:33:30:75:6F] (Manu) 4C000215494E54454C4C495F524F434B535F48575075F2FF0C (Apple) (UUID) 494E54454C4C495F524F434B535F4857 (Major) 5075 (Minor) F2FF (RSSI) 0C
^C***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
[2024-03-12T21:58:26] Scanning Stopped.
GoveeBTTempLogger Version 2.20240312.0 Built on: Mar 12 2024 at 14:57:30 (exiting)
```

closed 2024-03-12 by wcbonner