wcbonner/GoveeBTTempLogger #60: Go version?

lordofscripts, 2024-05-21:
I am looking for something like this but written in GO. Not necessarily all the functionality but just a library that fetches the data from the thermo/hygrometers to store as CSV (no plotting).

wcbonner, 2024-05-22:
@lordofscripts I don't know GO so can't really suggest a solution for you. 

I believe that some people are using this program to receive the Bluetooth messages and then process the data without using the graphing or logging facilities.

closed 2024-05-22 by wcbonner

lordofscripts, 2024-05-23:
Thanks for the prompt answer @wcbonner , I needed it in GO to integrate it with my GO application. I got my hands full with GO pet projects at the moment, I haven't used C/C++ in like 15 years :O since I bumped into C#.

GO is actually fun and not that difficult to learn if you know C-family languages. I actually learned Go just last year out of sheer curiosity. I thought it was some "new" language and it turned out it had been around for over 10 years, LOL.

Anyway, thanks and May the Govee Be With You :) I am still waiting for GOVEE to enhance their firmware with Local URL callbacks. It would open a whole new world for us GOVEE fans with tech knowledge.