wcbonner/GoveeBTTempLogger #61: SSH terminal session lags when GoveeBTTempLogger is running as a service on Pi 4

cnighswonger, 2024-06-20:
`uname -a`
Linux raspberrypi 6.6.33-v7+ #1770 SMP Wed Jun 12 17:44:10 BST 2024 armv7l GNU/Linux

`lsb_release -a`
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

`cat /proc/cpuinfo`
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
...
Hardware        : BCM2835
Revision        : a02082
Serial          : 00000000c5e47ad8
Model           : Raspberry Pi 3 Model B Rev 1.2

`uptime`
20:52:58 up 1 day, 18 min,  2 users,  load average: 0.02, 0.02, 0.00

GoveeBTTempLogger Version 2.20231201.0 Built on: Dec  1 2023 at 17:40:24

root      4081     1  0 07:03 ?        00:01:05 /usr/local/bin/goveebttemplogger --verbose 0 --log /var/log/goveebttemplogger --time 60 --svg /var/www/html/goveebttemplogger --battery 8 --minmax 8 --cache /var/cache/goveebttemplogger --download

When connected to this Pi via Putty SSH terminal, typed commands lag by several seconds. Stopping the goveebttemplogger process resolves this issue.

Has anyone else experienced this issue or could it just be local to this installation?

Thanks,
Chris

wcbonner, 2024-06-20:
> When connected to this Pi via Putty SSH terminal, typed commands lag by several seconds. Stopping the goveebttemplogger process resolves this issue.
> 
> Has anyone else experienced this issue or could it just be local to this installation?
> 

@cnighswonger I've never heard of this behavior before. From looking at the details you submitted it looks like the version you are running was code from last December and the current version is much more recent, but the network and Bluetooth code haven't changed significantly in that timeframe. Still, I might suggest pulling and compiling the latest code to see if that makes any difference.

My local code is on a Pi4 running 64bit PiOS, and I usually ssh in with the default Windows 11 ssh client. I can't see a way the ssh client itself would make a difference.

wcbonner, 2024-06-20:
I don't think I have anything running bullseye anymore. The machines I had easy access to I've updated to bookworm, and the remote machines have just been in maintenance mode. 

details from my Pi4:
wim@WimPi4-Sola:~ $ **uname -a**
`Linux WimPi4-Sola 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
`
wim@WimPi4-Sola:~ $ **lsb_release -a**
```
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
```
wim@WimPi4-Sola:~ $ cat /proc/cpuinfo
```
processor       : 0
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

...

Revision        : d03114
Serial          : 10000000270a58c5
Model           : Raspberry Pi 4 Model B Rev 1.4
```

details from a Pi3 running older OS:
wim@FlightTracker:~ $ **uname -a**
`Linux FlightTracker 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux
`
wim@FlightTracker:~ $ **lsb_release -a**
```
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

```
wim@FlightTracker:~ $ **cat /proc/cpuinfo**
```
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 44.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

...

Hardware        : BCM2835
Revision        : a02082
Serial          : 00000000c16a7c2c
Model           : Raspberry Pi 3 Model B Rev 1.2

```

cnighswonger, 2024-06-20:
@wcbonner thanks for the response.

I just realized that I was on my Pi 3 rather than 4.

I'll have a look again when I'm home. Its not a huge issue as I'm just logging data at this point. I suspect it something local and a fresh install of the OS might clear it up.

wcbonner, 2024-08-16:
@cnighswonger I just overhauled the Bluetooth code and made a new release. https://github.com/wcbonner/GoveeBTTempLogger/releases/tag/v3.20240816.0 

Does the new code still cause issues for your SSH terminal?

closed 2024-12-20 by wcbonner