wcbonner/GoveeBTTempLogger #62: Downloading stored data for H5104 (does not work?)

jer194, 2024-07-01:
Hi,

I am trying to download stored data from my H5104, but it doesn't seem to work.
Probably it is the same reason like for H5100 and H5105 (reading your comments)?

In /var/log/goveebttemplogger/ there is this file:

rw-r--r--  1 root root     27 Jan  1  1970 gvh-lastdownload.txt

It contains:

> cat gvh-lastdownload.txt
D2:38:32:30:1F:19	00:00:00

Thanks for the great tool btw :-)
Jakob.

wcbonner, 2024-07-01:
@jer194 Which model Raspberry are you using?

I've been having problems reading the historical data on my Pi4 with the latest Raspberry Pi OS (64 bit, so Debian with Pi additions) 

I'm attempting to learn the DBus Bluetooth interface since the "hci" interface was deprecated five years ago, and I've finally found some articles gathering data in something close to the method I am looking for. 

Other than that, I don't have a historical download solution working for all the devices right now, and have not been using the option on my primary machine because the download timeout causes it to miss too many advertisements. 

> I am trying to download stored data from my H5104, but it doesn't seem to work.
> 
> Probably it is the same reason like for H5100 and H5105 (reading your comments)?
>

jer194, 2024-07-01:
This is what I am using:

Dell Latitude 7490 Touch

> cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux trixie/sid"

>  uname -a
Linux dell 6.8.12-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.8.12-1 (2024-05-31) x86_64 GNU/Linux

(I have a raspberry as well somewhere, but I didn't use it recently anymore)

wcbonner, 2024-12-18:
> This is what I am using:
>
>
>
> Dell Latitude 7490 Touch
>
>
>
>
> PRETTY_NAME="Debian GNU/Linux trixie/sid"
>
>
>
>
> Linux dell 6.8.12-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.8.12-1 (2024-05-31) x86_64 GNU/Linux [...]

@jer194 I have been trying to implement downloading of data using DBus. It's not working yet, but I've also made minor changes around the HCI interface code that may allow the historical data to be downloaded on your GVH5104 device. 

I'm running on a Pi4 and the test location I'm running right now has a GVH5075, GVH5104, and GVH5174. 

I ran the current program (built December 13) with the --download --HCI options and it was able to download from all three devices. 

(I also had --log, --svg, --cache, and --verbose 2 options set at the time)

jer194, 2024-12-19:
Hi @wcbonner,

I have compiled the current goveebttemplogger for my rpi4 (following your excellent steb by step guide!),
and indeed I was able to download all historical data (for 14 days back, that's when I had to change the batteries), 
using this command:

> goveebttemplogger -d -H -l . -c

Not that I need it all the time, I'm continuously collecting the data now,
but it might be useful in case of failures ...

Thanks a lot for all your work and dedication!
Jakob.

closed 2024-12-20 by wcbonner