wcbonner/GoveeBTTempLogger #63: DBus BlueZ H5179 unknown if decoding

wcbonner, 2024-08-14:
I don't have a GVH5179 and haven't confirmed the decode routine works

wcbonner, 2024-08-14:
Old code supported H5179 as reported here: https://github.com/wcbonner/GoveeBTTempLogger/issues/36

I don't have a H5179 so need another person to test the new DBus code to verify that the H5179 is being decoded. @alec2dabreen Are you still running this program and have time to test the https://github.com/wcbonner/GoveeBTTempLogger/tree/BlueZ-DBus branch?

alec2dabreen, 2024-08-14:
Sorry @wcbonner -- I'm no longer in possession of a 5179. :(

wcbonner, 2024-08-14:
> Sorry @wcbonner -- I'm no longer in possession of a 5179. :(

Thanks for getting back to me. 😎

wcbonner, 2025-02-18:
Fixed and verified GVH5179 decoding

closed 2025-02-18 by wcbonner