wcbonner/GoveeBTTempLogger #75: Slow startup issue...

blackketter, 2024-10-06:
I've been running v2 for more than a year without any issue.  It's been great, nice work!  (400 days uptime!)

I made a small change to the goveebttemplogger.cpp to adjust the colors in the SVG output files.

I successfully rebuilt, reinstalled and relaunched the daemon.  Unfortunately it's been stuck using nearly 100% of a core for more than an hour.  I see from the README that "The causes the program to take longer to start up as it will attempt to read all of the old logged data into an internal memory structure as it starts."  Wondering if it's going to finish in a reasonable amount of time, or if there's some exponential slowdown that I need to find another solution for. 

Any suggestions? 

The syslog looks like this:
```
Oct  6 15:24:16 buster systemd[1]: Starting GoveeBTTempLogger service...
Oct  6 15:24:16 buster systemd[1]: Started GoveeBTTempLogger service.
Oct  6 15:24:16 buster goveebttemplogger[4131]: [2024-10-06T22:24:16] Reading: /var/www/html/goveebttemplogger/gvh-titlemap.txt
Oct  6 15:24:16 buster goveebttemplogger[4131]: [2024-10-06T22:24:16] GoveeBTTempLogger Version 2.20230901-1 Built on: Oct  6 2024 at 15:14:30
Oct  6 15:24:16 buster goveebttemplogger[4131]: [2024-10-06T22:24:16] Reading: /var/www/html/goveebttemplogger/gvh-A4C1381A352C-2023-09.txt
Oct  6 16:04:07 buster goveebttemplogger[4131]: [2024-10-06T23:04:07] Reading: /var/www/html/goveebttemplogger/gvh-A4C1381A352C-2023-10.txt
Oct  6 16:04:59 buster goveebttemplogger[4131]: [2024-10-06T23:04:59] Reading: /var/www/html/goveebttemplogger/gvh-A4C1381A352C-2023-11.txt
Oct  6 16:05:52 buster goveebttemplogger[4131]: [2024-10-06T23:05:52] Reading: /var/www/html/goveebttemplogger/gvh-A4C1381A352C-2023-12.txt
Oct  6 16:06:47 buster goveebttemplogger[4131]: [2024-10-06T23:06:47] Reading: /var/www/html/goveebttemplogger/gvh-A4C1381A352C-2024-01.txt
Oct  6 16:07:42 buster goveebttemplogger[4131]: [2024-10-06T23:07:42] Reading: /var/www/html/goveebttemplogger/gvh-A4C1381A352C-2024-02.txt
Oct  6 16:08:01 buster goveebttemplogger[4131]: [2024-10-06T23:08:01] Reading: /var/www/html/goveebttemplogger/gvh-A4C1381A352C-2024-03.txt
Oct  6 16:08:17 buster goveebttemplogger[4131]: [2024-10-06T23:08:17] Reading: /var/www/html/goveebttemplogger/gvh-A4C1381A352C-2024-04.txt
Oct  6 16:08:58 buster goveebttemplogger[4131]: [2024-10-06T23:08:58] Reading: /var/www/html/goveebttemplogger/gvh-A4C1381A352C-2024-05.txt
Oct  6 16:09:44 buster goveebttemplogger[4131]: [2024-10-06T23:09:44] Reading: /var/www/html/goveebttemplogger/gvh-A4C1381A352C-2024-06.txt
Oct  6 16:10:31 buster goveebttemplogger[4131]: [2024-10-06T23:10:31] Reading: /var/www/html/goveebttemplogger/gvh-A4C1381A352C-2024-07.txt
Oct  6 16:11:22 buster goveebttemplogger[4131]: [2024-10-06T23:11:22] Reading: /var/www/html/goveebttemplogger/gvh-A4C1381A352C-2024-08.txt
Oct  6 16:11:50 buster goveebttemplogger[4131]: [2024-10-06T23:11:50] Reading: /var/www/html/goveebttemplogger/gvh-A4C1381A352C-2024-09.txt
Oct  6 16:12:33 buster goveebttemplogger[4131]: [2024-10-06T23:12:33] Reading: /var/www/html/goveebttemplogger/gvh-A4C1381A352C-2024-10.txt
Oct  6 16:12:39 buster goveebttemplogger[4131]: [2024-10-06T23:12:39] Reading: /var/www/html/goveebttemplogger/gvh-A4C138DB1B37-2024-01.txt
Oct  6 16:12:39 buster goveebttemplogger[4131]: [2024-10-06T23:12:39] Reading: /var/www/html/goveebttemplogger/gvh-A4C138E43FC8-2023-09.txt
```
and that's the last entry for more than half an hour.

And my log directory looks like this:
```
dean@buster:/var/www/html/goveebttemplogger $ ls -l
total 683788
-rw-r--r-- 1 root root 18849153 Sep 30  2023 gvh-A4C1381A352C-2023-09.txt
-rw-r--r-- 1 root root 21385558 Oct 31  2023 gvh-A4C1381A352C-2023-10.txt
-rw-r--r-- 1 root root 20812238 Nov 30  2023 gvh-A4C1381A352C-2023-11.txt
-rw-r--r-- 1 root root 21916320 Dec 31  2023 gvh-A4C1381A352C-2023-12.txt
-rw-r--r-- 1 root root 22768794 Jan 31  2024 gvh-A4C1381A352C-2024-01.txt
-rw-r--r-- 1 root root  6501049 Feb 29  2024 gvh-A4C1381A352C-2024-02.txt
-rw-r--r-- 1 root root  5599906 Mar 31  2024 gvh-A4C1381A352C-2024-03.txt
-rw-r--r-- 1 root root 16811165 Apr 30 16:59 gvh-A4C1381A352C-2024-04.txt
-rw-r--r-- 1 root root 18841030 May 30 21:16 gvh-A4C1381A352C-2024-05.txt
-rw-r--r-- 1 root root 18319785 Jun 30 16:59 gvh-A4C1381A352C-2024-06.txt
-rw-r--r-- 1 root root 20806400 Jul 31 16:58 gvh-A4C1381A352C-2024-07.txt
-rw-r--r-- 1 root root 11114382 Aug 31 16:57 gvh-A4C1381A352C-2024-08.txt
-rw-r--r-- 1 root root 17335720 Sep 30 16:59 gvh-A4C1381A352C-2024-09.txt
-rw-r--r-- 1 root root  2377410 Oct  6 15:23 gvh-A4C1381A352C-2024-10.txt
-rw-r--r-- 1 root root    11768 Oct  6 15:19 gvh-A4C1381A352C-day.svg
-rw-r--r-- 1 root root     8733 Oct  6 14:00 gvh-A4C1381A352C-month.svg
-rw-r--r-- 1 root root     9405 Oct  6 15:00 gvh-A4C1381A352C-week.svg
-rw-r--r-- 1 root root    18646 Oct  6 00:00 gvh-A4C1381A352C-year.svg
-rw-r--r-- 1 root root       35 Jan  2  2024 gvh-A4C138DB1B37-2024-01.txt
-rw-r--r-- 1 root root 30186634 Sep 30  2023 gvh-A4C138E43FC8-2023-09.txt
-rw-r--r-- 1 root root 33148318 Oct 31  2023 gvh-A4C138E43FC8-2023-10.txt
-rw-r--r-- 1 root root 30139499 Nov 30  2023 gvh-A4C138E43FC8-2023-11.txt
-rw-r--r-- 1 root root 33912482 Dec 31  2023 gvh-A4C138E43FC8-2023-12.txt
-rw-r--r-- 1 root root 32942882 Jan 31  2024 gvh-A4C138E43FC8-2024-01.txt
-rw-r--r-- 1 root root 10414181 Feb 29  2024 gvh-A4C138E43FC8-2024-02.txt
-rw-r--r-- 1 root root  8872127 Mar 31  2024 gvh-A4C138E43FC8-2024-03.txt
-rw-r--r-- 1 root root 29328491 Apr 30 16:59 gvh-A4C138E43FC8-2024-04.txt
-rw-r--r-- 1 root root 32063055 May 31 16:59 gvh-A4C138E43FC8-2024-05.txt
-rw-r--r-- 1 root root 31081165 Jun 30 16:59 gvh-A4C138E43FC8-2024-06.txt
-rw-r--r-- 1 root root 32573553 Jul 31 16:58 gvh-A4C138E43FC8-2024-07.txt
-rw-r--r-- 1 root root 18891982 Aug 31 16:59 gvh-A4C138E43FC8-2024-08.txt
-rw-r--r-- 1 root root 26089891 Sep 30 16:59 gvh-A4C138E43FC8-2024-09.txt
-rw-r--r-- 1 root root  4097765 Oct  6 15:24 gvh-A4C138E43FC8-2024-10.txt
-rw-r--r-- 1 root root    11577 Oct  6 15:20 gvh-A4C138E43FC8-day.svg
-rw-r--r-- 1 root root     8749 Oct  6 14:00 gvh-A4C138E43FC8-month.svg
-rw-r--r-- 1 root root     9123 Oct  6 15:00 gvh-A4C138E43FC8-week.svg
-rw-r--r-- 1 root root    18647 Oct  6 00:00 gvh-A4C138E43FC8-year.svg
-rw-r--r-- 1 root root  3147198 Jul 31 16:58 gvh-C43038364C77-2024-07.txt
-rw-r--r-- 1 root root 13257295 Aug 31 16:59 gvh-C43038364C77-2024-08.txt
-rw-r--r-- 1 root root 19060506 Sep 30 16:59 gvh-C43038364C77-2024-09.txt
-rw-r--r-- 1 root root  3447626 Oct  6 15:24 gvh-C43038364C77-2024-10.txt
-rw-r--r-- 1 root root    11403 Oct  6 15:20 gvh-C43038364C77-day.svg
-rw-r--r-- 1 root root     8647 Oct  6 14:00 gvh-C43038364C77-month.svg
-rw-r--r-- 1 root root     9064 Oct  6 15:00 gvh-C43038364C77-week.svg
-rw-r--r-- 1 root root     5054 Oct  6 00:00 gvh-C43038364C77-year.svg
-rw-r--r-- 1 root root  3980405 Sep 28 11:10 gvh-C93534311E1E-2024-09.txt
-rw-r--r-- 1 root root     8562 Sep 28 11:09 gvh-C93534311E1E-day.svg
-rw-r--r-- 1 root root     1723 Sep 28 10:00 gvh-C93534311E1E-month.svg
-rw-r--r-- 1 root root     2333 Sep 28 11:00 gvh-C93534311E1E-week.svg
-rw-r--r-- 1 root root     1949 Sep 28 00:00 gvh-C93534311E1E-year.svg
-rw-r--r-- 1 root root  4937343 Jul 31 16:58 gvh-D93830340256-2024-07.txt
-rw-r--r-- 1 root root 33551317 Aug 31 16:59 gvh-D93830340256-2024-08.txt
-rw-r--r-- 1 root root 33333502 Sep 30 16:59 gvh-D93830340256-2024-09.txt
-rw-r--r-- 1 root root  6359104 Oct  6 15:24 gvh-D93830340256-2024-10.txt
-rw-r--r-- 1 root root    11447 Oct  6 15:20 gvh-D93830340256-day.svg
-rw-r--r-- 1 root root     8736 Oct  6 14:00 gvh-D93830340256-month.svg
-rw-r--r-- 1 root root     9077 Oct  6 15:00 gvh-D93830340256-week.svg
-rw-r--r-- 1 root root     5070 Oct  6 00:00 gvh-D93830340256-year.svg
-rw-r--r-- 1 root root     1184 Feb  9  2024 gvh-F1B21374BFC0-2024-02.txt
-rw-r--r-- 1 root root     2077 Feb  9  2024 gvh-F1B21374BFC0-day.svg
-rw-r--r-- 1 root root     2077 Feb  9  2024 gvh-F1B21374BFC0-week.svg
-rw-r--r-- 1 root root      130 Jul 13 15:36 gvh-lastdownload.txt
-rw-r--r-- 1 dean dean      105 Jul 27 10:27 gvh-titlemap.txt
-rw-r--r-- 1 dean dean      950 Jul 27 10:32 temperature.html
```

blackketter, 2024-10-07:
It finally finished after about an hour and 35 minutes.  Not the end of the world, but maybe a place for some optimization one day.

wcbonner, 2024-10-07:
> It finally finished after about an hour and 35 minutes.  Not the end of the world, but maybe a place for some optimization one day.  

I was running on a Pi4 and it was taking an hour to startup. That's why I developed the cache file option. When using a cache file in addition to the log file, it only works hard to integrate the new data logged since the cache was last written.

closed 2025-02-13 by wcbonner