wcbonner/GoveeBTTempLogger #80: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress

alzuno, 2024-11-15:
Hi @wcbonner,

I've been experiencing recurring service interruptions. Over the past month, the system hasn't remained up for more than 24 hours. Previously, it would sometimes run for a month without any issues.

From the logs, I noticed this recurring error: 

`/org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3918)`

After this error appears, the logging stops entirely. The only way to get the system working again is to reboot the Raspberry Pi.

Here a piece of logs from the system:

```
Nov 14 09:39:41 rasp3 goveebttemplogger[33743]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 14 09:39:41 rasp3 goveebttemplogger[33743]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 14 09:39:41 rasp3 goveebttemplogger[33743]: /org/bluez/hci0: org.freedesktop.DBus.Properties: SetPowered: true
Nov 14 09:39:41 rasp3 goveebttemplogger[33743]: Host Controller Address: B8:27:EB:FF:AD:6A BlueZ Adapter Path: /org/bluez/hci0
Nov 14 09:39:41 rasp3 goveebttemplogger[33743]: Connected to D-Bus as ":1.21"
Nov 14 09:39:41 rasp3 goveebttemplogger[33743]: Reading LastDownload: /var/log/goveebttemplogger/gvh-lastdownload.txt
Nov 14 09:39:40 rasp3 goveebttemplogger[33743]: Reading: /var/log/goveebttemplogger/gvh-A4C13856DF3C-2024-11.txt
Nov 14 09:39:40 rasp3 goveebttemplogger[33743]: Reading: /var/log/goveebttemplogger/gvh-A4C138043FDE-2024-11.txt
Nov 14 09:39:40 rasp3 goveebttemplogger[33743]: Reading: /var/cache/goveebttemplogger/gvh-A4C13856DF3C-cache.txt
Nov 14 09:39:40 rasp3 goveebttemplogger[33743]: Reading: /var/cache/goveebttemplogger/gvh-A4C138043FDE-cache.txt
Nov 14 09:39:40 rasp3 goveebttemplogger[33743]: Reading: /var/log/goveebttemplogger/gvh-thermometer-types.txt
Nov 14 09:39:40 rasp3 goveebttemplogger[33743]: Reading: /var/www/html/goveebttemplogger/gvh-titlemap.txt
Nov 14 09:39:40 rasp3 goveebttemplogger[33743]: GoveeBTTempLogger Version 3.20241023.1 Built on: Nov 13 2024 at 09:24:30 (starting)
Nov 14 09:39:40 rasp3 systemd[1]: Started GoveeBTTempLogger service.
Nov 14 09:39:33 rasp3 systemd[1]: goveebttemplogger.service: Consumed 17min 4.071s CPU time.
Nov 14 09:39:33 rasp3 systemd[1]: Stopped GoveeBTTempLogger service.
Nov 14 09:39:33 rasp3 systemd[1]: goveebttemplogger.service: Succeeded.
Nov 14 09:39:33 rasp3 goveebttemplogger[674]: GoveeBTTempLogger Version 3.20241023.1 Built on: Nov 13 2024 at 09:24:30 (exiting)
Nov 14 09:39:33 rasp3 goveebttemplogger[674]: Writing: /var/log/goveebttemplogger/gvh-thermometer-types.txt
Nov 14 09:39:33 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3918)
Nov 14 09:39:32 rasp3 systemd[1]: Stopping GoveeBTTempLogger service...
Nov 14 09:39:32 rasp3 goveebttemplogger[674]: ***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
Nov 14 09:30:56 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 14 09:30:56 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 14 09:30:56 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3918)
Nov 14 09:00:55 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 14 09:00:55 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 14 09:00:55 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3918)
Nov 14 08:30:54 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 14 08:30:54 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 14 08:30:54 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3918)
Nov 14 08:00:53 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 14 08:00:53 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 14 08:00:53 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3918)
Nov 14 07:30:52 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 14 07:30:52 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 14 07:30:52 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3918)
Nov 14 07:00:51 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 14 07:00:51 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 14 07:00:51 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3918)
Nov 14 06:30:50 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 14 06:30:50 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 14 06:30:50 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3918)
Nov 14 06:00:49 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 14 06:00:49 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 14 06:00:49 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3918)
Nov 14 05:30:48 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 14 05:30:48 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 14 05:30:48 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3918)
Nov 14 05:00:47 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 14 05:00:47 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 14 05:00:47 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3918)
Nov 14 04:30:46 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 14 04:30:46 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 14 04:30:46 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3918)
Nov 14 04:00:45 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 14 04:00:45 rasp3 goveebttemplogger[674]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
[... 106 lines cut ...]
[-------------------] [A4:C1:38:04:3F:DE] 00:00:00
[-------------------] [A4:C1:38:56:DF:3C] 00:00:00
[-------------------] [A4:C1:38:04:3F:DE] 00:00:00
[-------------------] [A4:C1:38:56:DF:3C] 00:00:00
[2024-11-15T14:00:01] 300 seconds or more have passed. Writing SVG Files
[2024-11-15T14:00:01] Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
[2024-11-15T14:00:01] Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
[2024-11-15T14:00:17] [A4:C1:38:04:3F:DE] (Temp) 21.1°C (Humidity)  60.8% (Battery)  47% (GVH5075)
[-------------------] [A4:C1:38:04:3F:DE] 00:00:00
[-------------------] [A4:C1:38:56:DF:3C] 00:00:00
[-------------------] [A4:C1:38:04:3F:DE] 00:00:00
[-------------------] [A4:C1:38:56:DF:3C] 00:00:00
[-------------------] [A4:C1:38:04:3F:DE] 00:00:00
[-------------------] [A4:C1:38:56:DF:3C] 00:00:00
```

Additional Information:
```
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```

Thanks for your help!

wcbonner, 2024-11-15:
@alzuno I'd seen some problems like this but was hoping that I'd fixed it. 

Are you running on a Raspberry Pi or another platform? (I'm guessing by your machine name in the logs that it's a Raspberry Pi3.)

Is the software running as the user goveebttemplogger instead of root? (one of my recent changes was to avoid running as root)

Are you running anything else that uses Bluetooth? (Keyboard, Mouse, or any other program?)

Am I correct that the Bluetooth appears to not work at all after this problem occurs, and the only way to recover it is to reboot the entire machine?

I've seen this issue and am not sure what's happening. There may be something I'm not cleaning up in the DBus communication that is causing BlueZ to leak memory and stop responding. It seems wrong that a user mode program can crash a piece of system software..

You should be able to revert to the old hardware Bluetooth Hardware Control Interface by adding the --HCI option to the end of the command line, but that probably won't recover the Bluetooth after it's locked up. Using HCI took exclusive control of the Bluetooth interface and has been deprecated for several years.

wcbonner, 2024-11-16:
@alzuno Commits I made today https://github.com/wcbonner/GoveeBTTempLogger/commit/c8e2cd16bce06fd4db5d36624addba28b6471faa and then https://github.com/wcbonner/GoveeBTTempLogger/commit/6b895c515abb1a28d4ec0a7aba391b6e14c7bcb8 bring the version up to 3.20241115.1 and I'm hoping will squash the issue you have for good.

I'm going to run this on my primary machines for a couple of days before creating a new release, but I'm interested if this fixes the issue on your platform.

alzuno, 2024-11-16:
Hi @wcbonner,

Thanks for your reply, and sorry for the delayed response (I’m on the "other side of the pond").

> Are you running on a Raspberry Pi or another platform? (I'm guessing by your machine name in the logs that it's a Raspberry Pi3.)

Yes, it's a Raspberry Pi3:
```
Hardware	: BCM2835
Revision	: a020d3
Model		: Raspberry Pi 3 Model B Plus Rev 1.3
```
And here’s the OS information:
```
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```
> 
> Is the software running as the user goveebttemplogger instead of root? (one of my recent changes was to avoid running as root)

It is running as the `goveebttemplogger` user. Here's the `ps aux` output:

`goveebt+     675  0.0  0.6  10480  5628 ?        Ss   Nov15   0:09 /usr/local/bin/goveebttemplogger --verbose 0 --log /var/log/goveebttemplogger --time 60 --svg /var/www/html/gove`

> Are you running anything else that uses Bluetooth? (Keyboard, Mouse, or any other program?)

No, it’s a headless setup. Nothing but electricity is connected to the Raspberry Pi, and no other program is using Bluetooth.

> Am I correct that the Bluetooth appears to not work at all after this problem occurs, and the only way to recover it is to reboot the entire machine?

Yes, that’s exactly the behavior. The only way to recover is to reboot the machine.

>@alzuno Commits I made today c8e2cd1 and then 6b895c5 bring the version up to 3.20241115.1 and I'm hoping will squash the issue you have for good.

>I'm going to run this on my primary machines for a couple of days before creating a new release, but I'm interested if this fixes the issue on your platform.

Great! I’m updating to the latest version now and will run it over the weekend. I’ll let you know the results.
Thanks!

alzuno, 2024-11-16:
Hi @wcbonner,

I updated the system to 3.20241115.1 version, and unfortunately it seems is not working.
As soon as the service is started, is just `/org/bluez/hci0: org.bluez.Adapter1: StartDiscovery` and then nothing else is happening.

Here's a journalctl output:
```
pi@rasp3:~ $ journalctl -r -u goveebttemplogger
-- Journal begins at Thu 2024-07-04 02:14:43 CEST, ends at Sat 2024-11-16 11:53:49 CET. --
Nov 16 11:53:49 rasp3 goveebttemplogger[1478]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 16 11:53:49 rasp3 goveebttemplogger[1478]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 16 11:53:49 rasp3 goveebttemplogger[1478]: /org/bluez/hci0: org.freedesktop.DBus.Properties: SetPowered: true
Nov 16 11:53:49 rasp3 goveebttemplogger[1478]: Host Controller Address: B8:27:EB:FF:AD:6A BlueZ Adapter Path: /org/bluez/h>
Nov 16 11:53:49 rasp3 goveebttemplogger[1478]: D-Bus connection ":1.18" Opened
Nov 16 11:53:49 rasp3 goveebttemplogger[1478]: Reading LastDownload: /var/log/goveebttemplogger/gvh-lastdownload.txt
Nov 16 11:53:48 rasp3 goveebttemplogger[1478]: Reading: /var/log/goveebttemplogger/gvh-A4C13856DF3C-2024-11.txt
Nov 16 11:53:48 rasp3 goveebttemplogger[1478]: Reading: /var/log/goveebttemplogger/gvh-A4C138043FDE-2024-11.txt
Nov 16 11:53:48 rasp3 goveebttemplogger[1478]: Reading: /var/cache/goveebttemplogger/gvh-A4C13856DF3C-cache.txt
Nov 16 11:53:48 rasp3 goveebttemplogger[1478]: Reading: /var/cache/goveebttemplogger/gvh-A4C138043FDE-cache.txt
Nov 16 11:53:48 rasp3 goveebttemplogger[1478]: Reading: /var/log/goveebttemplogger/gvh-thermometer-types.txt
Nov 16 11:53:48 rasp3 goveebttemplogger[1478]: Reading: /var/www/html/goveebttemplogger/gvh-titlemap.txt
Nov 16 11:53:48 rasp3 goveebttemplogger[1478]: GoveeBTTempLogger Version 3.20241115.1 Built on: Nov 16 2024 at 10:20:24 (s>
Nov 16 11:53:47 rasp3 systemd[1]: Started GoveeBTTempLogger service.
Nov 16 11:53:45 rasp3 systemd[1]: goveebttemplogger.service: Consumed 2.769s CPU time.
Nov 16 11:53:45 rasp3 systemd[1]: Stopped GoveeBTTempLogger service.
Nov 16 11:53:45 rasp3 systemd[1]: goveebttemplogger.service: Succeeded.
Nov 16 11:53:45 rasp3 goveebttemplogger[680]: GoveeBTTempLogger Version 3.20241115.1 Built on: Nov 16 2024 at 10:20:24 (ex>
Nov 16 11:53:45 rasp3 goveebttemplogger[680]: Writing: /var/log/goveebttemplogger/gvh-thermometer-types.txt
Nov 16 11:53:45 rasp3 goveebttemplogger[680]: D-Bus connection ":1.14" Closed
Nov 16 11:53:45 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already>
Nov 16 11:53:44 rasp3 systemd[1]: Stopping GoveeBTTempLogger service...
Nov 16 11:53:44 rasp3 goveebttemplogger[680]: ***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. ******>
Nov 16 11:29:37 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 16 11:29:37 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 16 11:29:37 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.freedesktop.DBus.Properties: SetPowered: true
Nov 16 11:29:37 rasp3 goveebttemplogger[680]: Host Controller Address: B8:27:EB:FF:AD:6A BlueZ Adapter Path: /org/bluez/hc>
Nov 16 11:29:37 rasp3 goveebttemplogger[680]: D-Bus connection ":1.14" Opened
Nov 16 11:29:37 rasp3 goveebttemplogger[680]: D-Bus connection ":1.12" Closed
Nov 16 11:29:37 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already>
Nov 16 11:10:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-week.svg With Titl>
Nov 16 11:10:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title>
Nov 16 11:10:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-week.svg With Titl>
Nov 16 11:10:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title>
-- Boot 83550d0c219841dfb643e11c2caa6fa9 --
Nov 16 11:00:02 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title>
Nov 16 11:00:02 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title>
-- Boot da42c8b1f2c14a3abec5b8d2526e0ba5 --
Nov 16 10:59:38 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 16 10:59:37 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 16 10:59:37 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.freedesktop.DBus.Properties: SetPowered: true
Nov 16 10:59:37 rasp3 goveebttemplogger[680]: Host Controller Address: B8:27:EB:FF:AD:6A BlueZ Adapter Path: /org/bluez/hc>
Nov 16 10:59:37 rasp3 goveebttemplogger[680]: D-Bus connection ":1.12" Opened
Nov 16 10:59:37 rasp3 goveebttemplogger[680]: Reading LastDownload: /var/log/goveebttemplogger/gvh-lastdownload.txt
Nov 16 10:59:37 rasp3 goveebttemplogger[680]: Reading: /var/log/goveebttemplogger/gvh-A4C13856DF3C-2024-11.txt
Nov 16 10:59:36 rasp3 goveebttemplogger[680]: Reading: /var/log/goveebttemplogger/gvh-A4C138043FDE-2024-11.txt
Nov 16 10:59:36 rasp3 goveebttemplogger[680]: Reading: /var/cache/goveebttemplogger/gvh-A4C13856DF3C-cache.txt
Nov 16 10:59:35 rasp3 goveebttemplogger[680]: Reading: /var/cache/goveebttemplogger/gvh-A4C138043FDE-cache.txt
Nov 16 10:59:35 rasp3 goveebttemplogger[680]: Reading: /var/log/goveebttemplogger/gvh-thermometer-types.txt
[... cut ...]
[-------------------] [A4:C1:38:04:3F:DE] 00:00:00
[-------------------] [A4:C1:38:56:DF:3C] 00:00:00
[-------------------] [A4:C1:38:04:3F:DE] 00:00:00
[-------------------] [A4:C1:38:56:DF:3C] 00:00:00
[2024-11-16T12:10:01] 300 seconds or more have passed. Writing SVG Files
[-------------------] [A4:C1:38:04:3F:DE] 00:00:00
[-------------------] [A4:C1:38:56:DF:3C] 00:00:00
[-------------------] [A4:C1:38:04:3F:DE] 00:00:00
[-------------------] [A4:C1:38:56:DF:3C] 00:00:00
```

Please let me know if you need anything else from me.
Thanks

wcbonner, 2024-11-16:
@alzuno I'm not sure why it's not working for you. 

My test would be to disable the service and reboot the machine. preferably power cycled, but sometimes doing that on a headless machine is a pain. 

```
sudo systemctl stop goveebttemplogger.service
sudo systemctl disable goveebttemplogger.service
sudo systemctl reboot
```
then at the command line run these commands, replacing "wim" with the name of your local logged in user. 

I made a minor change this morning, bumping the version to 3.20241116.0 but it shouldn't have changed the main logic.
```
pushd ~/GoveeBTTempLogger/ && git pull && popd && cmake -S ~/GoveeBTTempLogger/ -B ~/GoveeBTTempLogger/build && cmake --build ~/GoveeBTTempLogger/build && pushd ~/GoveeBTTempLogger/build && cpack . && popd
mkdir --parents /home/wim/gvh/log/ /home/wim/gvh/svg/ /home/wim/gvh/cache/
```
Run the program in default DBus mode. If it reports govee devices and temperatures, that's good. If not, I'm interested if you let it run for over half an hour if it closes the DBus connection and opens another with a different name successfully. 
```
/home/wim/GoveeBTTempLogger/build/goveebttemplogger --verbose 1 --log /home/wim/gvh/log/ --time 60 --svg /home/wim/gvh/svg/ --battery 8 --minmax 8 --cache /home/wim/gvh/cache/
```
If the DBus communication didn't work, try the same by adding the --HCI option to see if it gets data that way. 
```
/home/wim/GoveeBTTempLogger/build/goveebttemplogger --verbose 1 --log /home/wim/gvh/log/ --time 60 --svg /home/wim/gvh/svg/ --battery 8 --minmax 8 --cache /home/wim/gvh/cache/ --HCI
```

alzuno, 2024-11-17:
Hi @wcbonner,
I ran the tests you requested, and here are the results:

1. I disabled the service using the provided commands.
2. I then powered off the Raspberry Pi, physically disconnecting it from the power supply, and plugged it back in.
3. After restarting, I ran the provided commands.
4. The system immediately detected the Govee devices and reported their temperatures without any issues.

```
pi@rasp3:~ $ pwd
/home/pi
pi@rasp3:~ $ git clone https://github.com/wcbonner/GoveeBTTempLogger.git
Cloning into 'GoveeBTTempLogger'...
remote: Enumerating objects: 2265, done.
remote: Counting objects: 100% (983/983), done.
remote: Compressing objects: 100% (320/320), done.
remote: Total 2265 (delta 711), reused 891 (delta 643), pack-reused 1282 (from 1)
Receiving objects: 100% (2265/2265), 3.73 MiB | 6.74 MiB/s, done.
Resolving deltas: 100% (1264/1264), done.
pi@rasp3:~ $ pushd ~/GoveeBTTempLogger/ && git pull && popd && cmake -S ~/GoveeBTTempLogger/ -B ~/GoveeBTTempLogger/build && cmake --build ~/GoveeBTTempLogger/build && pushd ~/GoveeBTTempLogger/build && cpack . && popd
~/GoveeBTTempLogger ~
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Already up to date.
~
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'dbus-1'
--   Found dbus-1, version 1.12.28
-- Components to pack: 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/GoveeBTTempLogger/build
Scanning dependencies of target goveebttemplogger
[ 14%] Building CXX object CMakeFiles/goveebttemplogger.dir/goveebttemplogger.cpp.o
[ 28%] Building C object CMakeFiles/goveebttemplogger.dir/uuid.c.o
[ 42%] Building CXX object CMakeFiles/goveebttemplogger.dir/wimiso8601.cpp.o
[ 57%] Linking CXX executable goveebttemplogger
Setting Raw Priveleges on $<TARGET_FILE:goveebttemplogger>
[ 57%] Built target goveebttemplogger
Scanning dependencies of target gvh-organizelogs
[ 71%] Building CXX object CMakeFiles/gvh-organizelogs.dir/gvh-organizelogs.cpp.o
[ 85%] Building CXX object CMakeFiles/gvh-organizelogs.dir/wimiso8601.cpp.o
[100%] Linking CXX executable gvh-organizelogs
[100%] Built target gvh-organizelogs
~/GoveeBTTempLogger/build ~
CPack: Create package using DEB
CPack: Install projects
CPack: - Run preinstall target for: GoveeBTTempLogger
[... 101 lines cut ...]
[2024-11-16T21:26:08] [A4:C1:38:04:3F:DE] (Temp) 21.0°C (Humidity)  60.2% (Battery)  47% (GVH5075)
[-------------------] [A4:C1:38:04:3F:DE] 00:00:00
[-------------------] [A4:C1:38:56:DF:3C] 00:00:00
[2024-11-16T21:26:10] [A4:C1:38:04:3F:DE] (Temp) 20.9°C (Humidity)  60.2% (Battery)  47% (GVH5075)
[2024-11-16T21:26:11] [A4:C1:38:04:3F:DE] (Temp) 21.0°C (Humidity)  60.3% (Battery)  47% (GVH5075)
[2024-11-16T21:26:12] [A4:C1:38:56:DF:3C] (Temp) 20.7°C (Humidity)  57.0% (Battery)  85% (GVH5075)
[2024-11-16T21:26:13] [A4:C1:38:04:3F:DE] (Temp) 21.0°C (Humidity)  60.3% (Battery)  47% (GVH5075)
^C***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
[                   ] /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery
[                   ] /org/bluez/hci0: org.bluez.Adapter1: RemoveDevice /org/bluez/hci0/dev_A4_C1_38_56_DF_3C
[                   ] /org/bluez/hci0: org.bluez.Adapter1: RemoveDevice /org/bluez/hci0/dev_A4_C1_38_04_3F_DE
[2024-11-16T21:26:16] D-Bus connection ":1.16" Closed
[-------------------] [A4:C1:38:04:3F:DE] 00:00:00
[-------------------] [A4:C1:38:56:DF:3C] 00:00:00
[-------------------] [A4:C1:38:04:3F:DE] 00:00:00
[-------------------] [A4:C1:38:56:DF:3C] 00:00:00
[2024-11-16T21:26:16] Writing: /home/pi/gvh/log/gvh-thermometer-types.txt
GoveeBTTempLogger Version 3.20241116.0 Built on: Nov 16 2024 at 21:18:24 (exiting)
```

After this, I updated the system and rebooted it again. It's now been running for approximately 24 hours without any problems.

I'm not sure what was causing the previous issues, but for now, everything seems to be working correctly. I'll continue monitoring and will let you know if the issue reappears.

Thank you so much for your help! Let me know if there’s anything else you’d like me to test.

alzuno, 2024-11-17:
Additionally, in the running service logs, I’ve noticed that every 30 minutes the system closes a D-Bus connection and opens a new one with a different code.

```
Nov 17 16:32:39 rasp3 goveebttemplogger[680]: D-Bus connection ":1.54" Opened
Nov 17 16:32:39 rasp3 goveebttemplogger[680]: D-Bus connection ":1.50" Closed
```

Here’s a snippet from the logs:

```
-- Journal begins at Thu 2024-07-04 02:14:43 CEST, ends at Sun 2024-11-17 16:35:02 CET. --
Nov 17 16:35:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-week.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 17 16:35:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 17 16:35:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-week.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 17 16:35:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 17 16:32:39 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 17 16:32:39 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 17 16:32:39 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.freedesktop.DBus.Properties: SetPowered: true
Nov 17 16:32:39 rasp3 goveebttemplogger[680]: Host Controller Address: B8:27:EB:FF:AD:6A BlueZ Adapter Path: /org/bluez/hci0
Nov 17 16:32:39 rasp3 goveebttemplogger[680]: D-Bus connection ":1.54" Opened
Nov 17 16:32:39 rasp3 goveebttemplogger[680]: D-Bus connection ":1.50" Closed
Nov 17 16:32:39 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: RemoveDevice /org/bluez/hci0/dev_A4_C1_38_56_DF_3C
Nov 17 16:32:39 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: RemoveDevice /org/bluez/hci0/dev_A4_C1_38_04_3F_DE
Nov 17 16:32:39 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery
Nov 17 16:30:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 17 16:30:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 17 16:25:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 17 16:25:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 17 16:20:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 17 16:20:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 17 16:15:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 17 16:15:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 17 16:10:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 17 16:10:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 17 16:05:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-month.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 17 16:05:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-week.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 17 16:05:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 17 16:05:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-month.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 17 16:05:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-week.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 17 16:05:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 17 16:02:40 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 17 16:02:39 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 17 16:02:39 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.freedesktop.DBus.Properties: SetPowered: true
Nov 17 16:02:39 rasp3 goveebttemplogger[680]: Host Controller Address: B8:27:EB:FF:AD:6A BlueZ Adapter Path: /org/bluez/hci0
Nov 17 16:02:39 rasp3 goveebttemplogger[680]: D-Bus connection ":1.50" Opened
Nov 17 16:02:39 rasp3 goveebttemplogger[680]: D-Bus connection ":1.49" Closed
Nov 17 16:02:39 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: RemoveDevice /org/bluez/hci0/dev_A4_C1_38_04_3F_DE
Nov 17 16:02:39 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: RemoveDevice /org/bluez/hci0/dev_A4_C1_38_56_DF_3C
Nov 17 16:02:38 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery
[... cut ...]
Nov 17 15:02:36 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: RemoveDevice /org/bluez/hci0/dev_A4_C1_38_56_DF_3C
Nov 17 15:02:36 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: RemoveDevice /org/bluez/hci0/dev_A4_C1_38_04_3F_DE
Nov 17 15:02:36 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery
Nov 17 15:00:01 rasp3 goveebttemplogger[680]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
```

wcbonner, 2024-11-17:
> After this, I updated the system and rebooted it again. It's now been running for approximately 24 hours without any problems.
>
> I'm not sure what was causing the previous issues, but for now, [...]

@alzuno Glad to know that it seems to be working properly now. I still don't know what could have been causing the problems you saw. The changes I made should make things more robust long term.

I imaged a `Raspberry Pi Zero W Rev 1.1` I had sitting around yesterday to make sure the code still worked on a 32 bit platform and everything worked as expected. I wasn't sure if you were running the 32 bit or 64 bit OS on your platform and wanted to eliminate that is a reason for the code to fail. 

I'm going to close [this issue](https://github.com/wcbonner/GoveeBTTempLogger/issues/80). If it recurs, please reopen it instead of creating a new one.

alzuno, 2024-11-18:
Hi @wcbonner,

This morning, I noticed the system exhibiting the same behavior as before.

While it appears to be running longer than it did previously, it’s still encountering the _Error: Operation already in progress_.

```
-- Journal begins at Thu 2024-07-04 02:14:43 CEST, ends at Mon 2024-11-18 14:55:02 CET. --
Nov 18 14:38:45 rasp3 goveebttemplogger[678]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 18 14:38:45 rasp3 goveebttemplogger[678]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 18 14:38:45 rasp3 goveebttemplogger[678]: /org/bluez/hci0: org.freedesktop.DBus.Properties: SetPowered: true
Nov 18 14:38:45 rasp3 goveebttemplogger[678]: Host Controller Address: B8:27:EB:FF:AD:6A BlueZ Adapter Path: /org/bluez/hci0
Nov 18 14:38:45 rasp3 goveebttemplogger[678]: D-Bus connection ":1.17" Opened
Nov 18 14:38:45 rasp3 goveebttemplogger[678]: D-Bus connection ":1.12" Closed
Nov 18 14:38:45 rasp3 goveebttemplogger[678]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplo>
Nov 18 14:15:01 rasp3 goveebttemplogger[678]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 18 14:15:01 rasp3 goveebttemplogger[678]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-month.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 18 14:15:01 rasp3 goveebttemplogger[678]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-week.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 18 14:15:01 rasp3 goveebttemplogger[678]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 18 14:10:47 rasp3 goveebttemplogger[678]: Writing: /var/cache/goveebttemplogger/gvh-A4C138043FDE-cache.txt
Nov 18 14:10:01 rasp3 goveebttemplogger[678]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-month.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 18 14:10:01 rasp3 goveebttemplogger[678]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-week.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 18 14:10:01 rasp3 goveebttemplogger[678]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 18 14:09:46 rasp3 goveebttemplogger[678]: Writing: /var/cache/goveebttemplogger/gvh-A4C13856DF3C-cache.txt
Nov 18 14:08:45 rasp3 goveebttemplogger[678]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 18 14:08:45 rasp3 goveebttemplogger[678]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 18 14:08:45 rasp3 goveebttemplogger[678]: /org/bluez/hci0: org.freedesktop.DBus.Properties: SetPowered: true
Nov 18 14:08:45 rasp3 goveebttemplogger[678]: Host Controller Address: B8:27:EB:FF:AD:6A BlueZ Adapter Path: /org/bluez/hci0
Nov 18 14:08:45 rasp3 goveebttemplogger[678]: D-Bus connection ":1.12" Opened
Nov 18 14:08:45 rasp3 goveebttemplogger[678]: Reading LastDownload: /var/log/goveebttemplogger/gvh-lastdownload.txt
Nov 18 14:08:44 rasp3 goveebttemplogger[678]: Reading: /var/log/goveebttemplogger/gvh-A4C13856DF3C-2024-11.txt
Nov 18 14:08:43 rasp3 goveebttemplogger[678]: Reading: /var/log/goveebttemplogger/gvh-A4C138043FDE-2024-11.txt
Nov 18 14:08:43 rasp3 goveebttemplogger[678]: Reading: /var/cache/goveebttemplogger/gvh-A4C13856DF3C-cache.txt
Nov 18 14:08:43 rasp3 goveebttemplogger[678]: Reading: /var/cache/goveebttemplogger/gvh-A4C138043FDE-cache.txt
Nov 18 14:08:42 rasp3 goveebttemplogger[678]: Reading: /var/log/goveebttemplogger/gvh-thermometer-types.txt
Nov 18 14:08:42 rasp3 goveebttemplogger[678]: Reading: /var/www/html/goveebttemplogger/gvh-titlemap.txt
Nov 18 14:08:42 rasp3 goveebttemplogger[678]: GoveeBTTempLogger Version 3.20241116.0 Built on: Nov 16 2024 at 21:18:24 (starting)
Nov 18 14:08:42 rasp3 systemd[1]: Started GoveeBTTempLogger service.
-- Boot 1bb0052e0a74460a8032306bcc2d89c3 --
Nov 18 09:56:09 rasp3 systemd[1]: goveebttemplogger.service: Consumed 1min 4.351s CPU time.
Nov 18 09:56:09 rasp3 systemd[1]: Stopped GoveeBTTempLogger service.
Nov 18 09:56:09 rasp3 systemd[1]: goveebttemplogger.service: Succeeded.
Nov 18 09:56:09 rasp3 goveebttemplogger[673]: GoveeBTTempLogger Version 3.20241116.0 Built on: Nov 16 2024 at 21:18:24 (exiting)
Nov 18 09:56:09 rasp3 goveebttemplogger[673]: Writing: /var/log/goveebttemplogger/gvh-thermometer-types.txt
Nov 18 09:56:09 rasp3 goveebttemplogger[673]: D-Bus connection ":1.14" Closed
Nov 18 09:56:09 rasp3 goveebttemplogger[673]: /org/bluez/hci0: org.bluez.Adapter1: RemoveDevice /org/bluez/hci0/dev_A4_C1_38_56_DF_3C
Nov 18 09:56:09 rasp3 goveebttemplogger[673]: /org/bluez/hci0: org.bluez.Adapter1: RemoveDevice /org/bluez/hci0/dev_A4_C1_38_04_3F_DE
Nov 18 09:56:09 rasp3 goveebttemplogger[673]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery
Nov 18 09:56:09 rasp3 systemd[1]: Stopping GoveeBTTempLogger service...
Nov 18 09:56:09 rasp3 goveebttemplogger[673]: ***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
Nov 18 09:55:02 rasp3 goveebttemplogger[673]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 18 09:55:01 rasp3 goveebttemplogger[673]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 18 09:51:05 rasp3 goveebttemplogger[673]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
[... cut ...]
Nov 18 08:46:02 rasp3 goveebttemplogger[680]: ***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
Nov 18 08:32:39 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 18 08:32:39 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 18 08:32:39 rasp3 goveebttemplogger[680]: /org/bluez/hci0: org.freedesktop.DBus.Properties: SetPowered: true
```
Let me know if you’d like to add more details or logs!

P.S: I will try adding the --HCI option to see if it gets data that way and doesn't halt.
Will let you know the results.

alzuno, 2024-11-18:
Thanks @wcbonner 

> Can you remind me of your exact system:

```
pi@rasp3:~ $ cat /proc/cpuinfo
processor	: 0
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 00000000f15507c0
Model		: Raspberry Pi 3 Model B Plus Rev 1.3
pi@rasp3:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
pi@rasp3:~ $ cat /proc/version
Linux version 6.1.21-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
```

> Also check the logs if there are any bluetooth messages interspersed with the goveebttemplogger messages?

```
Nov 18 15:02:21 rasp3 systemd[1]: Stopping GoveeBTTempLogger service...
Nov 18 15:02:21 rasp3 goveebttemplogger[1719]: ***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
Nov 18 15:01:48 rasp3 goveebttemplogger[1719]: Download from device: [A4:C1:38:04:3F:DE] 2024-10-29 15:01:00 2024-11-18 15:01:00 (28800)
Nov 18 15:01:21 rasp3 goveebttemplogger[1719]: Using Controller Address: B8:27:EB:FF:AD:6A
Nov 18 15:01:21 rasp3 goveebttemplogger[1719]: Host Controller Address: B8:27:EB:FF:AD:6A BlueTooth Device ID: 0 HCI Name: hci0
Nov 18 15:01:21 rasp3 goveebttemplogger[1719]: Reading LastDownload: /var/log/goveebttemplogger/gvh-lastdownload.txt
Nov 18 15:01:20 rasp3 goveebttemplogger[1719]: Reading: /var/log/goveebttemplogger/gvh-A4C13856DF3C-2024-11.txt
Nov 18 15:01:19 rasp3 goveebttemplogger[1719]: Reading: /var/log/goveebttemplogger/gvh-A4C138043FDE-2024-11.txt
Nov 18 15:01:19 rasp3 goveebttemplogger[1719]: Reading: /var/cache/goveebttemplogger/gvh-A4C13856DF3C-cache.txt
Nov 18 15:01:19 rasp3 goveebttemplogger[1719]: Reading: /var/cache/goveebttemplogger/gvh-A4C138043FDE-cache.txt
Nov 18 15:01:19 rasp3 goveebttemplogger[1719]: Reading: /var/log/goveebttemplogger/gvh-thermometer-types.txt
[... 136 lines cut ...]
Nov 18 08:46:30 rasp3 goveebttemplogger[677]: Host Controller Address: B8:27:EB:FF:AD:6A BlueZ Adapter Path: /org/bluez/hci0
Nov 18 08:46:30 rasp3 goveebttemplogger[677]: D-Bus connection ":1.12" Opened
Nov 18 08:46:30 rasp3 goveebttemplogger[677]: Reading LastDownload: /var/log/goveebttemplogger/gvh-lastdownload.txt
Nov 18 08:46:29 rasp3 goveebttemplogger[677]: Reading: /var/log/goveebttemplogger/gvh-A4C13856DF3C-2024-11.txt
Nov 18 08:46:28 rasp3 goveebttemplogger[677]: Reading: /var/log/goveebttemplogger/gvh-A4C138043FDE-2024-11.txt
Nov 18 08:46:28 rasp3 goveebttemplogger[677]: Reading: /var/cache/goveebttemplogger/gvh-A4C13856DF3C-cache.txt
Nov 18 08:46:28 rasp3 goveebttemplogger[677]: Reading: /var/cache/goveebttemplogger/gvh-A4C138043FDE-cache.txt
Nov 18 08:46:28 rasp3 goveebttemplogger[677]: Reading: /var/log/goveebttemplogger/gvh-thermometer-types.txt
Nov 18 08:46:28 rasp3 goveebttemplogger[677]: Reading: /var/www/html/goveebttemplogger/gvh-titlemap.txt
Nov 18 08:46:28 rasp3 goveebttemplogger[677]: GoveeBTTempLogger Version 3.20241116.0 Built on: Nov 16 2024 at 21:18:24 (starting)
Nov 18 08:46:28 rasp3 systemd[1]: Started GoveeBTTempLogger service.
Nov 18 08:46:25 rasp3 bluetoothd[593]: Failed to set privacy: Rejected (0x0b)
Nov 18 08:46:25 rasp3 bluetoothd[593]: sap-server: Operation not permitted (1)
Nov 18 08:46:25 rasp3 bluetoothd[593]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
Nov 18 08:46:25 rasp3 bluetoothd[593]: Bluetooth management interface 1.22 initialized
Nov 18 08:46:25 rasp3 bluetoothd[593]: Starting SDP server
Nov 18 08:46:25 rasp3 systemd[1]: Started Bluetooth service.
Nov 18 08:46:25 rasp3 bluetoothd[593]: Bluetooth daemon 5.55
Nov 18 08:46:24 rasp3 systemd[1]: Starting Bluetooth service...
Nov 18 08:46:16 rasp3 systemd[1]: Condition check resulted in Bluetooth service being skipped.
-- Boot 226a8378705f4bd3a3cf07131b38b3c5 --
```

I will update and let you know the results after that.
Thanks,

alzuno, 2024-11-19:
After updating to version `3.20241117.0 Built on: Nov 18 2024`, the same error occurred again—after less than an hour of operation, the system stopped with the same issues.

```
-- Boot a5c05d4040a2476ea0852d2a433f6132 --
Nov 18 18:08:23 rasp3 systemd[1]: goveebttemplogger.service: Consumed 4.012s CPU time.
Nov 18 18:08:23 rasp3 systemd[1]: Stopped GoveeBTTempLogger service.
Nov 18 18:08:23 rasp3 systemd[1]: goveebttemplogger.service: Succeeded.
Nov 18 18:08:23 rasp3 goveebttemplogger[679]: GoveeBTTempLogger Version 3.20241117.0 Built on: Nov 18 2024 at 17:31:00 (exiting)
Nov 18 18:08:23 rasp3 goveebttemplogger[679]: Writing: /var/log/goveebttemplogger/gvh-thermometer-types.txt
Nov 18 18:08:23 rasp3 goveebttemplogger[679]: D-Bus connection ":1.14" Closed
Nov 18 18:08:23 rasp3 goveebttemplogger[679]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3941)
Nov 18 18:08:22 rasp3 systemd[1]: Stopping GoveeBTTempLogger service...
Nov 18 18:08:22 rasp3 goveebttemplogger[679]: ***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
Nov 18 17:55:07 rasp3 goveebttemplogger[679]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 18 17:55:07 rasp3 goveebttemplogger[679]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 18 17:55:07 rasp3 goveebttemplogger[679]: /org/bluez/hci0: org.freedesktop.DBus.Properties: SetPowered: true
Nov 18 17:55:07 rasp3 goveebttemplogger[679]: Host Controller Address: B8:27:EB:FF:AD:6A BlueZ Adapter Path: /org/bluez/hci0
Nov 18 17:55:07 rasp3 goveebttemplogger[679]: D-Bus connection ":1.14" Opened
Nov 18 17:55:07 rasp3 goveebttemplogger[679]: D-Bus connection ":1.12" Closed
Nov 18 17:55:07 rasp3 goveebttemplogger[679]: /org/bluez/hci0: org.bluez.Adapter1: RemoveDevice /org/bluez/hci0/dev_A4_C1_38_56_DF_3C
Nov 18 17:55:07 rasp3 goveebttemplogger[679]: /org/bluez/hci0: org.bluez.Adapter1: RemoveDevice /org/bluez/hci0/dev_A4_C1_38_04_3F_DE
Nov 18 17:55:07 rasp3 goveebttemplogger[679]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3941)
-- Boot adc58a2a4e1c4791b437c82b037d8c6c --
Nov 18 17:50:01 rasp3 goveebttemplogger[3545]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 18 17:50:01 rasp3 goveebttemplogger[3545]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 18 17:45:01 rasp3 goveebttemplogger[3545]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 18 17:45:01 rasp3 goveebttemplogger[3545]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 18 17:42:00 rasp3 goveebttemplogger[3545]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 18 17:42:00 rasp3 goveebttemplogger[3545]: /org/bluez/hci0: org.bluez.Adapter1: SetDiscoveryFilter
Nov 18 17:42:00 rasp3 goveebttemplogger[3545]: /org/bluez/hci0: org.freedesktop.DBus.Properties: SetPowered: true
Nov 18 17:42:00 rasp3 goveebttemplogger[3545]: Host Controller Address: B8:27:EB:FF:AD:6A BlueZ Adapter Path: /org/bluez/hci0
Nov 18 17:42:00 rasp3 goveebttemplogger[3545]: D-Bus connection ":1.21" Opened
Nov 18 17:42:00 rasp3 goveebttemplogger[3545]: Reading LastDownload: /var/log/goveebttemplogger/gvh-lastdownload.txt
Nov 18 17:42:00 rasp3 goveebttemplogger[3545]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 18 17:42:00 rasp3 goveebttemplogger[3545]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 18 17:41:58 rasp3 goveebttemplogger[3545]: Reading: /var/log/goveebttemplogger/gvh-A4C13856DF3C-2024-11.txt
Nov 18 17:41:56 rasp3 goveebttemplogger[3545]: Reading: /var/log/goveebttemplogger/gvh-A4C138043FDE-2024-11.txt
Nov 18 17:41:56 rasp3 goveebttemplogger[3545]: Reading: /var/cache/goveebttemplogger/gvh-A4C13856DF3C-cache.txt
Nov 18 17:41:56 rasp3 goveebttemplogger[3545]: Reading: /var/cache/goveebttemplogger/gvh-A4C138043FDE-cache.txt
Nov 18 17:41:56 rasp3 goveebttemplogger[3545]: Reading: /var/log/goveebttemplogger/gvh-thermometer-types.txt
Nov 18 17:41:56 rasp3 goveebttemplogger[3545]: Reading: /var/www/html/goveebttemplogger/gvh-titlemap.txt
Nov 18 17:41:56 rasp3 goveebttemplogger[3545]: GoveeBTTempLogger Version 3.20241117.0 Built on: Nov 18 2024 at 17:31:00 (starting)
Nov 18 17:41:56 rasp3 systemd[1]: Started GoveeBTTempLogger service.
Nov 18 17:41:56 rasp3 systemd[1]: goveebttemplogger.service: Consumed 9.242s CPU time.
Nov 18 17:41:56 rasp3 systemd[1]: Stopped GoveeBTTempLogger service.
Nov 18 17:41:56 rasp3 systemd[1]: goveebttemplogger.service: Succeeded.
Nov 18 17:41:56 rasp3 goveebttemplogger[3511]: GoveeBTTempLogger Version 3.20241117.0 Built on: Nov 18 2024 at 17:31:00 (exiting)
Nov 18 17:41:56 rasp3 goveebttemplogger[3511]: Writing: /var/log/goveebttemplogger/gvh-thermometer-types.txt
Nov 18 17:41:56 rasp3 systemd[1]: Stopping GoveeBTTempLogger service...
Nov 18 17:41:56 rasp3 goveebttemplogger[3511]: ***************** SIGINT: Caught Ctrl-C, finishing loop and quitting. *****************
Nov 18 17:41:45 rasp3 systemd[1]: goveebttemplogger.service: Current command vanished from the unit file, execution of the command list won't be resumed.
[... cut ...]
Do the Bluetooth errors I mentioned in my previous post sound familiar to you?

```
Nov 18 09:20:59 rasp3 bluetoothd[603]: Failed to set privacy: Rejected (0x0b)
Nov 18 09:20:59 rasp3 bluetoothd[603]: sap-server: Operation not permitted (1)
Nov 18 09:20:59 rasp3 bluetoothd[603]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
Nov 18 09:20:58 rasp3 bluetoothd[603]: Bluetooth management interface 1.22 initialized
Nov 18 09:20:58 rasp3 bluetoothd[603]: Starting SDP server
```

Thank you.

wcbonner, 2024-11-21:
@alzuno I added a bunch more error logging logic with the [3.20241120.0](https://github.com/wcbonner/GoveeBTTempLogger/commit/46be2528441f6ebf3ddd0e0b46d60741b600a6c7) code commit today, but since I still can't figure out why your system is freezing, I don't know if it will improve it at all. 

One thing I added was that I explicitly remove the match rules from the DBus connection before I close it. I was not doing that previously, and don't know if it may have been causing DBus to run out of resources. 

I hope that the --HCI option is continuing to run properly for you. I've also done some cleanup in the past week to have the --HCI code use the exact same decode routines as the DBus code. There had been a few thermometer types I'd added since I stopped doing the HCI development, and it was easier to remove the old code than maintain duplicate routines.

I may try to get my hands on a Raspberry Pi 3 Model B Plus over the holidays that I can run buster on and see if I can duplicate your issue. The Pi4 I have running Buster isn't exhibiting the problem, but my Pi4 and Pi5 have more ram, which may contribute to the problem. You said you are running headless, are you running the OS without running the GUI at all? or are you set to be able to VNC in and use the GUI on occasion?

alzuno, 2024-11-22:
Hi @wcbonner,

I updated to version `3.20241120.0` and let it run without the `--HCI `option for an afternoon. After a few hours, the system halted again.

```
Nov 21 15:56:22 rasp3 goveebttemplogger[689]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 21 15:56:22 rasp3 goveebttemplogger[689]: Host Controller Address: B8:27:EB:FF:AD:6A BlueZ Adapter Path: /org/bluez/hci0
Nov 21 15:56:22 rasp3 goveebttemplogger[689]: D-Bus connection ":1.22" Opened
Nov 21 15:56:22 rasp3 goveebttemplogger[689]: D-Bus connection ":1.21" Closed
Nov 21 15:56:22 rasp3 goveebttemplogger[689]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3791)
Nov 21 15:26:22 rasp3 goveebttemplogger[689]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 21 15:26:22 rasp3 goveebttemplogger[689]: Host Controller Address: B8:27:EB:FF:AD:6A BlueZ Adapter Path: /org/bluez/hci0
Nov 21 15:26:22 rasp3 goveebttemplogger[689]: D-Bus connection ":1.21" Opened
Nov 21 15:26:22 rasp3 goveebttemplogger[689]: D-Bus connection ":1.20" Closed
Nov 21 15:26:22 rasp3 goveebttemplogger[689]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3791)
Nov 21 14:56:22 rasp3 goveebttemplogger[689]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 21 14:56:22 rasp3 goveebttemplogger[689]: Host Controller Address: B8:27:EB:FF:AD:6A BlueZ Adapter Path: /org/bluez/hci0
Nov 21 14:56:22 rasp3 goveebttemplogger[689]: D-Bus connection ":1.20" Opened
Nov 21 14:56:22 rasp3 goveebttemplogger[689]: D-Bus connection ":1.19" Closed
Nov 21 14:56:22 rasp3 goveebttemplogger[689]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3791)
Nov 21 14:30:01 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 21 14:26:22 rasp3 goveebttemplogger[689]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 21 14:26:22 rasp3 goveebttemplogger[689]: Host Controller Address: B8:27:EB:FF:AD:6A BlueZ Adapter Path: /org/bluez/hci0
Nov 21 14:26:22 rasp3 goveebttemplogger[689]: D-Bus connection ":1.19" Opened
Nov 21 14:26:22 rasp3 goveebttemplogger[689]: D-Bus connection ":1.18" Closed
Nov 21 14:26:22 rasp3 goveebttemplogger[689]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3791)
Nov 21 14:25:01 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 21 14:25:01 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 21 14:20:01 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 21 14:20:01 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 21 14:15:02 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 21 14:15:01 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 21 14:10:01 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 21 14:10:01 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 21 14:05:02 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-month.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 21 14:05:02 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-week.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 21 14:05:02 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 21 14:05:02 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-month.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 21 14:05:02 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-week.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 21 14:05:01 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 21 14:00:01 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 21 14:00:01 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 21 13:56:23 rasp3 goveebttemplogger[689]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 21 13:56:22 rasp3 goveebttemplogger[689]: Host Controller Address: B8:27:EB:FF:AD:6A BlueZ Adapter Path: /org/bluez/hci0
Nov 21 13:56:22 rasp3 goveebttemplogger[689]: D-Bus connection ":1.18" Opened
Nov 21 13:56:22 rasp3 goveebttemplogger[689]: D-Bus connection ":1.17" Closed
[... cut ...]
Nov 21 11:45:01 rasp3 goveebttemplogger[689]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
-- Boot 4570649433b34482b7da98bc219dbbc7 --
```

Adding the `--HCI` option seems to have resolved the issue, as the system has been running stable for over 12 hours now.

I'm running Raspberry Pi OS Lite (no GUI) on a Raspberry Pi 3.
Let me know if you have a chance to test this on a similar setup over the holidays, or if I can help with anything else.

Thanks! I'll keep it running with `--HCI` for now.

wcbonner, 2024-11-22:
@alzuno If you want to try one more test, after it starts producing the error run the command:
```
systemctl status bluetooth
```
I'm interested in what it thinks is going on with bluetooth. 

```
wim@WimPi4-Sola:~ $ systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; preset: enabled)
     Active: active (running) since Wed 2024-11-20 12:16:14 PST; 1 day 18h ago
       Docs: man:bluetoothd(8)
   Main PID: 580 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 8738)
        CPU: 21min 59.209s
     CGroup: /system.slice/bluetooth.service
             └─580 /usr/libexec/bluetooth/bluetoothd

Nov 20 12:16:14 WimPi4-Sola bluetoothd[580]: Starting SDP server
Nov 20 12:16:15 WimPi4-Sola bluetoothd[580]: profiles/audio/vcp.c:vcp_init() D-Bus experimental not enabled
Nov 20 12:16:15 WimPi4-Sola bluetoothd[580]: src/plugin.c:plugin_init() Failed to init vcp plugin
Nov 20 12:16:15 WimPi4-Sola bluetoothd[580]: profiles/audio/mcp.c:mcp_init() D-Bus experimental not enabled
Nov 20 12:16:15 WimPi4-Sola bluetoothd[580]: src/plugin.c:plugin_init() Failed to init mcp plugin
Nov 20 12:16:15 WimPi4-Sola bluetoothd[580]: profiles/audio/bap.c:bap_init() D-Bus experimental not enabled
Nov 20 12:16:15 WimPi4-Sola bluetoothd[580]: src/plugin.c:plugin_init() Failed to init bap plugin
Nov 20 12:16:15 WimPi4-Sola bluetoothd[580]: Bluetooth management interface 1.22 initialized
Nov 20 12:16:15 WimPi4-Sola bluetoothd[580]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
Nov 20 12:16:15 WimPi4-Sola bluetoothd[580]: sap-server: Operation not permitted (1)
```
And then try running 
```
systemctl restart bluetooth
```
before resorting to a full reboot.

wcbonner, 2024-11-23:
I mistakenly asked for a buster build on the Pi3. It looks like my code automatically reverted to using the HCI interface on that platform. 

```
pibuster@pibuster:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
pibuster@pibuster:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 44.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 44.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 44.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 44.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a020d3
Serial          : 00000000325a3820
Model           : Raspberry Pi 3 Model B Plus Rev 1.3
pibuster@pibuster:~ $ cat /proc/version
Linux version 5.10.103-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1529 SMP Tue Mar 8 12:21:37 GMT 2022
pibuster@pibuster:~ $ journalctl --boot 0 --unit=bluetooth --unit=goveebttemplogger
-- Logs begin at Fri 2024-11-22 21:13:04 PST, end at Sat 2024-11-23 00:05:01 PST. --
Nov 22 21:13:09 pibuster systemd[1]: Starting Bluetooth service...
Nov 22 21:13:10 pibuster bluetoothd[478]: Bluetooth daemon 5.50
Nov 22 21:13:10 pibuster systemd[1]: Started Bluetooth service.
Nov 22 21:13:10 pibuster bluetoothd[478]: Starting SDP server
Nov 22 21:13:10 pibuster bluetoothd[478]: Bluetooth management interface 1.18 initialized
Nov 22 21:13:10 pibuster bluetoothd[478]: Sap driver initialization failed.
Nov 22 21:13:10 pibuster bluetoothd[478]: sap-server: Operation not permitted (1)
Nov 22 21:13:10 pibuster bluetoothd[478]: Endpoint registered: sender=:1.12 path=/org/bluez/hci0/A2DP/SBC/Source/1
Nov 22 21:13:10 pibuster bluetoothd[478]: Endpoint registered: sender=:1.12 path=/org/bluez/hci0/A2DP/SBC/Source/2
Nov 22 21:13:10 pibuster bluetoothd[478]: Failed to set privacy: Rejected (0x0b)
Nov 22 21:50:21 pibuster systemd[1]: Started GoveeBTTempLogger service.
Nov 22 21:50:21 pibuster goveebttemplogger[7227]: GoveeBTTempLogger Version 3.20241121.0 Built on: Nov 22 2024 at 21:16:07 (starting)
[... 87 lines cut ...]
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13801E144-day.svg With Title: A4C13801E144 (GVH5074)
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Writing: /var/www/html/goveebttemplogger/gvh-A4C1381D5062-day.svg With Title: A4C1381D5062 (GVH5074)
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138266CB7-day.svg With Title: A4C138266CB7 (GVH5074)
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Writing: /var/www/html/goveebttemplogger/gvh-A4C1384D40C1-day.svg With Title: A4C1384D40C1 (GVH5074)
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138815F79-day.svg With Title: A4C138815F79 (GVH5072)
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Writing: /var/www/html/goveebttemplogger/gvh-A4C1388DAA87-day.svg With Title: A4C1388DAA87 (GVH5075)
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Writing: /var/www/html/goveebttemplogger/gvh-A4C1389802C8-day.svg With Title: A4C1389802C8 (GVH5074)
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138D0C87C-day.svg With Title: A4C138D0C87C (GVH5074)
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138E2F9DA-day.svg With Title: A4C138E2F9DA (GVH5075)
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138E653B6-day.svg With Title: A4C138E653B6 (GVH5075)
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138F068A1-day.svg With Title: A4C138F068A1 (GVH5074)
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Writing: /var/www/html/goveebttemplogger/gvh-CC353435204B-day.svg With Title: CC353435204B (GVH5104)
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Writing: /var/www/html/goveebttemplogger/gvh-CE3533331259-day.svg With Title: CE3533331259 (GVH5105)
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Writing: /var/www/html/goveebttemplogger/gvh-D9383333614A-day.svg With Title: D9383333614A (GVH5100)
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Reading LastDownload: /var/log/goveebttemplogger/gvh-lastdownload.txt
Nov 23 00:13:03 pibuster goveebttemplogger[371]: D-Bus connection ":1.20" Opened
[... cut ...]
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Could not get list of adapters from BlueZ over DBus. Reverting to HCI interface.
Nov 23 00:13:03 pibuster goveebttemplogger[371]: D-Bus connection ":1.20" Closed
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Host Controller Address: B8:27:EB:F0:92:8A BlueTooth Device ID: 0 HCI Name: hci0
Nov 23 00:13:03 pibuster goveebttemplogger[371]: Using Controller Address: B8:27:EB:F0:92:8A
```

I'll see if I can get the system rebuilt with bullseye over the weekend.

alzuno, 2024-11-25:
Hi @wcbonner,

I’ve completed the testing you requested, and I can confirm that restarting the Bluetooth service resolved the issue. Here are the details:

**Bluetooth Status Before Restart**
When the system wasn't writing logs, the Bluetooth service was active but encountering errors.

```
pi@rasp3:~ $ systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2024-11-24 13:32:38 CET; 21h ago
       Docs: man:bluetoothd(8)
   Main PID: 593 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 779)
        CPU: 16min 37.453s
     CGroup: /system.slice/bluetooth.service
             └─593 /usr/libexec/bluetooth/bluetoothd

Nov 24 13:32:38 rasp3 systemd[1]: Starting Bluetooth service...
Nov 24 13:32:38 rasp3 bluetoothd[593]: Bluetooth daemon 5.55
Nov 24 13:32:38 rasp3 systemd[1]: Started Bluetooth service.
Nov 24 13:32:39 rasp3 bluetoothd[593]: Starting SDP server
Nov 24 13:32:39 rasp3 bluetoothd[593]: Bluetooth management interface 1.22 initialized
Nov 24 13:32:39 rasp3 bluetoothd[593]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
Nov 24 13:32:39 rasp3 bluetoothd[593]: sap-server: Operation not permitted (1)
Nov 24 13:32:39 rasp3 bluetoothd[593]: Failed to set privacy: Rejected (0x0b)
```
**Bluetooth Service Restart**

```
pi@rasp3:~ $ systemctl restart bluetooth
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to restart 'bluetooth.service'.
Authenticating as: ,,, (pi)
Password: 
==== AUTHENTICATION COMPLETE ===
pi@rasp3:~ $ systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2024-11-25 11:22:31 CET; 14s ago
       Docs: man:bluetoothd(8)
   Main PID: 11785 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 779)
        CPU: 89ms
     CGroup: /system.slice/bluetooth.service
             └─11785 /usr/libexec/bluetooth/bluetoothd

Nov 25 11:22:31 rasp3 systemd[1]: Starting Bluetooth service...
Nov 25 11:22:31 rasp3 bluetoothd[11785]: Bluetooth daemon 5.55
Nov 25 11:22:31 rasp3 systemd[1]: Started Bluetooth service.
Nov 25 11:22:31 rasp3 bluetoothd[11785]: Starting SDP server
Nov 25 11:22:31 rasp3 bluetoothd[11785]: Bluetooth management interface 1.22 initialized
Nov 25 11:22:31 rasp3 bluetoothd[11785]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
Nov 25 11:22:31 rasp3 bluetoothd[11785]: sap-server: Operation not permitted (1)
Nov 25 11:22:31 rasp3 bluetoothd[11785]: Failed to set privacy: Rejected (0x0b)
```
After restarting the Bluetooth service, the system returned to normal operation.

**Log with the system working again after the reboot from the bluetooth service:**
```
pi@rasp3:~ $ journalctl --boot 0 -r --unit=bluetooth --unit=goveebttemplogger
-- Journal begins at Thu 2024-07-04 02:14:43 CEST, ends at Mon 2024-11-25 11:40:02 CET. --
Nov 25 11:40:01 rasp3 goveebttemplogger[667]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-month.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 25 11:40:01 rasp3 goveebttemplogger[667]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-week.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 25 11:40:01 rasp3 goveebttemplogger[667]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 25 11:40:01 rasp3 goveebttemplogger[667]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-month.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 25 11:40:01 rasp3 goveebttemplogger[667]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-week.svg With Title: Room (A4:C1:38:04:3F:DE)
[... 35 lines cut ...]
Nov 25 10:04:55 rasp3 goveebttemplogger[667]: D-Bus connection ":1.63" Opened
Nov 25 10:04:55 rasp3 goveebttemplogger[667]: D-Bus connection ":1.62" Closed
Nov 25 10:04:55 rasp3 goveebttemplogger[667]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery: Error: Operation already in progress /home/pi/GoveeBTTempLogger/goveebttemplogger.cpp(3791)
Nov 25 09:40:03 rasp3 goveebttemplogger[667]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 25 09:40:03 rasp3 goveebttemplogger[667]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 25 09:35:03 rasp3 goveebttemplogger[667]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-week.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 25 09:35:03 rasp3 goveebttemplogger[667]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
Nov 25 09:35:01 rasp3 goveebttemplogger[667]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-week.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 25 09:35:01 rasp3 goveebttemplogger[667]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 25 09:34:56 rasp3 goveebttemplogger[667]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Nov 25 09:34:55 rasp3 goveebttemplogger[667]: Host Controller Address: B8:27:EB:FF:AD:6A BlueZ Adapter Path: /org/bluez/hci0
Nov 25 09:34:55 rasp3 goveebttemplogger[667]: D-Bus connection ":1.62" Opened
[... cut ...]
Nov 25 09:25:01 rasp3 goveebttemplogger[667]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138043FDE-day.svg With Title: Room (A4:C1:38:04:3F:DE)
Nov 25 09:20:01 rasp3 goveebttemplogger[667]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13856DF3C-day.svg With Title: Office (A4:C1:38:56:DF:3C)
```
I hope this information will be useful to you.

About the buster build reverting to use the HCI interface, it seems not to be the same on bullseye. As you can see in the logs if i don't use the --HCI option, the system uses Bluez.

blackketter, 2025-08-09:
Had this happen to me yesterday with the latest code too, seemed to recover on its own.  Raspi zero 2 w, bookworm. 
```
Aug 08 12:01:12 fridge goveebttemplogger[623]:    Last Download from device: [C4:30:38:36:4C:77] 2025-07-15 23:59:00 (GVH5104)
Aug 08 12:01:28 fridge goveebttemplogger[623]:    Last Download from device: [A4:C1:38:11:B9:AF] 2025-08-02 17:26:00 (GVH5075)
Aug 08 12:01:32 fridge goveebttemplogger[623]:    Last Download from device: [D9:38:30:34:02:56] 2025-08-02 01:43:00 (GVH5104)
Aug 08 12:01:32 fridge goveebttemplogger[623]:    Last Download from device: [A4:C1:38:1A:35:2C] 2025-08-03 19:57:00 (GVH5075)
Aug 08 12:01:58 fridge goveebttemplogger[623]:    Last Download from device: [C4:30:38:36:4C:77] 2025-07-15 23:59:00 (GVH5104)
Aug 08 12:05:01 fridge goveebttemplogger[623]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13811B9AF-day.svg With Title: Freezer2
Aug 08 12:05:01 fridge goveebttemplogger[623]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13811B9AF-week.svg With Title: Freezer2
Aug 08 12:05:01 fridge goveebttemplogger[623]: Writing: /var/www/html/goveebttemplogger/gvh-A4C13811B9AF-month.svg With Title: Freezer2
Aug 08 12:05:01 fridge goveebttemplogger[623]: Writing: /var/www/html/goveebttemplogger/gvh-A4C1381A352C-day.svg With Title: Refrigerator
Aug 08 12:05:01 fridge goveebttemplogger[623]: Writing: /var/www/html/goveebttemplogger/gvh-A4C1381A352C-week.svg With Title: Refrigerator
Aug 08 12:05:01 fridge goveebttemplogger[623]: Writing: /var/www/html/goveebttemplogger/gvh-A4C1381A352C-month.svg With Title: Refrigerator
Aug 08 12:05:01 fridge goveebttemplogger[623]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138E43FC8-day.svg With Title: Freezer
Aug 08 12:05:01 fridge goveebttemplogger[623]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138E43FC8-week.svg With Title: Freezer
Aug 08 12:05:01 fridge goveebttemplogger[623]: Writing: /var/www/html/goveebttemplogger/gvh-A4C138E43FC8-month.svg With Title: Freezer
Aug 08 12:05:01 fridge goveebttemplogger[623]: Writing: /var/www/html/goveebttemplogger/gvh-C43038364C77-day.svg With Title: Lab
Aug 08 12:05:01 fridge goveebttemplogger[623]: Writing: /var/www/html/goveebttemplogger/gvh-C43038364C77-week.svg With Title: Lab
Aug 08 12:05:01 fridge goveebttemplogger[623]: Writing: /var/www/html/goveebttemplogger/gvh-C43038364C77-month.svg With Title: Lab
Aug 08 12:05:01 fridge goveebttemplogger[623]: Writing: /var/www/html/goveebttemplogger/gvh-D93830340256-day.svg With Title: Kitchen
Aug 08 12:05:01 fridge goveebttemplogger[623]: Writing: /var/www/html/goveebttemplogger/gvh-D93830340256-week.svg With Title: Kitchen
Aug 08 12:05:01 fridge goveebttemplogger[623]: Writing: /var/www/html/goveebttemplogger/gvh-D93830340256-month.svg With Title: Kitchen
Aug 08 22:40:33 fridge goveebttemplogger[623]: /org/bluez/hci0: org.bluez.Adapter1: StopDiscovery
Aug 08 22:40:33 fridge goveebttemplogger[623]: D-Bus connection ":1.12920" Closed
Aug 08 22:40:33 fridge goveebttemplogger[623]: D-Bus connection ":1.14076" Opened
Aug 08 22:40:33 fridge goveebttemplogger[623]: /: org.freedesktop.DBus.ObjectManager: GetManagedObjects
Aug 08 22:40:33 fridge goveebttemplogger[623]: Host Controller Address: B8:27:EB:CC:A0:FA BlueZ Adapter Path: /org/bluez/hci0
Aug 08 22:40:33 fridge goveebttemplogger[623]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery: Error: Operation already in progress /home/dean/GoveeBTTempLogger/goveebttemplogger.cp>
Aug 08 22:40:33 fridge goveebttemplogger[623]: D-Bus connection ":1.14076" Closed
Aug 08 22:40:33 fridge goveebttemplogger[623]: D-Bus connection ":1.14077" Opened
Aug 08 22:40:33 fridge goveebttemplogger[623]: /: org.freedesktop.DBus.ObjectManager: GetManagedObjects
Aug 08 22:40:33 fridge goveebttemplogger[623]: Host Controller Address: B8:27:EB:CC:A0:FA BlueZ Adapter Path: /org/bluez/hci0
Aug 08 22:40:33 fridge goveebttemplogger[623]: /org/bluez/hci0: org.bluez.Adapter1: StartDiscovery
Aug 08 22:40:33 fridge goveebttemplogger[623]:    Last Download from device: [A4:C1:38:E4:3F:C8] 2025-08-02 08:09:00 (GVH5075)
Aug 08 22:40:34 fridge goveebttemplogger[623]:    Last Download from device: [D9:38:30:34:02:56] 2025-08-02 01:43:00 (GVH5104)
Aug 08 22:40:35 fridge goveebttemplogger[623]:    Last Download from device: [A4:C1:38:1A:35:2C] 2025-08-03 19:57:00 (GVH5075)
Aug 08 22:40:35 fridge goveebttemplogger[623]:    Last Download from device: [A4:C1:38:11:B9:AF] 2025-08-02 17:26:00 (GVH5075)
Aug 08 22:40:37 fridge goveebttemplogger[623]:    Last Download from device: [C4:30:38:36:4C:77] 2025-07-15 23:59:00 (GVH5104)
Aug 08 22:41:03 fridge goveebttemplogger[623]: Writing: /var/cache/goveebttemplogger/gvh-A4C13811B9AF-cache.txt
Aug 08 22:41:03 fridge goveebttemplogger[623]: Writing: /var/cache/goveebttemplogger/gvh-A4C1381A352C-cache.txt
Aug 08 22:41:04 fridge goveebttemplogger[623]: Writing: /var/cache/goveebttemplogger/gvh-A4C138E43FC8-cache.txt
Aug 08 22:41:04 fridge goveebttemplogger[623]: Writing: /var/cache/goveebttemplogger/gvh-C43038364C77-cache.txt
Aug 08 22:41:04 fridge goveebttemplogger[623]: Writing: /var/cache/goveebttemplogger/gvh-D93830340256-cache.txt
Aug 08 22:41:51 fridge goveebttemplogger[623]:    Last Download from device: [C4:30:38:36:4C:77] 2025-07-15 23:59:00 (GVH5104)
Aug 08 22:42:06 fridge goveebttemplogger[623]:    Last Download from device: [D9:38:30:34:02:56] 2025-08-02 01:43:00 (GVH5104)
[... cut ...]
Jul 26 22:38:23 fridge bluetoothd[383]: Bluetooth management interface 1.23 initialized
Jul 26 22:38:23 fridge systemd[1]: Started bluetooth.service - Bluetooth service.
Jul 26 22:38:23 fridge bluetoothd[383]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
Jul 26 22:38:23 fridge bluetoothd[383]: sap-server: Operation not permitted (1)
Aug 08 22:40:33 fridge bluetoothd[383]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
```

blackketter, 2025-08-10:
Fingers crossed. 

In the mean time, is there a timeout that could be adjusted to have the bluetooth restart after, say, 15 minutes of missing input?

duncanyoyo1, 2026-01-15:
First I want to say thank you for this great software, I really appreciate all the effort that went into reverse engineering these devices and the BT communications.

I use my RPi as a build machine for RetroArch cores and other emulators for retro handhelds ( The Anbernic RGXX H700 line specifically, same arch as the CPU in the RPi3 )

And I have noticed, when the system is under heavy load, the bluetooth will drop out. 

The log will get spammed with 

    Jan 15 05:49:06 raspberry goveebttemplogger[98484]: D-Bus connection ":1.11940" Closed
    Jan 15 05:49:06 raspberry goveebttemplogger[98484]: D-Bus connection ":1.11941" Opened

The only way to get the logger to start working again is to ```hciconfig hci0 down && hciconfig hci0 up```

I have --restart set, but it never triggers.

So I ended up writing my own watchdog service that looks at the modified date of the log files. If it doesn't see an update in a set amount of time, it kicks bluetooth and restarts the logger service.

Another small issue I had that isn't related to this specific bug, I live in a place where I frequently lose power. Often when this happens I get garbage written to the last line of the log files, as the process has an open handle to the file but it's not properly flushed. I added a section to the service that runs a small script that will parse and remove the garbage lines of my log files before starting the goveebttemplogger process. If I don't, the logger process will hang on reading the files and never actually start.

With those 2 changes the system is now basically fully tolerant to power losses and BT dropouts under load.

I know this doesn't fix the issue, more of a workaround, but perhaps it might give some ideas of a better way to at least mitigate it.