wcbonner/GoveeBTTempLogger #81: Downloading historical data

jer194, 2025-01-04:
xref #62 

Hi again :-)

Some more feedback:

- 1: Wrong battery level in historical data?

> goveebttemplogger -d -H -l . -c

> cat gvh-D23832301F19-2025-01.txt | grep 2024-12-15 | head -1
2024-12-15 07:45:00     2.3892  89.2    70

'70' seems to be the current battery level.
On Dec 15 it was around 80%.

wcbonner, 2025-01-04:
> xref #62
> 
> Hi again, just some more feedback on Download mode :-) :
> 
> * 1: Wrong battery level in historical data?
> 
> > goveebttemplogger -d -H -l . -c
> 
> > cat gvh-D23832301F19-2025-01.txt | grep 2024-12-15 | head -1
> > 2024-12-15 07:45:00     2.3892  89.2    70
> 
> '70' seems to be the current battery level. On Dec 15 it was around 80%. If the true level back then is not available, maybe it could be indicated by something like -1.

This is by design, since the historical data from the device does not have the battery data. The data is stored as an int so I could use a -1 in the storage for it, but the repercussions of making a change to the data in all instances, writing to logs, graphing, reading from logs, etc, Is something that I don't want to think about for a while. 

> * 2: Is it possible to just download historical data, then stop?
>   Right now, after 'download' it goes into scanning mode, and starts collecting regular data after >4 min.

There are a bunch of implications related to this with multiple thermometers. The very nature of the download is that you can't connect to a device until you've received an advertising packet, and there's a short window of time that you can stop scanning for packets and connect to the device. If the connection fails, then I've got to put it back in scanning mode and wait for another advertising packet to arrive. 

> * 3: An existing 'gvh-lastdownload.txt' file seems to prevent future downloads. It contains something like
>   D2:38:32:30:1F:19       2025-01-04T11:53:00
> 
> Instead, it goes immediately into regular scanning mode. After removing the 'gvh-lastdownload.txt' file, download mode works again. Maybe the behavior could be changed, to just download missing historical data, then stop?

The code is set to not download data more often than every two weeks because it uses much more battery than just listening to the advertisements. That date is the UTC date/time of the last successful download from a thermometer. For code related to the 14 days, see: https://github.com/wcbonner/GoveeBTTempLogger/blob/6bb7e7b7124e437a4cba443ad9d2b5e72d4b2228/goveebttemplogger.cpp#L225 and https://github.com/wcbonner/GoveeBTTempLogger/blob/6bb7e7b7124e437a4cba443ad9d2b5e72d4b2228/goveebttemplogger.cpp#L5094 and  https://github.com/wcbonner/GoveeBTTempLogger/blob/6bb7e7b7124e437a4cba443ad9d2b5e72d4b2228/goveebttemplogger.cpp#L3429 

> * 4: Question about the documentation:
> 
> "" -t (--time) Sets the frequency data is written to the logs. The time frequency of the data in the logs is based on bluetooth announcement time. This is how often the logfile is written. "" I assume that's the time in seconds, and in the context of data download it does not have any significance?

I think you are correct. The --time parameter is purely based on the main program loop time. So if you have it set so that it won't write the log file in less than five minute increments, and you download three weeks of historical data, all that data is stored in an internal buffer until the five minute interval is up, or an INT or HUP signal is received causing it to exit the main loop and flush its data to the file.

jer194, 2025-01-04:
Thanks for your detailed explanations!
All this is not as trivial as I had thought obviously ...
Still, it is already quite useful the way it is, thanks again!

closed 2025-02-13 by wcbonner