wcbonner/GoveeBTTempLogger #88: Corruption in log file on system reboot

blackketter, 2025-07-20:
My system crashed (Raspi Zero W, running bookworm) and the log files appear to have some corruption at the time of the crash.  Below is an example where it happens at 2025-07-20 11:39:35.  This appeared to happen in all 5 of my log files.  

This has happened before with an older version of the OS and GoveeBTTempLogger and I had to edit the log files so they would parse properly.

I've got the logging set to the default 60 second interval.

[gvh-A4C13811B9AF-2025-07.txt](https://github.com/user-attachments/files/21335308/gvh-A4C13811B9AF-2025-07.txt)

wcbonner, 2025-07-20:
I'll have to look and see if there's a better method of ignoring corruption during the reading of log files than I'm currently using. 

I looked at your data file in windows notepad.

```
2025-07-20 11:39:35	-16.1	48.4	54
                                                                                                                                        2025-07-20 13:23:26	-20.1	67.2	53
2025-07-20 13:23:28	-20	67.2	53
```

Do you know what the characters are on the beginning of the line where it started writing new data?

In my own problems with corruption early on I found that NULLs were injected in the file. I now strip all preceding nulls https://github.com/wcbonner/GoveeBTTempLogger/blob/4c2940cacf533d8fe868b8daa70368161f3ee5a9/goveebttemplogger.cpp#L396 but could possibly change it to simply throw away a logged line that doesn't start with a digit, since the line should start with a date. Looking at this code reminds me of how long ago I wrote it and the lack of error handling. 

The fact that you get corruption in multiple current files indicates that the corruption is happening at the OS level. I never have more than one file open at a time for writing.

blackketter, 2025-07-23:
To be clear, the code before your most recent change did handle the corrupted files, but my much older version did not (though I did confirm that this version had the null check.)  

 In any case, things seem to be working ok with the latest code.

wcbonner, 2025-07-25:
> To be clear, the code before your most recent change did handle the corrupted files, but my much older version did not (though I did confirm that this version had the null check.)
> 
> In any case, things seem to be working ok with the latest code.

Good to know that the latest fix seems to take care of the issue.

closed 2025-07-25 by wcbonner