wcbonner/GoveeBTTempLogger #89: Debian Trixie issue running goveebttemplogger

babelouest, 2025-08-26:
Hello,

I've upgraded my RPi to Debian Trixie and it seems that goveebttemplogger is no longer working. Debian Trixie is not officially released for Raspberry pi OS though, but I think it's worth mentioning if the issue comes from goveebttemplogger.

After the upgrade, goveebttemplogger stopped logging the govee sensors. Journald output is full of these error messages:

```
aoû 26 09:17:32 bedevere goveebttemplogger[679]: /: org.freedesktop.DBus.ObjectManager: GetManagedObjects
aoû 26 09:17:32 bedevere goveebttemplogger[679]: Host Controller Address: XX:XX:XX:XX:XX:XX BlueZ Adapter Path: /org/bluez/hci0
aoû 26 09:17:32 bedevere goveebttemplogger[679]: /org/bluez/hci0: org.freedesktop.DBus.Properties: SetPowered: true Error: Failed /home/nico/GoveeBTTempLogger/goveebttemplogger.cpp(3895)
aoû 26 09:17:32 bedevere goveebttemplogger[679]: D-Bus connection ":1.639362" Closed
aoû 26 09:17:32 bedevere goveebttemplogger[679]: D-Bus connection ":1.639363" Opened
```

I've reinstalled the debian package from the [last release](https://github.com/wcbonner/GoveeBTTempLogger/releases/tag/v3.20250218.3) and also tried to build from source, I still have the same error.

Do you have an idea where the problem is?

wcbonner, 2025-08-27:
> Hello,
>
> I've upgraded my RPi to Debian Trixie and it seems that goveebttemplogger is no longer working. Debian Trixie is not officially released for Raspberry pi OS though, but I think it's worth [...]

I've not yet tried Trixie. I was thinking about trying a build of it last week but needed time for other things. I looked at old Raspberry release notes and it seems that the official raspberry derivation of Debian usually runs about four months after the Debian release. That puts the release in early December. 

I don't know what the problem is. Can you report the version of BlueZ on Trixie in comparison with Bookworm?

Did you do a clean install of Trixie or upgrade an existing system by editing the apt files?

I won't have time to play with spare hardware until next weekend. The earliest I expect to do anything would be Labor Day.

babelouest, 2025-08-27:
Hello @wcbonner ,

> I've not yet tried Trixie. I was thinking about trying a build of it last week but needed time for other things. I looked at old Raspberry release notes and it seems that the official raspberry [...]

That's right, raspberry pi os is delayed with Debian official release, I wanted to early test goveebttemplogger with the new OS though.

> I don't know what the problem is. Can you report the version of BlueZ on Trixie in comparison with Bookworm?

Trixie has bluez 5.82, where Bookworm has bluez 5.66: https://packages.debian.org/search?searchon=names&keywords=bluez

Also, if relevant, Trixie has dbus 1.16.2 and Bookworm has dbus 1.14.10: https://packages.debian.org/search?searchon=names&keywords=dbus

> Did you do a clean install of Trixie or upgrade an existing system by editing the apt files?

I've upgraded from bookworm to trixie, I don't know if there are rc installer for raspberry pi os trixie yet.

> I won't have time to play with spare hardware until next weekend. The earliest I expect to do anything would be Labor Day.

That's ok, there's no rush for me, I can live without knowing my living room temperature... I can also test some patches if you want me to.

wcbonner, 2025-08-29:
@babelouest if you are not using Bluetooth for anything else on your Trixie Pi, I'd be interested in knowing if adding the --HCI option still works. It uses the hardware interface in BlueZ that's been deprecated, but has still worked in Bookworm. It takes exclusive access to the Bluetooth controller and listens for advertisements.

babelouest, 2025-08-29:
Hello @wcbonner ,

If I run with the option `--HCI`, I go further, but it still doesn't work:

```
goveebttemplogger@host:/root$ /usr/local/bin/goveebttemplogger --verbose 0 --log /var/log/goveebttemplogger --time 60 --download 7  --restart 3 --svg /var/www/html/goveebttemplogger --battery 8 --minmax 8 --cache /var/cache/goveebttemplogger --HCI
GoveeBTTempLogger Version 3.20250818.3 Built on: Aug 26 2025 at 07:47:46 (starting)
Reading: /var/log/goveebttemplogger/gvh-lastdownload.txt
Reading: /var/log/goveebttemplogger/gvh-thermometer-types.txt
Reading: /var/cache/goveebttemplogger/gvh-A4Cxxxxxxxxx-cache.txt
Reading: /var/log/goveebttemplogger/gvh-A4Cxxxxxxxxx-2025-08.txt
Host Controller Address: xx:xx:xx:xx:xx:xx BlueTooth Device ID: 0 HCI Name: hci0
[                   ] Error: Bluetooth device not found
GoveeBTTempLogger Version 3.20250818.3 Built on: Aug 26 2025 at 07:47:46 (exiting)
```

wcbonner, 2025-08-30:
> Hello [@wcbonner](https://github.com/wcbonner) ,
>
> If I run with the option `--HCI`, I go further, but it still doesn't work:
>
> ```
> goveebttemplogger@host:/root$ /usr/local/bin/goveebttemplogger [...]

This is interesting and may provide clues. It's possible that the Bluetooth controller in the Pi isn't supported without specific raspberry device drivers that haven't been updated for Trixie yet. Purely speculation on my part right now, but when I get around to trying this I may also plug in a second USB Bluetooth controller to see if it acts differently.

babelouest, 2025-08-30:
Indeed, if I run `bluetoothctl`, I have the following status in show:

```
Powered: no
PowerState: off-blocked
```

and if I run `power on`, I have the following error:

```
[bluetoothctl]> power on
Failed to set power on: org.bluez.Error.Failed
```

wcbonner, 2025-10-02:
@babelouest and @jer194 Updates on Trixie:

I just ran an in-place upgrade of my Pi5 to Trixie since it was Raspberry published this today: https://www.raspberrypi.com/news/trixie-the-new-version-of-raspberry-pi-os/ 

I mostly followed https://forums.raspberrypi.com/viewtopic.php?t=392376&sid=3c9c7a753f5259812e92beb280c59c64 with some information from https://linuxiac.com/how-to-upgrade-to-debian-13-trixie-from-12-bookworm/

I used these commands:

```
mkdir ~/apt ; cp /etc/apt/sources.list ~/apt ; cp -r /etc/apt/sources.list.d/ ~/apt
sudo sed -i 's/bookworm/trixie/g' /etc/apt/sources.list
sudo sed -i 's/bookworm/trixie/g' /etc/apt/sources.list.d/*
sudo apt update
sudo apt purge -y raspberrypi-ui-mods
sudo apt autoremove -y
sudo apt full-upgrade -y -o Dpkg::Options::="--force-confdef" -o Dpkg::Options::="--force-confnew" --purge --auto-remove
sudo apt install -y rpd-wayland-all rpd-x-all
sync ; sync ; sudo systemctl reboot
```
I then made sure I had the latest code and rebuilt the packages:
```
pushd  ~/GoveeBTTempLogger/ && git pull && popd && cmake -S ~/GoveeBTTempLogger -B ~/GoveeBTTempLogger/build && cmake --build ~/GoveeBTTempLogger/build && pushd ~/GoveeBTTempLogger/build && ctest -V . && cpack . && popd
```
I did a test run using the deprecated HCI interface, which still works:
```
/home/wim/GoveeBTTempLogger/build/goveebttemplogger --HCI --verbose 2 --log /home/wim/gvh/log/ --time 60 --svg /home/wim/gvh/svg/ --battery 8 --minmax 8 --cache /home/wim/gvh/cache/ --download 1
```
I did a test run with the dBus interface and it also ran properly:
```
/home/wim/GoveeBTTempLogger/build/goveebttemplogger --verbose 2 --log /home/wim/gvh/log/ --time 60 --svg /home/wim/gvh/svg/ --battery 8 --minmax 8 --cache /home/wim/gvh/cache/ --download 1
```

closed 2025-10-05 by wcbonner